istoreos/istoreos #600: 请问是否有适配NanoPi Neo3的计划？

windskyxb, 2023-02-18:
Neo3就是单网口的R2S

jjm2473, 2023-02-18:
直接用r2s固件有什么问题

windskyxb, 2023-02-19:
无法启动

windskyxb, 2023-02-19:
> 直接用r2s固件有什么问题

无法启动

windskyxb, 2023-02-19:
> 直接用r2s固件有什么问题

另一台指示灯正常，应该是正常启动了，但我无法进入管理后台

jjm2473, 2023-02-19:
试试另一个网口

windskyxb, 2023-02-19:
> 试试另一个网口

但Neo3是单网口(ˉ▽ˉ；)...

windskyxb, 2023-02-19:
> 试试另一个网口

[image]

jjm2473, 2023-02-19:
@windskyxb 你把neo3的网口接到主路由器，主路由应该会给neo3分配ip，电脑连主路由，看看分配的ip是什么，对应的ip应该可以打开后台

windskyxb, 2023-02-19:
> @windskyxb 你把neo3的网口接到主路由器，主路由应该会给neo3分配ip，电脑连主路由，看看分配的ip是什么，对应的ip应该可以打开后台

[image]
更换内存卡后r2s的固件可以正常运行在neo3上了

closed 2023-02-19 by windskyxb

WMZMW, 2024-11-30:
提示：NEO3刷入最新版R2S固件后，在通过浏览器访问管理界面时可能会出现ERR_CONNECTION_REFUSED，可通过以下方法解决：
1.使用USB转串口模块（如CH340等）连接NEO3板载TTL（GND相连，RX与TX交叉相连，波特率1500000），按回车进入后台
2.输入命令quickstart进入快速设置
3.选择Allow WAN Access
4.此时管理界面便可正常访问
注：实测CH340模块连接不稳定，可能会出现乱码情况，可更换更高级的串口模块，或按以下方法盲操：quickstart按下回车后，按两下方向键下，按两下回车，按四下方向键下，按一下回车