istoreos/istoreos #601: [√ ] 在路由器里安装docker又安装了宝塔都正常工作外网也能访问，重启路由器后，docker也能自动启动镜像，但是宝塔网管理页面无法访问，需要反复重启几次才有概率成功

sd3970566, 2023-02-18:
在路由器里安装docker又安装了宝塔都正常工作外网也能访问，重启路由器后，docker也能自动启动镜像，但是宝塔网管理页面无法访问，需要反复重启几次才有概率成功
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [√ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在路由器里安装docker又安装了宝塔都正常工作外网也能访问，重启路由器后，docker也能自动启动镜像，但是宝塔网管理页面无法访问，需要反复重启几次才有概率成功！有时候意外断电的情况下，导致重启路由器。网站服务是否正常开启纯靠运气··就不太给力啊，希望能够修复。  

### (2) 路由器型号和固件版本

A：固件版本
iStoreOS 21.02.3 2023020316
内核版本
5.4.188

### (3) 详细日志和/或截图

A：我路由器重启了好几遍。 日志很多。具体也不知道从哪里发起。我就docker的事件随便就近的时间发一段  和  路由器的日志 就近的发一段吧
 【docker日志】>这个是最后一次重启成功 启动宝塔服务后的日志。我就没再重启了
：[2023-02-19 00:04:03] container exec_die Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:08] container exec_create: /bin/sh -c curl -sf http://127.0.0.1:8888/btpanel || exit 1 Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:08] container exec_start: /bin/sh -c curl -sf http://127.0.0.1:8888/btpanel || exit 1 Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:08] container exec_die Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:13] container exec_create: /bin/sh -c curl -sf http://127.0.0.1:8888/btpanel || exit 1 Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:13] container exec_start: /bin/sh -c curl -sf http://127.0.0.1:8888/btpanel || exit 1 Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:13] container exec_die Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:18] container exec_create: /bin/sh -c curl -sf http://127.0.0.1:8888/btpanel || exit 1 Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:18] container exec_start: /bin/sh -c curl -sf http://127.0.0.1:8888/btpanel || exit 1 Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:18] container exec_die Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:23] container exec_create: /bin/sh -c curl -sf http://127.0.0.1:8888/btpanel || exit 1 Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:23] container exec_start: /bin/sh -c curl -sf http://127.0.0.1:8888/btpanel || exit 1 Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:24] container exec_die Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:29] container exec_create: /bin/sh -c curl -sf http://127.0.0.1:8888/btpanel || exit 1 Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:29] container exec_start: /bin/sh -c curl -sf http://127.0.0.1:8888/btpanel || exit 1 Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 Container Name: centos
[2023-02-19 00:04:29] container exec_die Container ID: 1530e8d2cfff75c14d90cb8bd3e292ec99d713410fced0419d161d52cd974606 

========================我是分隔符=========================
【路由器日志】>这个是最后一次重启成功 启动宝塔服务后的日志。我就没再重启了
Sat Feb 18 23:49:57 2023 daemon.info dnsmasq[4221]: using only locally-known addresses for domain invalid
Sat Feb 18 23:49:57 2023 daemon.info dnsmasq[4221]: using only locally-known addresses for domain bind
Sat Feb 18 23:49:57 2023 daemon.info dnsmasq[4221]: using only locally-known addresses for domain lan
Sat Feb 18 23:49:57 2023 daemon.warn dnsmasq[4221]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Sat Feb 18 23:49:57 2023 daemon.info dnsmasq[4221]: read /etc/hosts - 4 addresses
Sat Feb 18 23:49:57 2023 daemon.info dnsmasq[4221]: read /tmp/hosts/dhcp.cfg01411c - 6 addresses
Sat Feb 18 23:49:57 2023 authpriv.info dropbear[4311]: Not backgrounding
Sat Feb 18 23:49:57 2023 kern.warn kernel: [   12.126495] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
Sat Feb 18 23:49:57 2023 user.notice : Added device handler type: bonding
Sat Feb 18 23:49:57 2023 user.notice : Added device handler type: 8021ad
Sat Feb 18 23:49:57 2023 user.notice : Added device handler type: 8021q
Sat Feb 18 23:49:57 2023 user.notice : Added device handler type: macvlan
Sat Feb 18 23:49:57 2023 user.notice : Added device handler type: veth
Sat Feb 18 23:49:57 2023 user.notice : Added device handler type: bridge
Sat Feb 18 23:49:57 2023 user.notice : Added device handler type: Network device
Sat Feb 18 23:49:57 2023 user.notice : Added device handler type: tunnel
Sat Feb 18 23:49:57 2023 user.notice wan_drop: current proto pppoe
Sat Feb 18 23:49:57 2023 daemon.notice procd: /etc/rc.d/S39blockmount: /dev/sda1 UUID=292A-A6AD TYPE=vfat LABEL=kernel MOUNT=/boot
Sat Feb 18 23:49:57 2023 daemon.notice procd: /etc/rc.d/S39blockmount: /dev/sda2 UUID=0531f9ed-9895142a-7e7cd6b6-ad2d7ed5 TYPE=squashfs LABEL= MOUNT=/rom
Sat Feb 18 23:49:57 2023 daemon.notice procd: /etc/rc.d/S39blockmount: /dev/sda3 UUID=5b0c6698-8a6a-4f09-a365-41fd8ec55e54 TYPE=ext4 LABEL=etc MOUNT=/overlay
Sat Feb 18 23:49:57 2023 daemon.notice procd: /etc/rc.d/S39blockmount: /dev/sdb1 UUID=e4a21fee-be00-47dd-85e5-9815ecddc8a9 TYPE=ext4 LABEL= MOUNT=/mnt/sata2-1
Sat Feb 18 23:49:57 2023 daemon.notice procd: /etc/rc.d/S45nfs: exportfs: could not open /var/lib/nfs/.etab.lock for locking: errno 2 (No such file or directory)
Sat Feb 18 23:49:57 2023 daemon.notice procd: /etc/rc.d/S45nfs: point 1
Sat Feb 18 23:49:57 2023 daemon.notice procd: /etc/rc.d/S45nfs: exportfs: could not open /var/lib/nfs/.etab.lock for locking: errno 2 (No such file or directory)
Sat Feb 18 23:49:57 2023 daemon.notice procd: /etc/rc.d/S45nfs: exportfs: can't lock /var/lib/nfs/etab for writing
Sat Feb 18 23:49:57 2023 daemon.notice procd: /etc/rc.d/S45nfs: point 2
Sat Feb 18 23:49:57 2023 daemon.notice procd: /etc/rc.d/S45nfs: point 3
Sat Feb 18 23:49:57 2023 cron.err crond[5112]: crond (busybox 1.33.2) started, log level 5
Sat Feb 18 23:49:58 2023 kern.info kernel: [   12.940308] 8021q: adding VLAN 0 to HW filter on device eth1
Sat Feb 18 23:49:58 2023 kern.info kernel: [   12.942565] br-lan: port 1(eth1) entered blocking state
[... 94 lines cut ...]
Sat Feb 18 23:50:01 2023 daemon.notice procd: /etc/rc.d/S99nfsd: point 3
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[4221]: exiting on receipt of SIGTERM
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: Connected to system UBus
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: started, version 2.85 cachesize 150
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: DNS service limited to local subnets
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: UBus support enabled: connected to system bus
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq-dhcp[8991]: DHCP, IP range 192.168.9.100 -- 192.168.9.249, lease time 12h
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using only locally-known addresses for domain test
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using only locally-known addresses for domain onion
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using only locally-known addresses for domain localhost
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using only locally-known addresses for domain local
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using only locally-known addresses for domain invalid
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using only locally-known addresses for domain bind
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using only locally-known addresses for domain lan
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: reading /tmp/resolv.conf.d/resolv.conf.auto
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using only locally-known addresses for domain test
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using only locally-known addresses for domain onion
[... cut ...]
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using only locally-known addresses for domain local
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using only locally-known addresses for domain invalid
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using only locally-known addresses for domain bind
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using only locally-known addresses for domain lan
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using nameserver 114.114.114.114#53
Sat Feb 18 23:50:02 2023 daemon.info dnsmasq[8991]: using nameserver 223.5.5.5#53

jjm2473, 2023-02-22:
先确认一下，容器之间与依赖关系吗？如果有，那可能失败，这是docker自己的问题

jjm2473, 2023-02-22:
你发日志应该发启动失败的情况下的日志

closed 2023-03-13 by jjm2473