istoreos/istoreos #603: R2S最新固件 eth1网口自动协商100M全双工

zhaokelei, 2023-02-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
R2S最新固件 eth1网口自动协商100M全双工
### (1) 具体问题
刷入最新固件默认双网卡都是全双工1000m  过了一段时间去看发现lan口掉速了  100m全双工 无法更改
A：

### (2) 路由器型号和固件版本

A：r2s iStoreOS 21.02.3 2023021714

### (3) 详细日志和/或截图

A：

jjm2473, 2023-02-22:
检查下是不是网线或者对端的问题，或者刷回去年底的固件试试。目前还没其他人反馈。

closed 2023-03-13 by jjm2473

heyaaron-Wu, 2023-03-18:
我也有这个问题eth1是100m

jjm2473, 2023-03-18:
看看是不是网线太差，或者进灰了 https://github.com/istoreos/istoreos/issues/647

heyaaron-Wu, 2023-03-18:
我将网线互换也是这样 但是当r2s连我电脑就是1g双工

heyaaron-Wu, 2023-03-18:
可以联系你嘛

heyaaron-Wu, 2023-03-18:
方便联系嘛

zhaokelei, 2023-03-23:
**ethtool -s eth0 speed 1000 duplex full autoneg off**
手动关闭千兆全双工自动协商

开启千兆全双工自动协商
ethtool -s eth0 speed 1000 duplex full autoneg on

然后固件默认的网口互换下 可以测试看看  我的已经解决了  把这个命令设置到开启启动命令里面就可以了，先测试关闭 然后再测试打开  不行就换个好点的网线

heyaaron-Wu, 2023-04-01:
我ethtool -s eth0 speed 1000 duplex full autoneg off使用了这条命令以后无法上网必须重启之后重启完之后又变100m了

jjm2473, 2023-04-04:
一般都是网线质量问题，或者接口脏了。

heyaaron-Wu, 2023-04-19:
没有噢

heyaaron-Wu, 2023-04-19:
lan口没事就wan口出问题了

heyaaron-Wu, 2023-04-19:
两个接口互换也一样 网线也换了一样100m

laorentou666, 2024-10-02:
> 我ethtool -s eth0 speed 1000 duplex full autoneg off使用了这条命令以后无法上网必须重启之后重启完之后又变100m了

我这也是 输完lan口的灯直接灭了 后台进不去要重启才行 结果又变成100M了

heifengwang, 2025-04-19:
同样，25年还有这个问题，wan口偷偷的就掉到100M

Nil-soo, 2025-06-06:
25年也还有

jjm2473, 2025-06-07:
这算啥，我手上的R2S用24.10固件LAN口完全无法识别，一直报USB错误，只有22.03固件才正常

jjm2473, 2025-06-07:
如果还有问题就试试openwrt官方的固件看能不能复现，目前r2s的底层基本跟官方一致，也就加了风扇而已。