istoreos/istoreos #609: container init caused: process_linux.go

l123456789jy, 2023-02-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* ["x ] 没有类似的issue

## 2. 详细叙述
 升级到最新的版本，docker 启动新pull 的容器全部无法启动
### (1) 具体问题
Containers: start chinesesubfinder...code:400 OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init caused: process_linux.go:508: setting cgroup config for procHooks process caused: open /sys/fs/cgroup/docker/1c2c0f136c9faa16a41130218c1bd4ac514caa101506aeb4fdeb0312231b7908/io.weight: no such file or directory: unknown
A：

### (2) 路由器型号和固件版本
iStoreOS 21.02.3 2023021714  x86_64 j4125
A：

### (3) 详细日志和/或截图

A：

l123456789jy, 2023-02-23:
刷回 iStoreOS 21.02.3 2023020316  版本正常

jjm2473, 2023-03-02:
更新0224以后应该就正常了

closed 2023-03-02 by jjm2473