istoreos/istoreos #611: 添加kmod-inet-diag，openclash的process-name规则依赖此前置

LemonPAKA, 2023-02-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* ["x"] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：添加kmod-inet-diag，openclash的process-name规则依赖此前置，openwrt官方21.02.3源中没有kmod-inet-diag，无法直接安装使用

### (2) 路由器型号和固件版本

A：iStoreOS 21.02.3 2023021714   Hewlett-Packard HP t620 PLUS Quad Core TC

### (3) 详细日志和/或截图

A：

jjm2473, 2023-03-03:
更新下iStore，然后看看能不能安装，还有使用是否正常

LemonPAKA, 2023-03-03:
> 更新下iStore，然后看看能不能安装，还有使用是否正常

kmod-inet-diag是openclash软前置，没有这个前置也能安装和运行，但是正如我描述中的一样，无法使用process-name规则的功能，此功能依赖kmod-inet-diag，但是iStoreOS 21.02.3 2023021714中并没有内置此插件，同时openwrt 21.02.3官方源中也没有此插件。（尝试过学习怎么编译但是奈何脑子太笨

jjm2473, 2023-03-03:
下个版本固件会内置这个模块

closed 2023-03-10 by jjm2473