istoreos/istoreos #613: 运营商强制断网之后 会导致运营商那边显示连接了 但实际上无网络

baobeiboboda, 2023-02-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我这边运营商是大连联通，使用的是路由器拨号，运营商这边每96小时强制下线然后重新连接，但是每次强制下线再重新连接都会没有网络，联系运营商获取宽带上线与下线记录，在没有网络的期间运营商一侧显示上线成功 22:11分左右发现没有网了 手动重启了路由器 然后运营商那边显示了22:11 的下网记录

这次出现问题的时间在22年2月23日22:00:43 左右

运营商数据：
上网时间：2023-02-19 22:00:43
下网时间：2023-02-23 22:00:43
时长：96小时0分
出流量17608167K
入流量59183026K
上网时间：2023-02-23 22:00:44
下网时间2023-02-23 22:11:34
时长0小时11分
出流量36K
入流量64K

### (2) 路由器型号和固件版本

A：R86s
版本 [Powered by LuCI istoreos-21.02 branch (git-23.037.57600-91128de)](https://github.com/openwrt/luci) / [ArgonTheme v2.2.9.4](https://github.com/jerrykuku/luci-theme-argon) / iStoreOS 21.02.3 2023021714

### (3) 详细日志和/或截图

A：Thu Feb 23 21:54:55 2023 daemon.err nmbd[8611]: [2023/02/23 21:54:55.817047,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Thu Feb 23 21:54:55 2023 daemon.err nmbd[8611]:   query_name_response: Multiple (6) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Thu Feb 23 21:54:55 2023 daemon.err nmbd[8611]:   This response was from IP 192.168.2.220, reporting an IP address of 192.168.2.220.
Thu Feb 23 21:58:10 2023 daemon.info dnsmasq-dhcp[29179]: DHCPDISCOVER(br-lan) 54:48:e6:5e:d0:ba
Thu Feb 23 21:58:10 2023 daemon.info dnsmasq-dhcp[29179]: DHCPOFFER(br-lan) 192.168.2.102 54:48:e6:5e:d0:ba
Thu Feb 23 21:58:10 2023 daemon.info dnsmasq-dhcp[29179]: DHCPREQUEST(br-lan) 192.168.2.102 54:48:e6:5e:d0:ba
Thu Feb 23 21:58:10 2023 daemon.info dnsmasq-dhcp[29179]: DHCPACK(br-lan) 192.168.2.102 54:48:e6:5e:d0:ba chunmi-oven-steam30lv1_miapD0BA
Thu Feb 23 21:58:21 2023 daemon.warn dnsmasq[29179]: failed to create listening socket for fe80::ce1a:faff:feec:9800%pppoe-wan: Address not available
Thu Feb 23 21:58:21 2023 daemon.warn dnsmasq[29179]: failed to create listening socket for fe80::ce1a:faff:feec:9800%pppoe-wan: Address not available
Thu Feb 23 21:58:21 2023 daemon.warn dnsmasq[29179]: failed to create listening socket for fe80::ce1a:faff:feec:9800%pppoe-wan: Address not available
Thu Feb 23 21:58:21 2023 daemon.warn dnsmasq[29179]: failed to create listening socket for fe80::ce1a:faff:feec:9800%pppoe-wan: Address not available
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]: [2023/02/23 22:00:08.071981,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]:   This response was from IP 192.168.2.200, reporting an IP address of 192.168.2.220.
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]: [2023/02/23 22:00:08.072873,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]:   query_name_response: Multiple (3) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]:   This response was from IP 192.168.2.200, reporting an IP address of 192.168.2.220.
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]: [2023/02/23 22:00:08.072939,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]:   query_name_response: Multiple (4) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]:   This response was from IP 192.168.2.200, reporting an IP address of 192.168.2.220.
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]: [2023/02/23 22:00:08.073296,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]:   query_name_response: Multiple (5) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]:   This response was from IP 192.168.2.220, reporting an IP address of 192.168.2.220.
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]: [2023/02/23 22:00:08.073352,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]:   query_name_response: Multiple (6) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Thu Feb 23 22:00:08 2023 daemon.err nmbd[8611]:   This response was from IP 192.168.2.220, reporting an IP address of 192.168.2.220.
Thu Feb 23 22:00:43 2023 daemon.info pppd[7354]: LCP terminated by peer
Thu Feb 23 22:00:43 2023 daemon.info pppd[7354]: Connect time 5760.1 minutes.
Thu Feb 23 22:00:43 2023 daemon.info pppd[7354]: Sent 4258742684 bytes, received 892033711 bytes.
Thu Feb 23 22:00:43 2023 daemon.warn dnsmasq[29179]: failed to create listening socket for fe80::ce1a:faff:feec:9800%pppoe-wan: Address not available
Thu Feb 23 22:00:43 2023 daemon.warn dnsmasq[29179]: failed to create listening socket for fe80::ce1a:faff:feec:9800%pppoe-wan: Address not available
Thu Feb 23 22:00:43 2023 daemon.err odhcp6c[7447]: Failed to send RELEASE message to ff02::1:2 (Permission denied)
Thu Feb 23 22:00:43 2023 daemon.notice netifd: Network device 'pppoe-wan' link is down
Thu Feb 23 22:00:43 2023 daemon.notice netifd: Network alias 'pppoe-wan' link is down
[... 128 lines cut ...]
Thu Feb 23 22:01:07 2023 daemon.info dnsmasq[18906]: using only locally-known addresses for domain test
Thu Feb 23 22:01:07 2023 daemon.info dnsmasq[18906]: using only locally-known addresses for domain onion
Thu Feb 23 22:01:07 2023 daemon.info dnsmasq[18906]: using only locally-known addresses for domain localhost
Thu Feb 23 22:01:07 2023 daemon.info dnsmasq[18906]: using only locally-known addresses for domain local
Thu Feb 23 22:01:07 2023 daemon.info dnsmasq[18906]: using only locally-known addresses for domain invalid
Thu Feb 23 22:01:07 2023 daemon.info dnsmasq[18906]: using only locally-known addresses for domain bind
Thu Feb 23 22:01:07 2023 daemon.info dnsmasq[18906]: using nameserver 127.0.0.1#7874
Thu Feb 23 22:01:07 2023 daemon.info dnsmasq[18906]: using only locally-known addresses for domain lan
Thu Feb 23 22:01:07 2023 daemon.info dnsmasq[18906]: read /etc/hosts - 4 addresses
Thu Feb 23 22:01:07 2023 daemon.info dnsmasq[18906]: read /tmp/hosts/odhcpd - 3 addresses
Thu Feb 23 22:01:07 2023 daemon.info dnsmasq[18906]: read /tmp/hosts/dhcp.cfg01411c - 10 addresses
Thu Feb 23 22:01:07 2023 daemon.info dnsmasq-dhcp[18906]: read /etc/ethers - 0 addresses
Thu Feb 23 22:01:10 2023 daemon.notice ttyd[12653]:    vhost default: no cert
Thu Feb 23 22:01:35 2023 daemon.info dnsmasq[18906]: read /etc/hosts - 4 addresses
Thu Feb 23 22:01:35 2023 daemon.info dnsmasq[18906]: read /tmp/hosts/odhcpd - 2 addresses
Thu Feb 23 22:01:35 2023 daemon.info dnsmasq[18906]: read /tmp/hosts/dhcp.cfg01411c - 10 addresses
Thu Feb 23 22:01:35 2023 daemon.info dnsmasq-dhcp[18906]: read /etc/ethers - 0 addresses
Thu Feb 23 22:01:36 2023 daemon.info dnsmasq[18906]: read /etc/hosts - 4 addresses
Thu Feb 23 22:01:36 2023 daemon.info dnsmasq[18906]: read /tmp/hosts/odhcpd - 3 addresses
Thu Feb 23 22:01:36 2023 daemon.info dnsmasq[18906]: read /tmp/hosts/dhcp.cfg01411c - 10 addresses
Thu Feb 23 22:01:36 2023 daemon.info dnsmasq-dhcp[18906]: read /etc/ethers - 0 addresses
Thu Feb 23 22:01:56 2023 user.err ddns-scripts[16466]: ddns: GNU Wget Error: '4'
Thu Feb 23 22:01:56 2023 user.warn ddns-scripts[16466]: ddns: Transfer failed - retry 2/0 in 60 seconds
Thu Feb 23 22:02:56 2023 user.err ddns-scripts[16466]: ddns: GNU Wget Error: '4'
Thu Feb 23 22:02:56 2023 user.warn ddns-scripts[16466]: ddns: Transfer failed - retry 3/0 in 60 seconds

baobeiboboda, 2023-03-03:
重新刷到这个版本 第一个96小时 重连正常
[Powered by LuCI openwrt-21.02-cn branch (git-22.186.32466-892d5d9)](https://github.com/openwrt/luci)
/ [ArgonTheme v2.2.9.4](https://github.com/jerrykuku/luci-theme-argon) / OpenWrt 21.02.2 2022071510

baobeiboboda, 2023-03-06:
第二个96小时 重连异常 

日志如下

Mon Mar  6 21:26:41 2023 daemon.info dnsmasq-dhcp[11080]: DHCP, IP range 192.168.2.100 -- 192.168.2.249, lease time 12h
Mon Mar  6 21:26:41 2023 daemon.info dnsmasq[11080]: using only locally-known addresses for domain test
Mon Mar  6 21:26:41 2023 daemon.info dnsmasq[11080]: using only locally-known addresses for domain onion
Mon Mar  6 21:26:41 2023 daemon.info dnsmasq[11080]: using only locally-known addresses for domain localhost
Mon Mar  6 21:26:41 2023 daemon.info dnsmasq[11080]: using only locally-known addresses for domain local
Mon Mar  6 21:26:41 2023 daemon.info dnsmasq[11080]: using only locally-known addresses for domain invalid
Mon Mar  6 21:26:41 2023 daemon.info dnsmasq[11080]: using only locally-known addresses for domain bind
Mon Mar  6 21:26:41 2023 daemon.info dnsmasq[11080]: using nameserver 127.0.0.1#7874
Mon Mar  6 21:26:41 2023 daemon.info dnsmasq[11080]: using only locally-known addresses for domain lan
Mon Mar  6 21:26:41 2023 daemon.info dnsmasq[11080]: read /etc/hosts - 4 addresses
Mon Mar  6 21:26:41 2023 daemon.info dnsmasq[11080]: read /tmp/hosts/odhcpd - 4 addresses
Mon Mar  6 21:26:41 2023 daemon.info dnsmasq[11080]: read /tmp/hosts/dhcp.cfg01411c - 10 addresses
Mon Mar  6 21:26:41 2023 daemon.info dnsmasq-dhcp[11080]: read /etc/ethers - 0 addresses
Mon Mar  6 21:26:43 2023 daemon.notice ttyd[9973]:    vhost default: no cert
Mon Mar  6 21:27:30 2023 daemon.info dnsmasq[11080]: read /etc/hosts - 4 addresses
Mon Mar  6 21:27:30 2023 daemon.info dnsmasq[11080]: read /tmp/hosts/odhcpd - 3 addresses
Mon Mar  6 21:27:30 2023 daemon.info dnsmasq[11080]: read /tmp/hosts/dhcp.cfg01411c - 10 addresses
Mon Mar  6 21:27:30 2023 daemon.info dnsmasq-dhcp[11080]: read /etc/ethers - 0 addresses
Mon Mar  6 21:27:31 2023 daemon.info dnsmasq[11080]: read /etc/hosts - 4 addresses
Mon Mar  6 21:27:31 2023 daemon.info dnsmasq[11080]: read /tmp/hosts/odhcpd - 4 addresses
Mon Mar  6 21:27:31 2023 daemon.info dnsmasq[11080]: read /tmp/hosts/dhcp.cfg01411c - 10 addresses
Mon Mar  6 21:27:31 2023 daemon.info dnsmasq-dhcp[11080]: read /etc/ethers - 0 addresses
Mon Mar  6 21:28:15 2023 daemon.info dnsmasq[11080]: read /etc/hosts - 4 addresses
Mon Mar  6 21:28:15 2023 daemon.info dnsmasq[11080]: read /tmp/hosts/odhcpd - 3 addresses
Mon Mar  6 21:28:15 2023 daemon.info dnsmasq[11080]: read /tmp/hosts/dhcp.cfg01411c - 10 addresses
Mon Mar  6 21:28:15 2023 daemon.info dnsmasq-dhcp[11080]: read /etc/ethers - 0 addresses
Mon Mar  6 21:28:16 2023 daemon.info dnsmasq[11080]: read /etc/hosts - 4 addresses
Mon Mar  6 21:28:16 2023 daemon.info dnsmasq[11080]: read /tmp/hosts/odhcpd - 4 addresses
Mon Mar  6 21:28:16 2023 daemon.info dnsmasq[11080]: read /tmp/hosts/dhcp.cfg01411c - 10 addresses
Mon Mar  6 21:28:16 2023 daemon.info dnsmasq-dhcp[11080]: read /etc/ethers - 0 addresses
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]: [2023/03/06 21:28:37.254214,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]:   This response was from IP 192.168.2.200, reporting an IP address of 192.168.2.220.
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]: [2023/03/06 21:28:37.254350,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]:   query_name_response: Multiple (3) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]:   This response was from IP 192.168.2.200, reporting an IP address of 192.168.2.220.
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]: [2023/03/06 21:28:37.254409,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]:   query_name_response: Multiple (4) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]:   This response was from IP 192.168.2.200, reporting an IP address of 192.168.2.220.
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]: [2023/03/06 21:28:37.255044,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]:   query_name_response: Multiple (5) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]:   This response was from IP 192.168.2.220, reporting an IP address of 192.168.2.220.
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]: [2023/03/06 21:28:37.255103,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]:   query_name_response: Multiple (6) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Mon Mar  6 21:28:37 2023 daemon.err nmbd[8960]:   This response was from IP 192.168.2.220, reporting an IP address of 192.168.2.220.
Mon Mar  6 21:28:46 2023 user.info : luci: accepted login on / for root from 192.168.2.2
Mon Mar  6 21:29:37 2023 user.notice wan_drop: current proto pppoe
Mon Mar  6 21:29:37 2023 daemon.notice netifd: Interface 'CTE' is enabled
Mon Mar  6 21:29:37 2023 daemon.notice netifd: Interface 'CTE' has link connectivity
Mon Mar  6 21:29:37 2023 daemon.notice netifd: Interface 'CTE' is setting up now
[... cut ...]
Mon Mar  6 21:33:18 2023 daemon.err nmbd[15269]:   query_name_response: Multiple (6) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Mon Mar  6 21:33:18 2023 daemon.err nmbd[15269]:   This response was from IP 192.168.2.220, reporting an IP address of 192.168.2.220.
Mon Mar  6 21:33:18 2023 daemon.err smbd[15271]: [2023/03/06 21:33:18.170957,  0] ../../lib/util/become_daemon.c:136(daemon_ready)
Mon Mar  6 21:33:18 2023 daemon.err smbd[15271]:   daemon_ready: daemon 'smbd' finished starting up and ready to serve connections

baobeiboboda, 2023-03-14:
重连异常之后 又过了第一个4天 一切正常 第二个四天 又不正常了
日志如下

baobeiboboda, 2023-03-14:
Tue Mar 14 21:15:34 2023 daemon.err nmbd[14693]:   This response was from IP 192.168.2.220, reporting an IP address of 192.168.2.220.
Tue Mar 14 21:15:34 2023 daemon.err nmbd[14693]: [2023/03/14 21:15:34.175706,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Tue Mar 14 21:15:34 2023 daemon.err nmbd[14693]:   query_name_response: Multiple (6) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Tue Mar 14 21:15:34 2023 daemon.err nmbd[14693]:   This response was from IP 192.168.2.220, reporting an IP address of 192.168.2.220.
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]: [2023/03/14 21:20:34.247984,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]:   This response was from IP 192.168.2.200, reporting an IP address of 192.168.2.220.
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]: [2023/03/14 21:20:34.248109,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]:   query_name_response: Multiple (3) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]:   This response was from IP 192.168.2.220, reporting an IP address of 192.168.2.220.
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]: [2023/03/14 21:20:34.248167,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]:   query_name_response: Multiple (4) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]:   This response was from IP 192.168.2.220, reporting an IP address of 192.168.2.220.
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]: [2023/03/14 21:20:34.248360,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]:   query_name_response: Multiple (5) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]:   This response was from IP 192.168.2.220, reporting an IP address of 192.168.2.220.
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]: [2023/03/14 21:20:34.248415,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]:   query_name_response: Multiple (6) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Tue Mar 14 21:20:34 2023 daemon.err nmbd[14693]:   This response was from IP 192.168.2.220, reporting an IP address of 192.168.2.220.
Tue Mar 14 21:21:02 2023 daemon.info dnsmasq-dhcp[17878]: DHCPREQUEST(br-lan) 192.168.2.135 54:ef:44:36:6b:e9
Tue Mar 14 21:21:02 2023 daemon.info dnsmasq-dhcp[17878]: DHCPACK(br-lan) 192.168.2.135 54:ef:44:36:6b:e9 AC-Controller-6BE9
Tue Mar 14 21:21:19 2023 daemon.warn dnsmasq[17878]: failed to create listening socket for fe80::ce1a:faff:feec:9800%pppoe-wan: Address not available
Tue Mar 14 21:21:19 2023 daemon.warn dnsmasq[17878]: failed to create listening socket for fe80::ce1a:faff:feec:9800%pppoe-wan: Address not available
Tue Mar 14 21:21:19 2023 daemon.warn dnsmasq[17878]: failed to create listening socket for fe80::ce1a:faff:feec:9800%pppoe-wan: Address not available
Tue Mar 14 21:21:19 2023 daemon.warn dnsmasq[17878]: failed to create listening socket for fe80::ce1a:faff:feec:9800%pppoe-wan: Address not available
Tue Mar 14 21:21:25 2023 daemon.warn dnsmasq[17878]: failed to create listening socket for fe80::ce1a:faff:feec:9800%pppoe-wan: Address not available
Tue Mar 14 21:21:25 2023 daemon.warn dnsmasq[17878]: failed to create listening socket for fe80::ce1a:faff:feec:9800%pppoe-wan: Address not available
Tue Mar 14 21:23:19 2023 daemon.warn dnsmasq[17878]: possible DNS-rebind attack detected: tcls.igame.qq.com
Tue Mar 14 21:23:29 2023 daemon.warn dnsmasq[17878]: possible DNS-rebind attack detected: tqm.gamebaby.qq.com
Tue Mar 14 21:24:38 2023 daemon.warn dnsmasq[17878]: possible DNS-rebind attack detected: tcls.igame.qq.com
Tue Mar 14 21:24:44 2023 daemon.warn dnsmasq[17878]: possible DNS-rebind attack detected: tqm.gamebaby.qq.com
Tue Mar 14 21:25:34 2023 daemon.err nmbd[14693]: [2023/03/14 21:25:34.301246,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Tue Mar 14 21:25:34 2023 daemon.err nmbd[14693]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Tue Mar 14 21:25:34 2023 daemon.err nmbd[14693]:   This response was from IP 192.168.2.200, reporting an IP address of 192.168.2.220.
Tue Mar 14 21:25:34 2023 daemon.err nmbd[14693]: [2023/03/14 21:25:34.301369,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Tue Mar 14 21:25:34 2023 daemon.err nmbd[14693]:   query_name_response: Multiple (3) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Tue Mar 14 21:25:34 2023 daemon.err nmbd[14693]:   This response was from IP 192.168.2.220, reporting an IP address of 192.168.2.220.
Tue Mar 14 21:25:34 2023 daemon.err nmbd[14693]: [2023/03/14 21:25:34.301428,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
[... cut ...]
Tue Mar 14 21:52:10 2023 daemon.info dnsmasq[7337]: read /tmp/hosts/dhcp.cfg01411c - 10 addresses
Tue Mar 14 21:52:10 2023 daemon.info dnsmasq-dhcp[7337]: read /etc/ethers - 0 addresses
Tue Mar 14 21:52:11 2023 daemon.info dnsmasq[7337]: read /etc/hosts - 4 addresses
Tue Mar 14 21:52:11 2023 daemon.info dnsmasq[7337]: read /tmp/hosts/odhcpd - 4 addresses
Tue Mar 14 21:52:11 2023 daemon.info dnsmasq[7337]: read /tmp/hosts/dhcp.cfg01411c - 10 addresses
Tue Mar 14 21:52:11 2023 daemon.info dnsmasq-dhcp[7337]: read /etc/ethers - 0 addresses

jjm2473, 2023-04-11:
utun是什么？是不是翻墙了？

baobeiboboda, 2023-09-11:
> utun是什么？是不是翻墙了？

是的 openclash

jjm2473, 2023-09-11:
如果是openclash，那出问题很正常，这个插件bug很多。还是卸载掉，用其他插件试试，例如passwall