istoreos/istoreos #616: 当尝试向日葵连接某台电脑时，会出现使openwrt主路由直接崩溃断连的情况

stardust1117, 2023-02-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：光猫桥接，istoreos作为主路由pppoe拨号上网，当尝试向日葵连接某台主机时（大部分电脑正常）。会出现使openwrt主路由直接崩溃断连的情况，断连后路由器必须断电重启才能恢复，经过路由器重置也不行，经过大量尝试发现关闭了默认开启的开关Fullcone-NAT开关后就好了。是否能解决在Fullcone-NAT网络环境下的连接问题？

### (2) 路由器型号和固件版本

A：EasePi, istoreos, OpenWrt 21.02.3 r16845-3f075851ca

### (3) 详细日志和/或截图

A：
Sat Feb 25 08:51:00 2023 daemon.err nmbd[6281]:   Got SIGTERM: going down...
Sat Feb 25 08:51:00 2023 daemon.err smbd[6280]: [2023/02/25 08:51:00.302030,  0] ../../lib/util/become_daemon.c:136(daemon_ready)
Sat Feb 25 08:51:00 2023 daemon.err smbd[6280]:   daemon_ready: daemon 'smbd' finished starting up and ready to serve connections
Sat Feb 25 08:51:00 2023 daemon.err nmbd[7445]: [2023/02/25 08:51:00.389885,  0] ../../lib/util/become_daemon.c:136(daemon_ready)
Sat Feb 25 08:51:00 2023 daemon.err nmbd[7445]:   daemon_ready: daemon 'nmbd' finished starting up and ready to serve connections
Sat Feb 25 08:51:00 2023 daemon.err smbd[7459]: [2023/02/25 08:51:00.468426,  0] ../../lib/util/become_daemon.c:136(daemon_ready)
Sat Feb 25 08:51:00 2023 daemon.err smbd[7459]:   daemon_ready: daemon 'smbd' finished starting up and ready to serve connections
Sat Feb 25 08:51:01 2023 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sat Feb 25 08:51:01 2023 daemon.info avahi-daemon[5301]: Leaving mDNS multicast group on interface pppoe-wan.IPv6 with address fe80::900b:d5d2:a810:1df2.
Sat Feb 25 08:51:01 2023 daemon.info avahi-daemon[5301]: Joining mDNS multicast group on interface pppoe-wan.IPv6 with address 240e:390:992:8c4:900b:d5d2:a810:1df2.
Sat Feb 25 08:51:01 2023 daemon.info avahi-daemon[5301]: Registering new address record for 240e:390:992:8c4:900b:d5d2:a810:1df2 on pppoe-wan.*.
Sat Feb 25 08:51:01 2023 daemon.info avahi-daemon[5301]: Withdrawing address record for fe80::900b:d5d2:a810:1df2 on pppoe-wan.
Sat Feb 25 08:51:01 2023 daemon.notice netifd: Interface 'wan_6' is now up
Sat Feb 25 08:51:01 2023 daemon.info dnsmasq[7209]: reading /tmp/resolv.conf.d/resolv.conf.auto
Sat Feb 25 08:51:01 2023 daemon.info dnsmasq[7209]: using only locally-known addresses for domain test
Sat Feb 25 08:51:01 2023 daemon.info dnsmasq[7209]: using only locally-known addresses for domain onion
Sat Feb 25 08:51:01 2023 daemon.info dnsmasq[7209]: using only locally-known addresses for domain localhost
Sat Feb 25 08:51:01 2023 daemon.info dnsmasq[7209]: using only locally-known addresses for domain local
Sat Feb 25 08:51:01 2023 daemon.info dnsmasq[7209]: using only locally-known addresses for domain invalid
Sat Feb 25 08:51:01 2023 daemon.info dnsmasq[7209]: using only locally-known addresses for domain bind
Sat Feb 25 08:51:01 2023 daemon.info dnsmasq[7209]: using only locally-known addresses for domain lan
Sat Feb 25 08:51:01 2023 daemon.info dnsmasq[7209]: using nameserver 202.101.172.35#53
Sat Feb 25 08:51:01 2023 daemon.info dnsmasq[7209]: using nameserver 202.101.172.46#53
Sat Feb 25 08:51:01 2023 daemon.info dnsmasq[7209]: using nameserver 240e:1c:200::1#53
Sat Feb 25 08:51:01 2023 daemon.info dnsmasq[7209]: using nameserver 240e:1c:200::2#53
Sat Feb 25 08:51:01 2023 user.notice miniupnpd: restart miniupnpd due to pppoe-wan up
Sat Feb 25 08:51:01 2023 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Sat Feb 25 08:51:02 2023 user.notice ddns-scripts[8093]: myddns_ipv4: PID '8093' started at 2023-02-25 08:51
Sat Feb 25 08:51:02 2023 user.warn ddns-scripts[8093]: myddns_ipv4: Service section disabled!
Sat Feb 25 08:51:02 2023 user.warn ddns-scripts[8093]: myddns_ipv4: Stopped - TERMINATE
Sat Feb 25 08:51:02 2023 user.warn ddns-scripts[8093]: myddns_ipv4: PID '8093' exit WITH ERROR '1' at 2023-02-25 08:51
Sat Feb 25 08:51:02 2023 user.notice miniupnpd: restart miniupnpd due to pppoe-wan up
Sat Feb 25 08:51:02 2023 daemon.warn odhcpd[4018]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Sat Feb 25 08:51:02 2023 daemon.info avahi-daemon[5301]: Registering new address record for 240e:390:989:7810::1 on br-lan.*.
Sat Feb 25 08:51:02 2023 daemon.info dnsmasq[7209]: read /etc/hosts - 4 addresses
Sat Feb 25 08:51:02 2023 daemon.info dnsmasq[7209]: read /tmp/hosts/odhcpd - 0 addresses
Sat Feb 25 08:51:02 2023 daemon.info dnsmasq[7209]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sat Feb 25 08:51:02 2023 daemon.info dnsmasq-dhcp[7209]: read /etc/ethers - 0 addresses
Sat Feb 25 08:51:03 2023 daemon.err dockerd[6620]: time="2023-02-25T00:51:03.930445102Z" level=warning msg="grpc: addrConn.createTransport failed to connect to {unix:///var/run/docker/containerd/containerd.sock  <nil> 0 <nil>}. Err :connection error: desc = \"transport: error while dialing: dial unix:///var/run/docker/containerd/containerd.sock: timeout\". Reconnecting..." module=grpc
Sat Feb 25 08:51:05 2023 daemon.err modprobe: failed to find a module named aufs
Sat Feb 25 08:51:05 2023 daemon.err dockerd[6620]: time="2023-02-25T00:51:05.678739213Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Sat Feb 25 08:51:05 2023 daemon.err dockerd[6620]: time="2023-02-25T00:51:05.680521636Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Sat Feb 25 08:51:05 2023 daemon.err dockerd[6620]: time="2023-02-25T00:51:05.787660489Z" level=warning msg="Your kernel does not support cgroup blkio weight"
Sat Feb 25 08:51:05 2023 daemon.err dockerd[6620]: time="2023-02-25T00:51:05.787746875Z" level=warning msg="Your kernel does not support cgroup blkio weight_device"
Sat Feb 25 08:51:05 2023 daemon.info procd: - init complete -
Sat Feb 25 08:51:05 2023 daemon.err modprobe: failed to find a module named nf_conntrack
Sat Feb 25 08:51:06 2023 daemon.info avahi-daemon[5301]: Joining mDNS multicast group on interface docker0.IPv4 with address 172.17.0.1.
Sat Feb 25 08:51:06 2023 daemon.info avahi-daemon[5301]: New relevant interface docker0.IPv4 for mDNS.
[... 42 lines cut ...]
Sat Feb 25 08:55:24 2023 daemon.info dnsmasq-dhcp[7209]: DHCPREQUEST(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 08:55:24 2023 daemon.info dnsmasq-dhcp[7209]: DHCPACK(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 08:55:24 2023 daemon.warn dnsmasq[7209]: possible DNS-rebind attack detected: disp3.aiseewhaley.aisee.tv
Sat Feb 25 08:56:25 2023 daemon.info dnsmasq-dhcp[7209]: DHCPDISCOVER(br-lan) 04:5d:4b:4d:09:d5
Sat Feb 25 08:56:25 2023 daemon.info dnsmasq-dhcp[7209]: DHCPOFFER(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 08:56:25 2023 daemon.info dnsmasq-dhcp[7209]: DHCPREQUEST(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 08:56:25 2023 daemon.info dnsmasq-dhcp[7209]: DHCPACK(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 08:57:25 2023 daemon.info dnsmasq-dhcp[7209]: DHCPDISCOVER(br-lan) 04:5d:4b:4d:09:d5
Sat Feb 25 08:57:25 2023 daemon.info dnsmasq-dhcp[7209]: DHCPOFFER(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 08:57:25 2023 daemon.info dnsmasq-dhcp[7209]: DHCPREQUEST(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 08:57:25 2023 daemon.info dnsmasq-dhcp[7209]: DHCPACK(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 08:58:26 2023 daemon.info dnsmasq-dhcp[7209]: DHCPDISCOVER(br-lan) 04:5d:4b:4d:09:d5
Sat Feb 25 08:58:26 2023 daemon.info dnsmasq-dhcp[7209]: DHCPOFFER(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 08:58:26 2023 daemon.info dnsmasq-dhcp[7209]: DHCPREQUEST(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 08:58:26 2023 daemon.info dnsmasq-dhcp[7209]: DHCPACK(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 08:58:55 2023 daemon.err uhttpd[4838]: find: /tmp/appfilter/: No such file or directory
Sat Feb 25 08:59:24 2023 daemon.info dnsmasq-dhcp[7209]: DHCPDISCOVER(br-lan) 04:5d:4b:4d:09:d5
Sat Feb 25 08:59:24 2023 daemon.info dnsmasq-dhcp[7209]: DHCPOFFER(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 08:59:24 2023 daemon.info dnsmasq-dhcp[7209]: DHCPREQUEST(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 08:59:24 2023 daemon.info dnsmasq-dhcp[7209]: DHCPACK(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:00:17 2023 daemon.info dnsmasq-dhcp[7209]: DHCPDISCOVER(br-lan) 04:5d:4b:4d:09:d5
Sat Feb 25 09:00:17 2023 daemon.info dnsmasq-dhcp[7209]: DHCPOFFER(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:00:17 2023 daemon.info dnsmasq-dhcp[7209]: DHCPREQUEST(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:00:17 2023 daemon.info dnsmasq-dhcp[7209]: DHCPACK(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:01:03 2023 daemon.info dnsmasq-dhcp[7209]: DHCPDISCOVER(br-lan) 04:5d:4b:4d:09:d5

stardust1117, 2023-02-25:
Sat Feb 25 09:01:03 2023 daemon.info dnsmasq-dhcp[7209]: DHCPOFFER(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:01:03 2023 daemon.info dnsmasq-dhcp[7209]: DHCPREQUEST(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:01:03 2023 daemon.info dnsmasq-dhcp[7209]: DHCPACK(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:01:40 2023 daemon.info dnsmasq-dhcp[7209]: DHCPDISCOVER(br-lan) 04:5d:4b:4d:09:d5
Sat Feb 25 09:01:40 2023 daemon.info dnsmasq-dhcp[7209]: DHCPOFFER(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:01:40 2023 daemon.info dnsmasq-dhcp[7209]: DHCPREQUEST(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:01:40 2023 daemon.info dnsmasq-dhcp[7209]: DHCPACK(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:02:25 2023 daemon.info dnsmasq-dhcp[7209]: DHCPDISCOVER(br-lan) 04:5d:4b:4d:09:d5
Sat Feb 25 09:02:25 2023 daemon.info dnsmasq-dhcp[7209]: DHCPOFFER(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:02:25 2023 daemon.info dnsmasq-dhcp[7209]: DHCPREQUEST(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:02:25 2023 daemon.info dnsmasq-dhcp[7209]: DHCPACK(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:03:26 2023 daemon.info dnsmasq-dhcp[7209]: DHCPDISCOVER(br-lan) 04:5d:4b:4d:09:d5
Sat Feb 25 09:03:26 2023 daemon.info dnsmasq-dhcp[7209]: DHCPOFFER(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:03:26 2023 daemon.info dnsmasq-dhcp[7209]: DHCPREQUEST(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5
Sat Feb 25 09:03:26 2023 daemon.info dnsmasq-dhcp[7209]: DHCPACK(br-lan) 192.168.2.226 04:5d:4b:4d:09:d5

stardust1117, 2023-02-25:
Sat Feb 25 08:50:58 2023 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Sat Feb 25 08:50:58 2023 daemon.notice netifd: Interface 'wan_6' is enabled
Sat Feb 25 08:50:58 2023 daemon.notice netifd: Interface 'wan_6' has link connectivity
Sat Feb 25 08:50:58 2023 daemon.notice netifd: Interface 'wan_6' is setting up now
Sat Feb 25 08:50:58 2023 daemon.notice procd: /etc/rc.d/S99dockerd: iptables: No chain/target/match by that name.
Sat Feb 25 08:50:59 2023 daemon.notice procd: /etc/rc.d/S99nfsd: point 1
Sat Feb 25 08:50:59 2023 daemon.notice procd: /etc/rc.d/S99nfsd: point 2
Sat Feb 25 08:50:59 2023 daemon.notice procd: /etc/rc.d/S99nfsd: point 3
Sat Feb 25 08:50:59 2023 kern.warn kernel: [   25.796623] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
Sat Feb 25 08:50:59 2023 kern.info kernel: [   25.804807] NFSD: starting 10-second grace period (net ffffff8008fe5700)
Sat Feb 25 08:50:59 2023 daemon.notice rpc.mountd[6791]: Version 2.5.2 starting
Sat Feb 25 08:50:59 2023 daemon.notice rpc.statd[6790]: Version 2.5.2 starting
Sat Feb 25 08:50:59 2023 daemon.warn rpc.statd[6790]: Flags: No-Daemon TI-RPC
Sat Feb 25 08:50:59 2023 daemon.notice rpc.statd[6790]: Initializing NSM state
Sat Feb 25 08:50:59 2023 daemon.warn rpc.statd[6790]: Running as root.  chown /var/lib/nfs to choose different user
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[3001]: exiting on receipt of SIGTERM
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: Connected to system UBus
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: started, version 2.85 cachesize 150
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: DNS service limited to local subnets
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: UBus support enabled: connected to system bus
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq-dhcp[6908]: DHCP, IP range 192.168.2.2 -- 192.168.2.248, lease time 12h
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using only locally-known addresses for domain test
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using only locally-known addresses for domain onion
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using only locally-known addresses for domain localhost
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using only locally-known addresses for domain local
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using only locally-known addresses for domain invalid
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using only locally-known addresses for domain bind
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using only locally-known addresses for domain lan
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: reading /tmp/resolv.conf.d/resolv.conf.auto
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using only locally-known addresses for domain test
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using only locally-known addresses for domain onion
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using only locally-known addresses for domain localhost
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using only locally-known addresses for domain local
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using only locally-known addresses for domain invalid
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using only locally-known addresses for domain bind
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using only locally-known addresses for domain lan
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using nameserver 202.101.172.35#53
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: using nameserver 202.101.172.46#53
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: read /etc/hosts - 4 addresses
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq-dhcp[6908]: read /etc/ethers - 0 addresses
Sat Feb 25 08:50:59 2023 daemon.notice ttyd[6913]: ttyd 1.6.3 (libwebsockets 3.1.0)
Sat Feb 25 08:50:59 2023 daemon.notice ttyd[6913]: tty configuration:
Sat Feb 25 08:50:59 2023 daemon.notice ttyd[6913]:   start command: /bin/login
Sat Feb 25 08:50:59 2023 daemon.notice ttyd[6913]:   close signal: SIGHUP (1)
Sat Feb 25 08:50:59 2023 daemon.notice ttyd[6913]:   terminal type: xterm-256color
Sat Feb 25 08:50:59 2023 daemon.notice ttyd[6913]:  Using foreign event loop...
Sat Feb 25 08:50:59 2023 daemon.notice ttyd[6913]: Creating Vhost 'default' port 7681, 2 protocols, IPv6 off
Sat Feb 25 08:50:59 2023 daemon.notice ttyd[6913]:  Listening on port: 7681
Sat Feb 25 08:50:59 2023 daemon.err nmbd[6281]: [2023/02/25 08:50:59.797504,  0] ../../lib/util/become_daemon.c:136(daemon_ready)
Sat Feb 25 08:50:59 2023 daemon.err nmbd[6281]:   daemon_ready: daemon 'nmbd' finished starting up and ready to serve connections
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[6908]: exiting on receipt of SIGTERM
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[7209]: Connected to system UBus
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[7209]: started, version 2.85 cachesize 150
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[7209]: DNS service limited to local subnets
[... cut ...]
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[7209]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq-dhcp[7209]: read /etc/ethers - 0 addresses
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[7209]: read /etc/hosts - 4 addresses
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq[7209]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sat Feb 25 08:50:59 2023 daemon.info dnsmasq-dhcp[7209]: read /etc/ethers - 0 addresses
Sat Feb 25 08:51:00 2023 daemon.err nmbd[6281]: [2023/02/25 08:51:00.274206,  0] ../../source3/nmbd/nmbd.c:60(terminate)

jjm2473, 2023-03-02:
崩溃没有自动重启？

stardust1117, 2023-03-02:
没有，必须断电重新插电才能恢复
 

星际尘埃
***@***.***

stardust1117, 2023-03-02:
向日葵工程师的回复
感谢您的耐心等待。工程师判断是您公司内网拦截访问向日葵，1.建议您咨询网络管理员是否有网络安全策略拦截了向日葵；
2.若单位服务器有防火墙或屏蔽了很多服务器，建议联系网络管理员开放支持我司服务器和端口号：
服务器：*.oray.net；远控常用端口：tcp80443 4009 8000；udp:4118；（远控）3000；剩下的都是动态的，随机的5位数远程端口，所有5位数端口需全开放

因我们服务器较多无法全部给您提供，具体按您向日葵日志所连接的服务器进行添加白名单，域名解析后即可得出IP
日志查看：请在名称带server的日志里查看所连接的服务器，域名+443端口的均为服务器。向日葵日志获取的方式：退出向日葵，右键桌面向日葵图标打开文件夹所在位置，查看log文件夹。
3. 建议关闭Fullcone-NAT使用

stardust1117, 2023-03-02:
我在恩山看到有一个跟我一样情况的，他是关了Turbo ACC好的。但我的istoreos默认应该没有turbo acc吧
https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=4437907&highlight=%CF%F2%C8%D5%BF%FB
“已经关闭了1天SFE了，好像自动断线(必须重启恢复)的问题也没有发生了，我继续观察一下”

jjm2473, 2023-03-02:
没有turboacc，也没有sfe。

如果你的路由器有接显示器，断线以后看看显示器有没有输出什么日志

jjm2473, 2023-03-18:
按下面的文档配置日志持久化，出现问题以后等个几分钟，然后按一下重启键看看能不能重启，不能重启就断电重启。重启以后提取 /root/system.log 和 /root/system.log.old 文件
https://doc.linkease.com/zh/guide/easepi/question.html#q-%E5%A6%82%E4%BD%95%E8%BF%9B%E8%A1%8C%E6%95%85%E9%9A%9C%E6%8E%92%E6%9F%A5

stardust1117, 2023-03-21:
好的我有时间复现操作一下

btdwv, 2023-05-19:
> 按下面的文档配置日志持久化，出现问题以后等个几分钟，然后按一下重启键看看能不能重启，不能重启就断电重启。重启以后提取 /root/system.log 和 /root/system.log.old 文件 https://doc.linkease.com/zh/guide/easepi/question.html#q-%E5%A6%82%E4%BD%95%E8%BF%9B%E8%A1%8C%E6%95%85%E9%9A%9C%E6%8E%92%E6%9F%A5

我也遇到同样的问题，在自己的ARS2上尝试了一下
21:05:50后过了几秒连接向日葵，立刻就断网了，大约21:07:33时重启软路由，下面是system.log的日志

Fri May 19 21:05:27 2023 daemon.info dnsmasq[12718]: read /etc/hosts - 4 addresses
Fri May 19 21:05:27 2023 daemon.info dnsmasq[12718]: read /tmp/hosts/dhcp.cfg01411c - 6 addresses
Fri May 19 21:05:27 2023 daemon.info dnsmasq-dhcp[12718]: read /etc/ethers - 0 addresses
Fri May 19 21:05:39 2023 daemon.info dnsmasq[12718]: read /etc/hosts - 4 addresses
Fri May 19 21:05:39 2023 daemon.info dnsmasq[12718]: read /tmp/hosts/dhcp.cfg01411c - 6 addresses
Fri May 19 21:05:39 2023 daemon.info dnsmasq-dhcp[12718]: read /etc/ethers - 0 addresses
Fri May 19 21:06:38 2023 daemon.err nmbd[7462]: [2023/05/19 21:06:38.149191,  0] ../../source3/nmbd/nmbd_become_lmb.c:398(become_local_master_stage2)
Fri May 19 21:06:38 2023 daemon.err nmbd[7462]:   *****
Fri May 19 21:06:38 2023 daemon.err nmbd[7462]:
Fri May 19 21:06:38 2023 daemon.err nmbd[7462]:   Samba name server ARS2 is now a local master browser for workgroup WORKGROUP on subnet 192.168.50.1
Fri May 19 21:06:38 2023 daemon.err nmbd[7462]:
Fri May 19 21:06:38 2023 daemon.err nmbd[7462]:   *****
Fri May 19 21:07:33 2023 daemon.info procd: - shutdown -
Fri May 19 21:07:33 2023 user.notice DEBUG: System Shutting Down
Fri May 19 21:07:33 2023 daemon.notice procd: /etc/rc.d/K00debug: DEBUG: System Shutting Down
Fri May 19 21:07:33 2023 user.notice DEBUG: System Shutting Down
Fri May 19 21:07:34 2023 daemon.info dnsmasq[12718]: exiting on receipt of SIGTERM
Fri May 19 21:07:34 2023 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri May 19 21:07:34 2023 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri May 19 21:07:34 2023 daemon.notice procd: /etc/rc.d/K01AdGuardHome: udhcpc: started, v1.33.2
Fri May 19 21:07:34 2023 daemon.notice procd: /etc/rc.d/K01AdGuardHome: udhcpc: sending discover
Fri May 19 21:07:37 2023 daemon.notice procd: /etc/rc.d/K01AdGuardHome: udhcpc: no lease, failing
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: Connected to system UBus
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: started, version 2.85 cache disabled
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: DNS service limited to local subnets
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: UBus support enabled: connected to system bus
Fri May 19 21:07:37 2023 daemon.info dnsmasq-dhcp[17178]: DHCP, IP range 192.168.50.100 -- 192.168.50.249, lease time 12h
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: using only locally-known addresses for domain test
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: using only locally-known addresses for domain onion
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: using only locally-known addresses for domain localhost
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: using only locally-known addresses for domain local
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: using only locally-known addresses for domain invalid
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: using only locally-known addresses for domain bind
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: using nameserver 127.0.0.1#7874
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: using only locally-known addresses for domain lan
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: read /etc/hosts - 4 addresses
Fri May 19 21:07:37 2023 daemon.info dnsmasq[17178]: read /tmp/hosts/dhcp.cfg01411c - 6 addresses
Fri May 19 21:07:37 2023 daemon.info dnsmasq-dhcp[17178]: read /etc/ethers - 0 addresses
Fri May 19 21:07:38 2023 daemon.notice procd: /etc/rc.d/K01AdGuardHome: AdGuardHome service disabled
Fri May 19 21:07:38 2023 daemon.err nmbd[7462]: [2023/05/19 21:07:38.255694,  0] ../../source3/nmbd/nmbd.c:60(terminate)
Fri May 19 21:07:38 2023 daemon.err nmbd[7462]:   Got SIGTERM: going down...
Fri May 19 21:07:38 2023 cron.err crond[17370]: crond (busybox 1.33.2) started, log level 5
Fri May 19 21:07:38 2023 daemon.notice procd: /etc/rc.d/K10autotimeset: Auto set has stop.
Fri May 19 21:07:38 2023 daemon.notice miniupnpd[11660]: shutting down MiniUPnPd
Fri May 19 21:07:43 2023 daemon.info dnsmasq[17178]: exiting on receipt of SIGTERM
Fri May 19 21:07:44 2023 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri May 19 21:07:44 2023 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri May 19 21:07:47 2023 daemon.info dnsmasq[18182]: Connected to system UBus
Fri May 19 21:07:47 2023 daemon.info dnsmasq[18182]: started, version 2.85 cache disabled
Fri May 19 21:07:47 2023 daemon.info dnsmasq[18182]: DNS service limited to local subnets
Fri May 19 21:07:47 2023 daemon.info dnsmasq[18182]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Fri May 19 21:07:47 2023 daemon.info dnsmasq[18182]: UBus support enabled: connected to system bus

jjm2473, 2023-05-27:
日志没记录到发生异常时的情况，可能是网卡挂掉了，但是内核又没崩溃，所以后面还能软重启。
试试杂项设置里面关掉wan口加速，还有关掉防火墙的FullconeNAT，或者只关掉其中一个。

btdwv, 2023-05-27:
> 

关WAN加速，开启FullconeNAT    断网了，大约17:27手动按开关重启
关FullconeNAT，开WAN加速       正常
都关掉                                           正常

btdwv, 2023-05-27:
> 日志没记录到发生异常时的情况，可能是网卡挂掉了，但是内核又没崩溃，所以后面还能软重启。 试试杂项设置里面关掉wan口加速，还有关掉防火墙的FullconeNAT，或者只关掉其中一个。

1. 关WAN加速，开启FullconeNAT 断网了，大约17:27手动按开关重启
2. 关FullconeNAT，开WAN加速 正常
3. 都关掉 正常

下面是断网、重启时的日志
Sat May 27 17:20:05 2023 daemon.notice ttyd[10502]: HTTP / - 192.168.50.110
Sat May 27 17:20:05 2023 daemon.notice ttyd[10502]: HTTP /token - 192.168.50.110
Sat May 27 17:20:05 2023 daemon.notice ttyd[10502]: WS   /ws - 192.168.50.110, clients: 1
Sat May 27 17:20:05 2023 daemon.notice ttyd[10502]: started process, pid: 3472
Sat May 27 17:20:08 2023 daemon.info dnsmasq[1561]: read /etc/hosts - 4 addresses
Sat May 27 17:20:08 2023 daemon.info dnsmasq[1561]: read /tmp/hosts/dhcp.cfg01411c - 6 addresses
Sat May 27 17:20:08 2023 daemon.info dnsmasq-dhcp[1561]: read /etc/ethers - 0 addresses
Sat May 27 17:20:11 2023 auth.info login[3472]: root login on 'pts/0'
Sat May 27 17:24:06 2023 daemon.warn miniupnpd[1727]: HTTP Connection from 192.168.50.228 closed unexpectedly
Sat May 27 17:24:37 2023 daemon.err modprobe: failed to find a module named tun
Sat May 27 17:24:43 2023 daemon.info dnsmasq[1561]: exiting on receipt of SIGTERM
Sat May 27 17:24:43 2023 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat May 27 17:24:43 2023 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: Connected to system UBus
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: started, version 2.85 cache disabled
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: DNS service limited to local subnets
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: UBus support enabled: connected to system bus
Sat May 27 17:24:46 2023 daemon.info dnsmasq-dhcp[6300]: DHCP, IP range 192.168.50.100 -- 192.168.50.249, lease time 12h
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: using only locally-known addresses for domain test
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: using only locally-known addresses for domain onion
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: using only locally-known addresses for domain localhost
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: using only locally-known addresses for domain local
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: using only locally-known addresses for domain invalid
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: using only locally-known addresses for domain bind
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: using nameserver 127.0.0.1#7874
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: using only locally-known addresses for domain lan
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: read /etc/hosts - 4 addresses
Sat May 27 17:24:46 2023 daemon.info dnsmasq[6300]: read /tmp/hosts/dhcp.cfg01411c - 6 addresses
Sat May 27 17:24:46 2023 daemon.info dnsmasq-dhcp[6300]: read /etc/ethers - 0 addresses
Sat May 27 17:26:17 2023 daemon.err modprobe: failed to find a module named tun
Sat May 27 17:26:22 2023 daemon.info dnsmasq[6300]: exiting on receipt of SIGTERM
Sat May 27 17:26:22 2023 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat May 27 17:26:22 2023 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: Connected to system UBus
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: started, version 2.85 cache disabled
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: DNS service limited to local subnets
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: UBus support enabled: connected to system bus
Sat May 27 17:26:25 2023 daemon.info dnsmasq-dhcp[7953]: DHCP, IP range 192.168.50.100 -- 192.168.50.249, lease time 12h
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: using only locally-known addresses for domain test
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: using only locally-known addresses for domain onion
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: using only locally-known addresses for domain localhost
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: using only locally-known addresses for domain local
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: using only locally-known addresses for domain invalid
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: using only locally-known addresses for domain bind
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: using nameserver 127.0.0.1#7874
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: using only locally-known addresses for domain lan
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: read /etc/hosts - 4 addresses
Sat May 27 17:26:25 2023 daemon.info dnsmasq[7953]: read /tmp/hosts/dhcp.cfg01411c - 6 addresses
Sat May 27 17:26:25 2023 daemon.info dnsmasq-dhcp[7953]: read /etc/ethers - 0 addresses
Sat May 27 17:27:12 2023 daemon.info procd: - shutdown -
Sat May 27 17:27:12 2023 user.notice DEBUG: System Shutting Down
Sat May 27 17:27:12 2023 daemon.notice procd: /etc/rc.d/K00debug: DEBUG: System Shutting Down
[... cut ...]
Sat May 27 17:28:19 2023 kern.info kernel: [   22.447455] br-lan: port 1(eth1) entered disabled state
Sat May 27 17:28:19 2023 kern.info kernel: [   22.453063] device eth1 entered promiscuous mode
Sat May 27 17:28:19 2023 kern.info kernel: [   22.458993] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
Sat May 27 17:28:19 2023 daemon.notice netifd: Interface 'lan' is enabled
Sat May 27 17:28:19 2023 daemon.notice netifd: Interface 'lan' is setting up now
Sat May 27 17:28:19 2023 daemon.notice netifd: Interface 'lan' is now up

jjm2473, 2023-06-01:
嗯，确实没记录到任何日志，如果只是fullconenat开启状态下才会崩溃，那就关掉fullconenat吧，反正有upnp一般也够用了

lawzhougc, 2023-09-09:
我的istoreos是做为旁路由使用  这段时间碰到向日葵不定时掉线的问题

jjm2473, 2023-09-10:
@lawzhougc 你的设备是ARS2吗，如果不是，那跟这个bug完全无关，去找向日葵的客服。