istoreos/istoreos #621: 请求对BCM43142/BCM943142HM驱动支持

S0DX, 2023-02-28:
请求支持老旧无线网卡

jjm2473, 2023-03-02:
先去查openwrt有没有支持，有的话，直接按照人家的教程就可以安装。使用最新的固件

jjm2473, 2023-03-18:
`opkg update; opkg install kmod-b43 ` 应该就支持了

S0DX, 2023-03-18:
> lspci能够看到网卡信息，但是使用dmesg可以看到返回消息Driver 'b43' was unable to register with bus_type 'bcma' because the bus was not initialized.

jjm2473, 2023-03-18:
确实kmod-bcma不一致导致

jjm2473, 2023-03-19:
已经修复，等固件发布

closed 2023-03-19 by jjm2473

S0DX, 2023-03-20:
此次更新后依然无法使用，dmsg显示dmesg | grep b43
[   14.501480] b43: module has no kallsyms
[   14.512942] b43legacy: module has no kallsyms

jjm2473, 2023-03-20:
> 此次更新后依然无法使用，dmsg显示dmesg | grep b43 [ 14.501480] b43: module has no kallsyms [ 14.512942] b43legacy: module has no kallsyms

这两个只是警告，我加上的，意思是使用了openwrt官方的内核模块。

jjm2473, 2023-03-20:
> 同时网络接口状态也无法查看，如有需要我可以提供主机网络映射地址提供调试。

不是说wifi吗，跟这个有什么关系？

jjm2473, 2023-03-20:
你可以安装openwrt官方固件看看能否支持，如果可以，istoreos一般也可以。