istoreos/istoreos #625: 关于单臂旁路由下，不开启DHCP的问题。

juzidefennu, 2023-03-01:
在单臂旁路由模式下面使用，旁路由不开启DHCP（主路由分配IP且网关为主路由）
使用一段时间后会发现，设置旁路由为网关的电脑出现访问网络慢，甚至加载不正常。
其他固件解决方式是
1.关闭LAN口桥接
2.防火墙自定义添加 iptables -t nat -I POSTROUTING -o eth0 -j MASQUERADE
可以缓解。
使用固件版本号为：istoreos-21.02.3-2023021714
请问你们有没有出现这个问题呢？而且是怎么解决的？

jjm2473, 2023-03-02:
如果你的客户端网关都不是旁路由了，那还跟旁路由有什么关系？除非旁路由里面有arp欺骗程序

jjm2473, 2023-03-02:
如果客户端无法上网，参考下这个 https://github.com/istoreos/istoreos/issues/598

juzidefennu, 2023-03-02:
> 如果客户端无法上网，参考下这个 #598
好的，谢谢。今天回家试试此方式。
因为，不想所有设备默认获得某个功能，所以用的主路由分配IP且网关为主路由。
需要使用那个功能时候，手动设置上网设备为旁路由网关。使用时间一长就会出现上述症状。
这个问题 一直都是按照这个 
https://www.right.com.cn/forum/thread-2983767-1-1.html
帖子在解决，但是会有问题，访问速度会限制在一个数值。

jjm2473, 2023-03-02:
看帖子的内容，作者应该是不太了解iptables和网络路由的机制。

实际上帖子中加的iptables规则就是相当于勾选防火墙的“IP 动态伪装”，没有任何区别。

还有，使用伪装以后，会有性能损耗。原本旁路由收到第一个包以后，如果发现不需要代理，可以发送ICMP重定向包让客户端和主路由之间直接转发剩下的流量，也就是之后的流量不需要再经过旁路由；使用IP伪装以后，所有流量都会经过旁路由，对于单网口设备，原本的全双工就相当于变成半双工了。

closed 2023-03-02 by jjm2473

juzidefennu, 2023-03-02:
> 看帖子的内容，作者应该是不太了解iptables和网络路由的机制。
> 
> 实际上帖子中加的iptables规则就是相当于勾选防火墙的“IP 动态伪装”，没有任何区别。
> 
> 还有，使用伪装以后，会有性能损耗。原本旁路由收到第一个包以后，如果发现不需要代理，可以发送ICMP重定向包让客户端和主路由之间直接转发剩下的流量，也就是之后的流量不需要再经过旁路由；使用IP伪装以后，所有流量都会经过旁路由，对于单网口设备，原本的全双工就相当于变成半双工了。

好的，非常感谢。学习了。