istoreos/istoreos #631: 21.02.3 2023022415版本IPv4和ipv6显示错误

lmabbe-lm, 2023-03-05:
21.02.3 2023022415版本IPv4和ipv6显示错误
这个是系统显示的
[image]
这个是baidu显示的ipv4地址
[image]

ipv6不应该显示fe80开头的，
应该显示外网的240开头的吧
[image]

LemonPAKA, 2023-03-05:
IPV4是因为你这个是运营商给的内网地址，IPV6是因为你是通过自动分配的WAN_6而不是WAN6所以检测错误了

lmabbe-lm, 2023-03-06:
> IPV4是因为你这个是运营商给的内网地址，IPV6是因为你是通过自动分配的WAN_6而不是WAN6所以检测错误了

那这个ipv4，怎么改才能变成外网的地址啊

LemonPAKA, 2023-03-07:
> > IPV4是因为你这个是运营商给的内网地址，IPV6是因为你是通过自动分配的WAN_6而不是WAN6所以检测错误了
> 
> 那这个ipv4，怎么改才能变成外网的地址啊

找运营商加钱（

jjm2473, 2023-03-10:
ipv4是因为你拨号得到的ip不是公网地址，而是运营商的大内网，对于互联网来说只能看到你的公网地址。

ipv6的可能有bug

jinxiao, 2023-03-11:
ipv6的我也有这个问题，读取的是wan口的IP而不是WAN6的
[image]
[image]

jjm2473, 2023-03-17:
新固件已修复

closed 2023-03-17 by jjm2473