istoreos/istoreos #638: iStoreOS版本21.02.3 2023022415 在 R2S 上会频繁掉线

huangxiangdan, 2023-03-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：每几个小时就会掉线，日志显示为lan

### (2) 路由器型号和固件版本

A：iStoreOS版本21.02.3 2023022415，R2S连lan口做单臂软路由

### (3) 详细日志和/或截图

A：日志被我冲掉了，没存下来，大致是r8152开始报错，然后显示eth1 is down。然后就断网了。现象和https://github.com/QiuSimons/YAOF/discussions/366 非常类似，用ethtool -K eth0 tx off rx off 没有根本解决问题，还是一段时间会断网，但解决方案好像是对的，就是换掉r8152驱动。

colindog, 2023-03-12:
21.02.3 2023031010 也还是会有

jjm2473, 2023-03-13:
试试 `ethtool -K eth1 tx off rx off`

jjm2473, 2023-03-13:
r8152驱动最近升级过，主要是为了支持 RTL8156 方案的 USB有线网卡

jjm2473, 2023-03-13:
运行测速程序几分钟会出现掉线问题吗，还是说闲置一段时间以后才会掉线？

jjm2473, 2023-03-13:
记录一些可能有用的链接
* https://forum.armbian.com/topic/20608-pull-request-enable-rockchip64-xhci-hcd-usb-trb-ent-quirk-for-rk3328/
* http://git.mis.ks.ua/US1GHQ/Armbian/commit/a8148d886a54e1e912b31fdf1cbbff44883589b1

colindog, 2023-03-13:
我第一次遇到是刚升 2023022415，晚上回家网络就掉了，重启了R2S，半个小时不到又掉；暂时拆掉了，周末升级了21.02.3 2023031010，半个小时左右网络又掉了，目前是刷回了 1216的版本，一切正常。

jjm2473, 2023-03-13:
试试这个固件 https://we.tl/t-0Ib94tCJcd

huangxiangdan, 2023-03-13:
我试了ethtool -K eth1 tx off rx off，没有作用。不会有频繁的报错信息（kern.warn kernel: [ 227.065809] r8152 4-1:1.0 eth1: Tx status -108），但几个小时之后照样掉线。
基本上是流量比较大的时候掉线的。

jjm2473, 2023-03-13:
@huangxiangdan 试试我发的测试固件

huangxiangdan, 2023-03-13:
我试了，但发现这个版本passwall安装失败了...没安装passwall，其他的跑起来估计没意义。

`-rw-r--r--    1 root     root      14025894 Mar 13 22:29 /tmp/is-root/tmp/PassWall_a53.run
84d47f73f4f1566fd3830706d78d3059  /tmp/is-root/tmp/PassWall_a53.run
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing OneClick install  100%  
Downloading https://istore.linkease.com/repo/all/compat/Packages.gz
Updated list of available packages in /var/opkg-lists/istore_compat
Downloading https://istore.linkease.com/repo/all/compat/Packages.sig
Signature check passed.
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/targets/rockchip/armv8/packages/Packages.sig
Signature check passed.
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/base/Packages.sig
Signature check passed.
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/luci/Packages.sig
Signature check passed.
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/packages/Packages.sig
Signature check passed.
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/routing/Packages.sig
Signature check passed.
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/telephony/Packages.sig
Signature check passed.
Package chinadns-ng (1.0-beta.25-1) installed in root is up to date.
Package dns2socks (2.1-2) installed in root is up to date.
Package dns2tcp (1.1.0-1) installed in root is up to date.
Package haproxy (2.6.6-1) installed in root is up to date.
Package ipt2socks (1.1.3-2) installed in root is up to date.
Package libudns (0.4-1) installed in root is up to date.
Package luci-app-passwall (4.55-2) installed in root is up to date.
Package luci-i18n-passwall-zh-cn (4.55-2) installed in root is up to date.
Package microsocks (1.0.2-2) installed in root is up to date.
Package naiveproxy (107.0.5304.87-3-1) installed in root is up to date.
Package pdnsd-alt (1.2.9b-par-2) installed in root is up to date.
Package shadowsocks-libev-ss-local (3.3.5-1) installed in root is up to date.
Package shadowsocks-libev-ss-redir (3.3.5-1) installed in root is up to date.
Package shadowsocks-libev-ss-server (3.3.5-1) installed in root is up to date.
Package shadowsocksr-libev-ssr-check (2.5.6-9) installed in root is up to date.
Package shadowsocksr-libev-ssr-local (2.5.6-9) installed in root is up to date.
Package shadowsocksr-libev-ssr-redir (2.5.6-9) installed in root is up to date.
Package shadowsocksr-libev-ssr-server (2.5.6-9) installed in root is up to date.
Package simple-obfs (0.0.5-1) installed in root is up to date.
Package tcping (0.3-1) installed in root is up to date.
Package unzip (6.0-8) installed in root is up to date.
Package xray-core (1.6.5-1) installed in root is up to date.
Package shadowsocks-libev-ss-local (3.3.5-1) installed in root is up to date.
Package shadowsocks-libev-ss-redir (3.3.5-1) installed in root is up to date.
ln: libmbedcrypto.so.3: File exists
ln: libmbedcrypto.so.7: File exists
Error loading shared library libmbedcrypto.so.7: No such file or directory (needed by /usr/bin/ss-local)
Error relocating /usr/bin/ss-local: mbedtls_cipher_update: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_free: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_auth_decrypt: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_setkey: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_finish: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_hmac: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_hmac_starts: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_reset: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_hmac_update: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md5_ret: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_starts: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_update: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_set_iv: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_get_size: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_init: symbol not found
[... cut ...]
Error relocating /usr/bin/ss-local: mbedtls_cipher_setup: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_free: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_info_from_string: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_hmac_finish: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_auth_encrypt: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_info_from_string: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_init: symbol not found`

jjm2473, 2023-03-13:
> 我试了，但发现这个版本passwall安装失败了...没安装passwall，其他的跑起来估计没意义。

如果你不用ss协议，可以忽略这个错误

jjm2473, 2023-03-13:
建议直接跑网络测速，如果有问题，应该马上就会掉线

huangxiangdan, 2023-03-13:
内网测速压测了100秒，跑到400多兆带宽。没出问题。其余的再观察。

jjm2473, 2023-03-13:
OK

jjm2473, 2023-03-14:
新的测试固件，性能可能有所提升 https://we.tl/t-KTHS2VbUxJ

huangxiangdan, 2023-03-14:
通过Teamviewer登录回家里，测了100s，但Teamviewer机器性能跟不上，只跑了100兆，晚上回去再压测看看。

jjm2473, 2023-03-14:
好

jjm2473, 2023-03-14:
如果是之前的bug，掉线以后过一会儿应该能重新连上，会自动恢复

jjm2473, 2023-03-14:
0310那个发布的版本在压测时会掉线吗，或者压测过程中速度变成0

huangxiangdan, 2023-03-14:
0310那个发布的版本我之前压测没有这么长时间，不会掉线，也没有速度变成0。而且我不管是这次还是之前，掉线之后从来不会重新连接...只能强制重启恢复。

jjm2473, 2023-03-14:
还有试试这个测试固件，性能和稳定性应该更好  https://we.tl/t-hXRhyC2nQb

jjm2473, 2023-03-14:
有重现方法就好办了，试试今晚发的 test-park 固件

huangxiangdan, 2023-03-14:
测了，今晚的固件牛X，测试直接跑到700M，翻墙试了几个4K，非常流畅。再观察一段时间试试稳定性。

jjm2473, 2023-03-14:
r2s提升这么明显吗？r68s使用2.5G的USB网卡大概有不到1/4的提升。

huangxiangdan, 2023-03-14:
还是哪里有问题。说完没多久就出问题了。我把日志贴下面。我只装了passwall，旁路由供几台机器翻墙。

> Tue Mar 14 23:23:22 2023 daemon.info dnsmasq[8643]: using 12 more nameservers
Tue Mar 14 23:23:22 2023 daemon.info dnsmasq[8643]: read /etc/hosts - 4 addresses
Tue Mar 14 23:23:22 2023 daemon.info dnsmasq[8643]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Tue Mar 14 23:23:22 2023 daemon.info dnsmasq-dhcp[8643]: read /etc/ethers - 0 addresses
Tue Mar 14 23:23:22 2023 cron.err crond[8822]: crond (busybox 1.33.2) started, log level 5
Tue Mar 14 23:23:58 2023 user.info : luci: accepted login on / for root from 192.168.31.143
Tue Mar 14 23:27:44 2023 daemon.info dnsmasq-dhcp[8643]: DHCPINFORM(br-lan) 192.168.31.164 14:10:9f:d8:27:c1
Tue Mar 14 23:27:44 2023 daemon.info dnsmasq-dhcp[8643]: DHCPACK(br-lan) 192.168.31.164 14:10:9f:d8:27:c1 mac-sc
Tue Mar 14 23:31:07 2023 daemon.info dnsmasq-dhcp[8643]: DHCPINFORM(br-lan) 192.168.31.164 14:10:9f:d8:27:c1
Tue Mar 14 23:31:07 2023 daemon.info dnsmasq-dhcp[8643]: DHCPACK(br-lan) 192.168.31.164 14:10:9f:d8:27:c1 mac-sc
Tue Mar 14 23:37:31 2023 daemon.info dnsmasq-dhcp[8643]: DHCPINFORM(br-lan) 192.168.31.164 14:10:9f:d8:27:c1
Tue Mar 14 23:37:31 2023 daemon.info dnsmasq-dhcp[8643]: DHCPACK(br-lan) 192.168.31.164 14:10:9f:d8:27:c1 mac-sc
Tue Mar 14 23:42:53 2023 kern.warn kernel: [ 1226.437684] ------------[ cut here ]------------
Tue Mar 14 23:42:53 2023 kern.info kernel: [ 1226.437714] NETDEV WATCHDOG: eth1 (r8152): transmit queue 0 timed out
Tue Mar 14 23:42:53 2023 kern.warn kernel: [ 1226.437835] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:473 dev_watchdog+0x2f4/0x300
Tue Mar 14 23:42:53 2023 kern.notice kernel: [ 1226.437841] Modules linked in: xt_FULLCONENAT rtl8192cu rtl8192c_common rtl_usb qcserial pppoe ppp_async option mt76x0u mt76x0_common l2tp_ppp iwlmvm iwldvm ipw cdc_mbim ath10k_pci ath10k_core ath wireguard usb_wwan usb_serial_simple ti_usb_3410_5052 sr9700 smsc95xx sierra_net sierra rtlwifi rtl8812au rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic plusb pl2303 oti6858 mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt7663u mt7663_usb_sdio_common mt7615_common mt76_usb mt76_connac_lib mt76 mos7840 mos7720 mmc_spi mct_u232 mcs7830 mac80211 libchacha20poly1305 libblake2s keyspan kalmia iwlwifi ipt_REJECT io_edgeport huawei_cdc_ncm garmin_gps ftdi_sio dm9601 cypress_m8 cp210x chaoskey chacha_neon ch341 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem belkin_sa ax88179_178a asix ark3116 aqc111 zstd xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_helper xt_ecn
Tue Mar 14 23:42:53 2023 kern.notice kernel: [ 1226.438327]  xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_bpf xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xr_usb_serial_common visor via_velocity via_rhine usbserial usbnet usblp usbatm ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_fsm ts_bm slhc sky2 skge sis900 sis190 sfp sfc_falcon sfc sch_mqprio sch_cake rtl8150 rng_core r8168 r8152 r8125 r6040 poly1305_neon pegasus pcnet32 of_mmc_spi ntfs3 niu nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda
Tue Mar 14 23:42:53 2023 kern.notice kernel: [ 1226.438789]  nf_conncount ne2k_pci mlx5_core mlx4_en mlx4_core mdio_i2c mdio_gpio mdio_bitbang macvlan lzo libcurve25519_generic libchacha libblake2s_generic kaweth iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables igc iavf hso hid_cp2112 forcedeth ezusb exfat ethoc et131x e100 dmx_usb crc7 crc_ccitt compat cm109 cls_flower cdc_wdm cdc_acm br_netfilter bnx2x bnx2 be2net atl2 atl1e atl1c atl1 asn1_decoder alx 8390 8139too 8139cp fuse sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_csum em_ipset cls_bpf act_bpf act_ctinfo act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact configs sg pwm_fan drivetemp usb_f_acm u_serial usb_f_mass_storage usb_f_hid usb_f_ecm u_ether libcomposite ledtrig_usbport trelay spi_ks8995 siit xt_set ip_set_list_set
Tue Mar 14 23:42:53 2023 kern.notice kernel: [ 1226.439298]  ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink net2280 st sr_mod cdrom rtl8366s rtl8366rb rtl8306 ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip17xx b53_mdio b53_common nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3 nfsd nfs msdos bonding 3c59x ip6_gre ip_gre gre vmxnet3 ixgbevf ixgbe igb i2c_algo_bit e1000 ifb dummy nat46 l2tp_ip6 l2tp_ip l2tp_eth sctp mdio l2 [... cut ...]
Tue Mar 14 23:55:26 2023 daemon.notice ddnsto[22429]: dial ipv4 failed ret=-102 retry=2 remote=tunnel.ddnsto.com:4445
Tue Mar 14 23:55:43 2023 daemon.notice ddnsto[22429]: dial ipv4 failed ret=-102 retry=3 remote=tunnel.ddnsto.com:4445
Tue Mar 14 23:56:02 2023 daemon.notice ddnsto[22429]: dial ipv4 failed ret=-102 retry=4 remote=tunnel.ddnsto.com:4445
Tue Mar 14 23:56:02 2023 daemon.notice ddnsto[5391]: check reboot failed, ret=-1
Tue Mar 14 23:56:56 2023 daemon.notice ddnsto[22836]: dial ipv4 failed ret=-102 retry=1 remote=tunnel.ddnsto.com:4445

jjm2473, 2023-03-14:
测速固件加上 `ethtool -K eth1 tx off rx off` 看看行不行

jjm2473, 2023-03-15:
这个是换了驱动的固件  https://we.tl/t-g2khr4thPU

huangxiangdan, 2023-03-15:
昨晚发的，我加上了ethtool -K eth1 tx off rx off 正在测试，要是不稳定我就换你新发的固件。

jjm2473, 2023-03-16:
加上`ethtool -K eth1 tx off rx off`以后速度和稳定性怎么样

huangxiangdan, 2023-03-16:
昨晚好像断了一次，不太确定，后来就刷降级的固件iStoreOS 21.02.3 2023031510-test-cdc，今天再回去测试下。 

| |
黄祥旦
|
|
***@***.***
|

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2023年03月16日 10:18 |
| 收件人 | ***@***.***> |
| 抄送人 | Danny ***@***.***> ,
***@***.***> |
| 主题 | Re: [istoreos/istoreos] iStoreOS版本21.02.3 2023022415 在 R2S 上会频繁掉线 (Issue #638) |

加上ethtool -K eth1 tx off rx off以后速度和稳定性怎么样

huangxiangdan, 2023-03-17:
请教一个问题，我最近试了不管哪个版本做单臂软路由，什么插件都没装，软路由仅仅配置了网关，都会出现访问困难的问题。具体表现是打开百度，需要加载好长时间，后来抓包看到ICMP的Time-to-live exceeded的报错信息。看起来像出现了环路的问题。

我试着做了一下方法debug
1. 把我的笔记本网关设置成路由器（非软路由）的地址，一切正常
2. 把软路由拿到公司去，相同的配置，网关设置成软路由地址，也一切正常。
3. 重启路由器和软路由，问题照旧。

目前不确定是我的路由器问题还是软路由问题，路由器是小米的，基本没做过特殊配置。
不知道有没有碰到过类似的情况。

jjm2473, 2023-03-18:
1. 关掉ipv6
2. https://github.com/istoreos/istoreos/issues/598

closed 2023-03-20 by jjm2473

simplysexyboy, 2023-08-15:
断lan 什么原因？

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.176 (build@buildserver) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 2023070614) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Thu Aug 3 02:21:10 2023
[    0.000000] Machine model: FriendlyElec NanoPi R2S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] On node 0 totalpages: 261632
[    0.000000]   DMA zone: 4088 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 261632 pages, LIFO batch:63
[    0.000000] cma: Reserved 8 MiB at 0x000000003f400000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 21 pages/cpu s47064 r8192 d30760 u86016
[    0.000000] pcpu-alloc: s47064 r8192 d30760 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1001708K/1046528K available (8638K kernel code, 2404K rwdata, 2424K rodata, 1728K init, 734K bss, 36628K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000009] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001403] Console: colour dummy device 80x25
[    0.001876] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.002824] pid_max: default: 32768 minimum: 301
[    0.003485] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004167] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.007398] rcu: Hierarchical SRCU implementation.
[    0.008192] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.009916] smp: Bringing up secondary CPUs ...
[    0.011100] Detected VIPT I-cache on CPU1
[    0.011204] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.012039] Detected VIPT I-cache on CPU2
[    0.012114] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.012954] Detected VIPT I-cache on CPU3
[    0.013027] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.013164] smp: Brought up 1 node, 4 CPUs
[    0.016391] SMP: Total of 4 processors activated.
[    0.016823] CPU features: detected: 32-bit EL0 Support
[    0.017296] CPU features: detected: CRC32 instructions
[    0.018113] CPU: All CPU(s) started at EL2
[    0.018598] alternatives: patching kernel code
[    0.030321] KASLR disabled due to lack of seed
[... 444 lines cut ...]
[   23.125057] br-lan: port 1(eth1) entered forwarding state
[   23.204507] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:00] driver [RTL8211E Gigabit Ethernet] (irq=POLL)
[   23.205460] rk_gmac-dwmac ff540000.ethernet: Set MAC tx_delay: 0x24 rx_delay: 0x18 for PHY 0x001cc915
[   23.213531] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[   23.214252] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[   23.214868] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[   24.171340] br-lan: port 1(eth1) entered disabled state
[   26.334200] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   26.335060] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   26.832995] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[   26.834107] r8152 4-1:1.0 eth1: carrier on
[   26.835272] br-lan: port 1(eth1) entered blocking state
[... cut ...]
[  103.531864] overlayfs: upperdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  103.533182] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  103.534359] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  103.535522] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.