istoreos/istoreos #639: R4S 使用 2023031010 版无法启动

tanshiqi, 2023-03-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：R4S 使用 2023031010 版无法启动

### (2) 路由器型号和固件版本

A：R4S

### (3) 详细日志和/或截图

A：无法获取IP，手工设置为192.168.100.1后也无法访问

jjm2473, 2023-03-13:
系统指示灯正常吗，如果正常，那重新刷sd卡试试

tanshiqi, 2023-03-13:
灯亮的，但是不知道是不是正常，换了2张卡都一样，启动不了，换成2023030414就没问题

jjm2473, 2023-03-13:
系统的灯在启动过程中应该会闪，启动完毕会常亮，看看是不是这样。

你说的是20230224吧，没有20230304

tanshiqi, 2023-03-13:
目前正常的是istoreos-21.02.3-2023030414-r4s-squashfs.img.gz

jjm2473, 2023-03-13:
2023030414没有发布过啊，是测试固件吗

jjm2473, 2023-03-13:
你看看启动过程中灯的状态是否正常

tanshiqi, 2023-03-13:
这里下载的：https://fw.koolcenter.com/iStoreOS/r4s/
[image]
难道地址错了？

jjm2473, 2023-03-13:
你还是看看0310启动过程中的灯状态对不对吧

tanshiqi, 2023-03-13:
可以了，等了很久，获取到IP了，时间比之前的版本长了很多哎

jjm2473, 2023-03-13:
奇怪，难道是ip冲突？

第一次启动一两分钟很正常，看看之后启动要多久。

tanshiqi, 2023-03-13:
设置为固定IP后，启动没问题了，30秒左右，现在都正常了，非常感谢