istoreos/istoreos #642: r66s接口状态丢失

rootphantomer, 2023-03-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：r66s接口状态丢失，但不影响使用，点进到网口配置里也没有内容

### (2) 路由器型号和固件版本

A：R66S，21.02.3 2023031010

### (3) 详细日志和/或截图

A：
[image]

[image]

jjm2473, 2023-03-13:
你看看 /etc/config/network 文件是不是损坏了

rootphantomer, 2023-03-13:
@jjm2473 目测没有损坏。
[image]

jjm2473, 2023-03-14:
登录后台以后，打开后台的这个地址看看 `/cgi-bin/luci/istore/network/interface/config/` ，例如 http://192.168.100.1/cgi-bin/luci/istore/network/interface/config/

如果有报错信息，就复制下来

rootphantomer, 2023-03-15:
[image]
```
{"error":"exit status 1","success":-1000}
```

jjm2473, 2023-03-15:
触发错误以后，终端执行 `logread | grep quickstart` 看看

rootphantomer, 2023-03-15:
> 触发错误以后，终端执行 `logread | grep quickstart` 看看

[image]

jjm2473, 2023-03-16:
可能你自己升级过系统的软件包？恢复出厂设置以后再看看有没有同样的问题

rootphantomer, 2023-03-16:
> ip -pretty -detail -json link show >/dev/null

[image]

>可能你自己升级过系统的软件包？恢复出厂设置以后再看看有没有同样的问题

恩，升级过一些。

jjm2473, 2023-03-17:
@rootphantomer 哦，你把ip-full包卸载了，你的这个ip命令是精简版。执行 `opkg update; opkg install ip-full` 就可以恢复

closed 2023-03-17 by jjm2473

rootphantomer, 2023-03-17:
@jjm2473 ip-full 有，卸载重装ip-full的时候发现有个上层依赖组件没了，重装ip-full后就好了。谢谢。

jjm2473, 2023-03-17:
看你截图的ip命令是从busybox来的，要么没装ip-full，要么你更新的时候失败了