istoreos/istoreos #647: r66s 21.02.3 2023031010 版本使用一段时间后会网络会降速

rootphantomer, 2023-03-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：r66s 21.02.3 2023031010 版本使用一段时间后会网络会降速。通过 系统便利工具 进行测试会从本来的500+Mbps，降速到了95Mbps，看了下千兆协商成百兆协议，重启后测试恢复正常，但大概30分钟左右后就变回百兆宽带速度。上一个版本是没遇到这个问题。

### (2) 路由器型号和固件版本

A：r66s 21.02.3 2023031010

### (3) 详细日志和/或截图

A：
[image]

[image]

rootphantomer, 2023-03-15:
换了其他固件也一样，不是istore的问题，好像是D2开了Turbo ACC后，导致降速。关了以后目前正常。

closed 2023-03-15 by rootphantomer

rootphantomer, 2023-03-16:
> 换了其他固件也一样，不是istore的问题，是我有个usb转网口的设备，插在usb接口上，但是没有使用导致的。拔了就没出问题了

也不是usb也不是acc的问题，固件换了两个千兆速率后，过段时间就变百兆。。。。感觉是上层路由的问题

jjm2473, 2023-03-17:
那换网线试试

rootphantomer, 2023-03-17:
> 那换网线试试

网口吹了口气，可能是网口进灰了，吹了气后没问题了。。。

jjm2473, 2023-03-17:
居然是因为这个 🤣