istoreos/istoreos #648: 安装jellyfin插件 设置汉语(简体)后 无效

CleverMod, 2023-03-15:
## 1.关于你要提交的问题
设置中文后重启服务 重新安装插件 卸载容器和镜像后重新安装 均不能解决
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [是 ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：R4S   21.02.3-2023031010

CleverMod, 2023-03-16:
> 是istore的jellyfin插件有问题，还是jellyfin里面的插件有问题？
istore安装jellyfin然后 打开jellyfin后台  第一步设置汉语简体后   jellyfin的后台并没有切换成中文   重启服务 容器镜像卸载重装都不行

jjm2473, 2023-03-16:
那跟istore的jellyfin插件无关，要么你配置的数据目录有问题，例如ntfs或者无法写入，要么jellyfin自身bug，不过还是前者的可能性更大

closed 2023-03-16 by jjm2473

CleverMod, 2023-03-16:
> 那跟istore的jellyfin插件无关，要么你配置的数据目录有问题，例如ntfs或者无法写入，要么jellyfin自身bug，不过还是前者的可能性更大

目录应该是什么格式的能指点指点么  搞了半天了没搞好

jjm2473, 2023-03-16:
哪有什么格式，在ext4分区里面的一个文件夹就行了。假设你的ext4的分区挂载到了 `/mnt/sda1` ，那你就可以填 `/mnt/sda1/config/jellyfin`。

这种基础知识就别在这里问了好吧，问群里的人，就算是小白都知道。