istoreos/istoreos #656: R4SE 固件刷入emmc无法启动

lucas1024, 2023-03-19:
R4SE 固件刷入emmc无法启动

jjm2473, 2023-03-19:
无R4SE测试，R4SE是其他人贡献的代码，而且未收到其他人反馈无法启动。
你是怎么刷机的？

lucas1024, 2023-03-20:
官方固件的EMMC刷机助手，还有官方给出的带EFlasher的刷机包都试过
iStoreOS用过0224和0317，都是刷进EMMC后重新通电，只有power的红灯是亮的
设备用的是R4SE带独立MAC的版本

jjm2473, 2023-03-20:
.img.gz固件下载以后是不是没解压成img？ 仔细查看eMMC刷机教程 https://doc.linkease.com/zh/guide/istoreos/install_rockchip_sysupgrade.html

jjm2473, 2023-03-20:
比较简单的方式是先刷到tf卡启动，启动之后登录后台，将固件命名成`1.img.gz`后，在“系统-文件传输”里面将固件上传上去，然后打开终端执行 `zcat /tmp/upload/1.img.gz | dd of=/dev/mmcblk0 bs=4096`（注意将mmcblk0改成emmc的实际设备名称，可能是mmcblk0也可能是mmcblk1）

jjm2473, 2023-03-20:
usb刷机的话，参考这个教程 https://supes.top/nanopi-r4se%E7%BA%BF%E5%88%B7openwrt%E8%87%B3emmc%E6%95%99%E7%A8%8B/

注意也是要先解压固件

jjm2473, 2023-03-20:
还有这个教程 https://github.com/DHDAXCW/DoorNet-1-2/blob/mere/emmc.md

closed 2023-03-22 by jjm2473

lucas1024, 2023-03-25:
USB刷机和DD命令都用过，刷进EMMC还是无法启动，用fdisk -l查看刷过的emmc只有三个分区
[image]

jjm2473, 2023-03-26:
本来就三个分区，要那么多分区干嘛。写进emmc以后，要把sd卡弹出

lucas1024, 2023-03-26:
sd卡弹出来等了半小时，系统也没启动，只有power灯饰亮的，sys灯不亮

jjm2473, 2023-03-26:
你那个有9个分区的是什么？sd卡还是emmc？

istoreos只有三个分区。

lucas1024, 2023-03-26:
九个分区的是SD卡，上面三个分区的是EMMC里的istoreos，EMMC里面系统没办法启动，插SD卡看看EMMC是不是空的