istoreos/istoreos #660: 请求在网络-DHCP/DNS 界面添加 自定义挟持域名或者自定义dns解析 选项

ForestL18, 2023-03-22:
请求在网络-DHCP/DNS 界面添加 自定义挟持域名或者自定义dns解析 选项，否则每次都需要到命令行终端修改hosts很麻烦

jjm2473, 2023-03-22:
我靠，本来就有啊，能不能自己研究下，能不能谷歌搜索下，居然还发issue，居然还好意思说修改hosts。

很多地方都能添加域名映射：
1. “网络-主机名映射-主机/域名列表”，“新增”；
2. “网络-DHCP/DNS-常规设置”，“地址”选项添加，格式是“/abc.com/10.1.1.1”。

如果你要劫持局域网所有的dns请求，而不只是使用软路由dns的客户端，在“网络-DHCP/DNS-常规设置” 勾选 “DNS 重定向”就行了。

closed 2023-03-22 by jjm2473

ForestL18, 2023-03-22:
> 我靠，本来就有啊，能不能自己研究下，能不能谷歌搜索下，居然还发issue，居然还好意思说修改hosts。
> 
> 很多地方都能添加域名映射：
> 
> 1. “网络-主机名映射-主机/域名列表”，“新增”；
> 2. “网络-DHCP/DNS-常规设置”，“地址”选项添加，格式是“/abc.com/10.1.1.1”。
> 
> 如果你要劫持局域网所有的dns请求，而不只是使用软路由dns的客户端，在“网络-DHCP/DNS-常规设置” 勾选 “DNS 重定向”就行了。

本来在网上基本不和别人争吵，但我今天就要喷你。
①回答个issue给你牛坏了是吧，不知道还以为Linux是你开源的呢，op是你一力写成的呢。本事没见到多少，狗叫声倒是不小。②issue我想提就提，你不想答可以不答，没人想听你狗叫。怎么？op是大家都能讨论的，到你这就变成你的了是吧，我觉得istoreOS和其他op界面略有不同，我不能提点意见？
③你要真这么牛你自己写个op，我保证不去你那提issue。比别人懂多点就居高临下，咄咄逼人？你受到的的教育就是这么教你的？（当然我也不能排除你没受过教育）
④国内的开源环境为什么烂，一方面是白嫖确实多，还有一方面就是你这种开发者，可能别人尊重你喊你声大佬，你还当真了？
⑤最后我只想说，开源项目中，别人想提什么提什么，只要不违反开源规则。你这么会叫不如去干点别的？回村当个喇叭不好吗？

jjm2473, 2023-03-26:
> 我觉得istoreOS和其他op界面略有不同

"网络"菜单里面就是官方op的原版luci，19版本以上都这样，这都不知道？我倒是想问你为什么不去官方op提issue呢？

你说的“其他op”肯定不包括官方的op，因为只有部分非官方的op的luci还停留在17版本，例如国内其他几个openwrt的分支版本。

> 别人想提什么提什么

这里不是论坛，不是给你吹水的。

imxyd-com, 2024-03-10:
还真不是想提什么提什么，未能满足“提问的艺术”，是很无礼、无耻、无德的行为。
提问的艺术中，最简单通俗易懂的一条是：“先自行查阅网络，尽其所能未能找到答案时。”

zq500480, 2025-09-14:
T-mobile的wifi-calling是这样设置，address=/depdg.epc.mnc260.mcc310.pub.3gppnetwork.org/208.54.49.131
address=/ss.epdg.epc.mnc260.mcc310.pub.3gppnetwork.org/208.54.36.3
address=/epdg.epc.mnc260.mcc310.pub.3gppnetwork.org/208.54.35.163
address=/ss.epdg.epc.mnc260.mcc310.pub.3gppnetwork.org/208.54.35.163
address=/ss.epdg.epc.geo.mnc260.mcc310.pub.3gppnetwork.org /208.54.87.3，去掉全面address.