istoreos/istoreos #665: 希望内核能开启Kernel Debug Info功能？

Olympus-E, 2023-03-22:
x86固件似乎是关闭的，kernel panic自动重启以后不会生成crashlog文件，尝试过人为制造内核崩溃也没有生成。有没有可能开启这个功能？最近总是kernel panic，debug需要日志文件

[](url)
Global settings settings --->
[*] Compile the kernel with symbol table information
(CONFIG_KERNEL_KALLSYMS=y)
[*] Compile the kernel with debug information
(CONFIG_KERNEL_DEBUG_INFO=y)

Olympus-E, 2023-03-23:
自己编译了一个系统，但似乎只开这两个不行

感觉内核可以启用kdump和kexec支持？

[https://www.open-mesh.org/projects/devtools/wiki/Crashdumps_with_kexec](url)

在虚拟机环境下测试不成功，可能是因为Vmware的缘故，但手头上没有其他物理机可以测试了。

jjm2473, 2023-03-26:
这个可以通过crashlog补丁把崩溃日志写到内存里，崩溃重启以后还能读出来，之后看看x86行不行。目前3568和ars2都有这个功能

jjm2473, 2023-04-06:
openwrt已经把crashlog补丁删了，所以没法开启内核日志写入内存的功能，自己用串口调试吧。

closed 2023-04-06 by jjm2473

jjm2473, 2023-04-06:
不知道这个有没有用 https://www.kernel.org/doc/html/v5.10/admin-guide/ramoops.html