istoreos/istoreos #669: 边栏的 “网络存储” 丢失

GeQ1an, 2023-03-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：边栏的 “网络存储” 项丢失。今天才注意到这个问题，为此重置了设备。重置后是有的，设置网络、修改内网地址、更改密码时都在，但在我执行了 [安装 OpenClash 依赖的命令](https://github.com/vernesong/OpenClash/releases/tag/v0.45.103-beta) 后，管理后台要求我重新登录，登陆后边栏的 “网络存储” 丢失。是否和依赖安装或更新有关？边栏丢失造成了使用不便，请求帮忙测试找到原因。

### (2) 路由器型号和固件版本

A：x86 N5105，iStoreOS 21.02.3 2023032001

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2023-03-31:
不要安装 `luci-compat luci luci-base`，以免更新到官方版本，官方版本就是没有“网络存储”菜单的。

opkg自己会解决依赖问题，不要瞎JB添加依赖。

对于你的情况，iStore里面安装systools，然后打开插件，选择“修复系统软件”，点击执行，然后会自动重启，之后刷新页面应该就行了。

closed 2023-03-31 by jjm2473

GeQ1an, 2023-04-01:
大佬 NB！果然是依赖的问题，很抱歉添麻烦了，因为对 OpenWrt 有关的知识储备比较匮乏，感谢。🙏🏻

StarkOnes, 2024-06-11:
我也遇到一样的问题，也是安装了那几个插件搞出来的问题，下载了systools试了，修复不了，但安装“alist网盘”后会出来“网络存储”，卸载后又没了。。。