istoreos/istoreos #672: 8125BG网卡，4网口2.5G。最新版本拨号失败

guugg, 2023-03-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：官网现在最新版本，开始拨号失败，等了几分钟拨成功了，使用一会就掉了（只设置拨号其他都是默认设置）。  后面怀疑版本问题在系统页面更新，直接导致连拨号都没成功过。 

### (2) 路由器型号和固件版本

A：N3160 主机  +  x86_efi

jjm2473, 2023-03-26:
应该跟固件无关，最近都没更新过8125驱动，也没更新网络相关的组件。

如果还有问题，就发下系统日志。

guugg, 2023-03-26:
> 应该跟固件无关，最近都没更新过8125驱动，也没更新网络相关的组件。
> 
> 如果还有问题，就发下系统日志。

Sun Mar 26 15:26:38 2023 daemon.warn pppd[11882]: Timeout waiting for PADO packets Sun Mar 26 15:26:38 2023 daemon.err pppd[11882]: Unable to complete PPPo Discovery Sun Mar 26 15:26:38 2023 daemon.info pppd[11882]: Exit.
Sun Mar 26 15:26:38 2023 daemon.notice netifd: Interface wan is now down
Sun Mar 26 15:26:38 2023 daemon.notice netifid: Interface wan is setting up now
Sum Mar 26 15:26:38 2023 daemon.err insmod: module is already loaded - stric
Sun Mar 26 15:2638 2023 daemon.err insmod: module is already loaded - epp_generis
Sun Mar 26 15:26:38 2023 daemon.err insmod: module is already loaded - pppos
Sum Mar 26 15:26:38 2023 daemon.err insmod: module is already loaded - pppos
Sun Mar 26 15:26:38 2023 daemon.notice netifd: wan (12084): ucie Entry not found
Sun Mar 26 15:26:38 2023 daemon.info popd[12101]: Plugin pppoe.so loaded.
Sun Mar 26 15:2638 2023 daemon.info ppod[12101]: PPPoE plugin from pppd 2.4.9
Sun Mar 26 15.26:38 2023 daemon.notice pod[12101]: appd 2.4.5 started by root, id O Sun Mar 26 15:26:53 2023 daemon.war pppd[12101]: Timeout waiting for PADO packets Sun Mar 26 15:25:53 2023 daemon.err pppd12101]: Unable to complete PPPoE Discomery Sun Mar 25 15:26:53 2023 daemon.info ppod[12101]: Exit.
Sun Mar 26 15:26:53 2023 daemon notice netifd: Interface "wan' is now down
Sun Mar 26 15:26:53 2023 daemon.notice netifd: Interface wan is setting up now
Son Mar 26 15:26:53 2023 daemon.err insmod: module is already loaded - sic
Sun Mar 26 15:26:58. 2023 daeron.err insmed: module is already loaded - pp_ generic
Sun Mar 26 15:26:53 2023 daemon.err inamod: module is already loaded - oppox Sun Mar 26 15:26:53 2023 daemon.err insmod: module is already loaded - pppoe
Sun Mar 26 15:26.53 2023 daemon.notice netiforwan (123091: uck: Entry not found
Sum Mar 26 15:26:53 2023 daemoninfo pood[12320]: Plugin popoe.so loaded.
Sun Mar 26 15:26:53 2023 daemon.info pod[12320]: PPPoE plugin from pppd 2.4.9
Sun Mar 26. 15:25:53 2023 daemon.notice pppd(12320]: god 2.4.9 started by root, uid o Sun Mar 26 15:27:08 2023 daemonaxam pope[12320]: Timeout waiting for PADO packets Sun Mar 26 15:27:08 2023 daemon.err pppdit2320]: Linable to complete PPPoE Discoverys Sun Mar 26 15:27:08 2023 daemon.info pod[12320]: Exit.
Son Mar 26 15:27:08 2023 daernon.notice netifd: Interface wan is now down
Sun Mar 26 15:27:08 2023 daemon notice netifd: Interface "wan is setting up now
Sun Mar 26 15:27:00 2023 daemon.err insmod: module is already loaded - sle
Sun Mar 26 15:27:09 2023 daemon.err insmod: module is already loaded - ppp_generic
Sun Mar 26 15:27:09 2023 daemon.err insmod: modulle is already loaded - pppos Sun Mar 26 15:27:09 2023 daemon.err insmod: module is already loaded - pppoe
Sun Mar 26 15:27:08 2023 daemon.motice netifd: wan (12522|: uce Entry not found
Sun Mar 26 15:27:09 2023 daemon info pppd[12539]: Plugin pppoe.so loaded.
Sun Mar 25 15:27:09 2023 daemon.info pppd[12538]: PPPoE plugin from pard 2.4.9
Sun Mar 26 15.27:09 2023 daemoninotice ppod[12539|: ppod 2.4.8 started by root und O Sun Mar 25 15:27:24 2023 daemon.warn pppd[12539): Timeout waiting for PA DO packets
Sun Mar 26 15:27:24 2023 daemon.err pppa[12539]: Unable to complete PPPoE Dissouery
Sun Mar 26 15:27.24 2023 daemon.info popa[125391: Exit.
Sun Mar 26 15:27:24 2023 daemon.notice netifd: Interface wan" is now down
Sun Mar 26 15:27:24 2023 daemonnotice netifd: Interface wan is setting up roug
Sun Mar 26 15:27:24 2023 daemon.err insmod: module is already loaded - sile
Sun Mar 26 15:27:24 2023 daemon.err insmod: module is already loaded - ppp generio
Sun Mar 26 15:27:24 2023 daemon.err insmod: module is already loaded - oppas Sun Mar 26 15:27.24 2023 daemon.err insmod: module is stready loaded - appoe Sun Mar 26 15:27:24 2023 daemon.notice netifd:wan 01274215 uck: Entry not faund
Sun Mar 26 15:27.24 2023 daemon.info ppod[127591: Plugin pppos.so lcaded.
Sun Mar 26 15:27.24 2023 daemon.info pppo[127591: PPPoE plugin from popd 2.4.8
Sun Mar 26 15:27:24 2023 daemon.notice pppod[12759] pppd 2.4.8 started by root, wid o

guugg, 2023-03-26:
- 下面日志的版本是`istoreos-21.02.3-2022121613-x86-64-squashfs-combined-efi.img`

```
Sun Mar 26 22:31:01 2023 daemon.warn pppd[13919]: Timeout waiting for PADO packets
Sun Mar 26 22:31:01 2023 daemon.err pppd[13919]: Unable to complete PPPoE Discovery
Sun Mar 26 22:31:01 2023 daemon.info pppd[13919]: Exit.
Sun Mar 26 22:31:01 2023 daemon.notice netifd: Interface 'wan' is now down
Sun Mar 26 22:31:01 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sun Mar 26 22:31:01 2023 daemon.err insmod: module is already loaded - slhc
Sun Mar 26 22:31:01 2023 daemon.err insmod: module is already loaded - ppp_generic
Sun Mar 26 22:31:01 2023 daemon.err insmod: module is already loaded - pppox
Sun Mar 26 22:31:01 2023 daemon.err insmod: module is already loaded - pppoe
Sun Mar 26 22:31:01 2023 daemon.notice netifd: wan (14371): uci: Entry not found
Sun Mar 26 22:31:01 2023 daemon.info pppd[14389]: Plugin pppoe.so loaded.
Sun Mar 26 22:31:01 2023 daemon.info pppd[14389]: PPPoE plugin from pppd 2.4.9
Sun Mar 26 22:31:01 2023 daemon.notice pppd[14389]: pppd 2.4.9 started by root, uid 0
Sun Mar 26 22:31:16 2023 daemon.warn pppd[14389]: Timeout waiting for PADO packets
Sun Mar 26 22:31:16 2023 daemon.err pppd[14389]: Unable to complete PPPoE Discovery
Sun Mar 26 22:31:16 2023 daemon.info pppd[14389]: Exit.
Sun Mar 26 22:31:16 2023 daemon.notice netifd: Interface 'wan' is now down
Sun Mar 26 22:31:16 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sun Mar 26 22:31:16 2023 daemon.err insmod: module is already loaded - slhc
Sun Mar 26 22:31:16 2023 daemon.err insmod: module is already loaded - ppp_generic
Sun Mar 26 22:31:16 2023 daemon.err insmod: module is already loaded - pppox
Sun Mar 26 22:31:16 2023 daemon.err insmod: module is already loaded - pppoe
Sun Mar 26 22:31:16 2023 daemon.notice netifd: wan (14681): uci: Entry not found
Sun Mar 26 22:31:16 2023 daemon.info pppd[14700]: Plugin pppoe.so loaded.
Sun Mar 26 22:31:16 2023 daemon.info pppd[14700]: PPPoE plugin from pppd 2.4.9
Sun Mar 26 22:31:16 2023 daemon.notice pppd[14700]: pppd 2.4.9 started by root, uid 0
Sun Mar 26 22:31:28 2023 daemon.err pppd[14700]: select (waitForPADO): Interrupted system call
Sun Mar 26 22:31:28 2023 daemon.warn pppd[14700]: Timeout waiting for PADO packets
Sun Mar 26 22:31:28 2023 daemon.err pppd[14700]: Unable to complete PPPoE Discovery
Sun Mar 26 22:31:28 2023 daemon.info pppd[14700]: Exit.
Sun Mar 26 22:31:28 2023 daemon.notice netifd: Interface 'wan' is now down
Sun Mar 26 22:31:28 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sun Mar 26 22:31:28 2023 daemon.err insmod: module is already loaded - slhc
Sun Mar 26 22:31:28 2023 daemon.err insmod: module is already loaded - ppp_generic
Sun Mar 26 22:31:28 2023 daemon.err insmod: module is already loaded - pppox
Sun Mar 26 22:31:28 2023 daemon.err insmod: module is already loaded - pppoe
Sun Mar 26 22:31:28 2023 daemon.notice netifd: wan (14950): uci: Entry not found
Sun Mar 26 22:31:28 2023 daemon.info pppd[14989]: Plugin pppoe.so loaded.
Sun Mar 26 22:31:28 2023 daemon.info pppd[14989]: PPPoE plugin from pppd 2.4.9
Sun Mar 26 22:31:29 2023 daemon.notice pppd[14989]: pppd 2.4.9 started by root, uid 0
Sun Mar 26 22:31:44 2023 daemon.warn pppd[14989]: Timeout waiting for PADO packets
Sun Mar 26 22:31:44 2023 daemon.err pppd[14989]: Unable to complete PPPoE Discovery
Sun Mar 26 22:31:44 2023 daemon.info pppd[14989]: Exit.
Sun Mar 26 22:31:44 2023 daemon.notice netifd: Interface 'wan' is now down
Sun Mar 26 22:31:44 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sun Mar 26 22:31:44 2023 daemon.err insmod: module is already loaded - slhc
Sun Mar 26 22:31:44 2023 daemon.err insmod: module is already loaded - ppp_generic
Sun Mar 26 22:31:44 2023 daemon.err insmod: module is already loaded - pppox
Sun Mar 26 22:31:44 2023 daemon.err insmod: module is already loaded - pppoe
Sun Mar 26 22:31:44 2023 daemon.notice netifd: wan (15204): uci: Entry not found
Sun Mar 26 22:31:44 2023 daemon.info pppd[15222]: Plugin pppoe.so loaded.
Sun Mar 26 22:31:44 2023 daemon.info pppd[15222]: PPPoE plugin from pppd 2.4.9
Sun Mar 26 22:31:44 2023 daemon.notice pppd[15222]: pppd 2.4.9 started by root, uid 0
Sun Mar 26 22:31:53 2023 daemon.info dnsmasq[7377]: read /etc/hosts - 4 addresses
Sun Mar 26 22:31:53 2023 daemon.info dnsmasq[7377]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sun Mar 26 22:31:53 2023 daemon.info dnsmasq-dhcp[7377]: read /etc/ethers - 0 addresses
Sun Mar 26 22:31:58 2023 user.notice wan_drop: current proto pppoe
Sun Mar 26 22:31:58 2023 daemon.info avahi-daemon[5434]: Registering new address record for fe80::a236:9fff:fe50:c305 on br-lan.*.
Sun Mar 26 22:31:58 2023 daemon.info avahi-daemon[5434]: Withdrawing address record for fe80::a236:9fff:fe50:c305 on br-lan.
Sun Mar 26 22:31:59 2023 daemon.warn pppd[15222]: Timeout waiting for PADO packets
Sun Mar 26 22:31:59 2023 daemon.err pppd[15222]: Unable to complete PPPoE Discovery
Sun Mar 26 22:31:59 2023 daemon.info pppd[15222]: Exit.
Sun Mar 26 22:31:59 2023 daemon.notice netifd: Interface 'wan' is now down
Sun Mar 26 22:31:59 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sun Mar 26 22:31:59 2023 daemon.err insmod: module is already loaded - slhc
Sun Mar 26 22:31:59 2023 daemon.err insmod: module is already loaded - ppp_generic
Sun Mar 26 22:31:59 2023 daemon.err insmod: module is already loaded - pppox
[... 63 lines cut ...]
[... cut ...]
Sun Mar 26 22:33:21 2023 daemon.err insmod: module is already loaded - ppp_generic
Sun Mar 26 22:33:21 2023 daemon.err insmod: module is already loaded - pppox
Sun Mar 26 22:33:21 2023 daemon.err insmod: module is already loaded - pppoe
Sun Mar 26 22:33:21 2023 daemon.notice netifd: wan (17586): uci: Entry not found
Sun Mar 26 22:33:21 2023 daemon.info pppd[17604]: Plugin pppoe.so loaded.
Sun Mar 26 22:33:21 2023 daemon.info pppd[17604]: PPPoE plugin from pppd 2.4.9
Sun Mar 26 22:33:21 2023 daemon.notice pppd[17604]: pppd 2.4.9 started by root, uid 0
```

guugg, 2023-03-27:
用的是铭瑄n3160主板不知道是不是不支持啊

jjm2473, 2023-03-28:
跟主板应该没关系

guugg, 2023-03-29:
> `Timeout waiting for PADO packets`确定不是自己没配置好光猫吗

我使用猫棒的，win10跟路由器op系统都可以秒拨号。应该没问题吧？ 会不会是网卡需要更新驱动了

> 跟主板应该没关系

jjm2473, 2023-03-29:
你的主板有多个网口吗，有没有插错口子？

guugg, 2023-03-29:
> 你的主板有多个网口吗，有没有插错口子？

是多网口的，我把每一个网口都试了；昨晚拨掉网卡用主板拨号也没成功，刷成win10就拨成功，win 数据中心 2022装不了驱动拨号失败。  刷其他op也是一样可以自动获取IP上网，不能拨号。 脑瘫了～

guugg, 2023-03-29:
现在突然想起来当时第一次刷忘了是刷哪个版本的包能拨号的了（等十分钟左右拨成功的）。 后面按教程开ipv6公网保存，就掉了然后才一直拨不成功的；也改过Mac地址。

guugg, 2023-03-30:
> 你的主板有多个网口吗，有没有插错口子？

挖槽~ 一时的大意。 不知道什么时候把猫棒的收发器按到只有optic ok灯亮。  所以才导致都拨号失败了win也拨失败，现在摁回2.5G就正常了。

更让我好奇的是！！！为什么 1G跟2.5G灯不亮的情况下红米AX6000路由器（ImmortalWrt固件）可以拨成功？？   就是因为这个能拨成功所以才没怀疑猫棒的问题     shit~

closed 2023-03-30 by guugg

jjm2473, 2023-04-04:
也许跑到了10G？🤣