istoreos/istoreos #675: 无法安装passwall 和SSrPLUS

BLACKHAN8316, 2023-03-26:
安装PassWall_a53.run，报错
Error relocating /usr/bin/ss-local: mbedtls_cipher_setkey: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_finish: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_hmac: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_hmac_starts: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_reset: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_hmac_update: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md5_ret: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_starts: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_update: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_set_iv: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_get_size: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_init: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_setup: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_setup: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_free: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_info_from_string: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_hmac_finish: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_auth_encrypt: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_info_from_string: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_init: symbol not found

jjm2473, 2023-03-26:
如果不用老的ss协议，就忽略这个错误，反正其他已经安装成功了。

closed 2023-03-26 by jjm2473

liliang233, 2023-04-18:
> 如果不用老的ss协议，就忽略这个错误，反正其他已经安装成功了。

要用要用！