istoreos/istoreos #677: 更新固件后硬盘挂载失败，恢复后路径出错

tinshou, 2023-03-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
前两天更新了固件之后，发现装系统的硬盘掉了，只剩下1.94/2G(原来是用两百g的固态装系统，另外1T拿来存数据)，因为这两g系统盘也满了，无法安装插件。于是在高级设置中查看挂载信息，发现overlayfs:/overlay只挂载了这2g的磁盘，于是把两百多g的磁盘挂载成了overlayfs，但是成功之后1T的磁盘又掉了，于是又把数据磁盘挂载上，但是现在虽然系统盘挂载上了两百g，但是路径好像不一致，导致安装的插件都无法查看，易有云从首页进入也是找不到磁盘。

### (1) 具体问题

A：帮忙看下怎么恢复原来的系统设置，因为前几天刚注册了个pt站，新手任务还没完成，现在所有插件都用不了，pt也快过期了。

### (2) 路由器型号和固件版本

A：主机是自己组装的q1900，固件版本21.02.3 2023032001

### (3) 详细日志和/或截图

A：
[image]
[image]

如果有进展请联系我，我在客服和群里都有留言，但是没有回复，如果有需要，付费也可以，麻烦帮忙恢复，感谢。电话18680549175 微信同号

GreenTeodoro839, 2023-03-28:
原来的磁盘数据在/overlay，你可以在首页点那个沙箱按钮，选退出沙箱就行

tinshou, 2023-03-29:
> 另外在网络公开电话号码是一个很危险的行为，建议尽快删除

感谢提醒，已经删除

tinshou, 2023-03-29:
> 原来的磁盘数据在/overlay，你可以在首页点那个沙箱按钮，选退出沙箱就行

好的，我试一下，感谢

tinshou, 2023-03-29:
> 原来的磁盘数据在/overlay，你可以在首页点那个沙箱按钮，选退出沙箱就行

我试了退出沙盒也没有恢复。然后我想着说原来系统的配置应该是在2g的那个分区，所以想把这个分区重新设置为overlayfs:overlay，但是尝试添加加载点为根文件系统，也执行了提示的命令，还是没有生效，系统盘还是200g那个分区，这个大佬知道怎么更改吗？
[image]
[image]

tinshou, 2023-03-29:
[image]

tinshou, 2023-03-29:
[image]

GreenTeodoro839, 2023-03-29:
> [image]

你把原来的分区占满了，导致无法修改原来的配置，已经废了，只能重装

jjm2473, 2023-03-30:
2G的空间一直都是overlay，不需要你配置。你能配置的只有沙箱，而且沙箱也不需要手动配置，首页就有。其他系统无关的盘自己挂载就行了，别挂载到/或者/overlay！

closed 2023-03-30 by jjm2473