istoreos/istoreos #681: 请问istore后台系统日志出现大量 tracker.dcdn.baidu.com代表什么

Zxklinux, 2023-03-30:
如下

Thu Mar 30 10:16:12 2023 daemon.warn dnsmasq[19209]: possible DNS-rebind attack detected: tracker.dcdn.baidu.com
Thu Mar 30 10:16:12 2023 daemon.warn dnsmasq[19209]: possible DNS-rebind attack detected: tracker.dcdn.baidu.com
Thu Mar 30 10:16:12 2023 daemon.warn dnsmasq[19209]: possible DNS-rebind attack detected: tracker.dcdn.baidu.com
Thu Mar 30 10:16:12 2023 daemon.warn dnsmasq[19209]: possible DNS-rebind attack detected: tracker.dcdn.baidu.com
Thu Mar 30 10:16:12 2023 daemon.warn dnsmasq[19209]: possible DNS-rebind attack detected: tracker.dcdn.baidu.com
Thu Mar 30 10:16:12 2023 daemon.warn dnsmasq[19209]: possible DNS-rebind attack detected: tracker.dcdn.baidu.com
Thu Mar 30 10:16:12 2023 daemon.warn dnsmasq[19209]: possible DNS-rebind attack detected: tracker.dcdn.baidu.com
Thu Mar 30 10:16:12 2023 daemon.warn dnsmasq[19209]: possible DNS-rebind attack detected: tracker.dcdn.baidu.com
Thu Mar 30 10:16:12 2023 daemon.warn dnsmasq[19209]: possible DNS-rebind attack detected: tracker.dcdn.baidu.com
。。。

这是作者在统计什么东西吗？还是有什么其他原因，谢谢

jjm2473, 2023-03-30:
不是，目前没啥统计。这个一般是内网的设备发出的，你把内网设备都断开看看还有没有，可能是什么盒子？

jjm2473, 2023-03-30:
https://zhuanlan.zhihu.com/p/114207971

https://blog.csdn.net/boliang319/article/details/38279195

closed 2023-03-30 by jjm2473