istoreos/istoreos #685: Exsi7虚拟的IstoreOS提示在线升级，固件下载后升级不了。

599662746, 2023-04-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

Exsi7虚拟的【iStoreOS 21.02.3 2023032001】固件，提示有新的固件。然后根据提示在线升级到【istoreos-21.02.3-2023033111-x86-64-squashfs-combined.img.gz】，第三步重启的时候，固件没更新上，还是提示固件更新。

到官网下载【istoreos-21.02.3-2023033111-x86-64-squashfs-combined.img.gz】手动备份升级依旧升级不了。

建议在线升级功能可以设置为不检测升级或某段时间内不提示升级。（个人有强迫症）

### (2) 路由器型号和固件版本

A：
虚拟机版本：VMware ESXi, 7.0.3, 20328353
现用固件：iStoreOS 21.02.3 2023032001
提示升级并自动下载的固件：istoreos-21.02.3-2023033111-x86-64-squashfs-combined.img.gz

------------------------------------------------------------------------------------
刷写镜像已上传。下面是列出的校验和及文件大小，将它们与原始文件进行比较以确保数据完整性。
单击下面的“继续”开始刷写。

大小: 138.18 MiB
MD5: 562b47bebd2f37edef8abe5257e5ae68
SHA256: c7aac89aa0ac5fbd5beffb76be83bfb21fccbdfec38dc5fec8dfa63f2ff82de8
------------------------------------------------------------------------------------

### (3) 详细日志和/或截图

A：
系统日志和内核日志太多了，没找到有用的信息。

599662746, 2023-04-01:
重新创建了一个虚拟机，就配置了wan口网络，其他都没配置，在线升级系统，好了。可能是我改了根目录的挂在路径了吧。

599662746, 2023-04-01:
我虚拟的iStore系统扩容到了10G，然后，我分了6G给根目录使用。如下：
--------------------------------------------------------------------------------------------
修改分区后，不可在线升级：
root@iStoreOS:~/Downloads# lsblk
NAME   MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda      8:0    0   10G  0 disk
├─sda1   8:1    0  128M  0 part /boot
├─sda2   8:2    0  256M  0 part /rom
├─sda3   8:3    0    2G  0 part /overlay
└─sda4   8:4    0    6G  0 part /
sr0     11:0    1 1024M  0 rom
root@iStoreOS:~/Downloads#
------------------------------------
root@iStoreOS:~/Downloads# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               131.0M    131.0M         0 100% /rom
tmpfs                     1.9G      3.7M      1.9G   0% /tmp
/dev/sda3                 1.9G    760.0M      1.1G  39% /overlay
/dev/sda4                 5.8G      1.4G      4.2G  25% /
tmpfs                     4.0M         0      4.0M   0% /mnt
/dev/sda1               126.0M      5.1M    118.4M   4% /boot
/dev/sda1               126.0M      5.1M    118.4M   4% /boot
tmpfs                   512.0K         0    512.0K   0% /dev
cgroup                    1.9G         0      1.9G   0% /sys/fs/cgroup
tmpfs                   512.0K         0    512.0K   0% /sys/module/md_mod/parameters/new_array
/dev/sda3                 1.9G    760.0M      1.1G  39% /overlay/upper/opt/docker
overlay                   1.9G    760.0M      1.1G  39% /overlay/upper/opt/docker/overlay2/66dc842520b003292fc36ee2380784369c5aded6bea674572220eff7980b0459/merged
overlay                   1.9G    760.0M      1.1G  39% /overlay/upper/opt/docker/overlay2/4f7863188849d1d990b82e85657ed4d3ae0e78f8f33c8b5751f5ceabf3f720f1/merged
root@iStoreOS:~/Downloads#
-----------------------------------------------------------------

------------------------------------------------------------------
默认分区，可在线升级：
root@iStoreOS:~# lsblk
NAME   MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda      8:0    0  2.4G  0 disk
├─sda1   8:1    0  128M  0 part /boot
├─sda2   8:2    0  256M  0 part /rom
└─sda3   8:3    0    2G  0 part /overlay
root@iStoreOS:~#
----------------------------
root@iStoreOS:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               131.0M    131.0M         0 100% /rom
tmpfs                     1.9G      4.5M      1.9G   0% /tmp
/dev/sda3                 1.9G    174.9M      1.7G   9% /overlay
overlayfs:/overlay        1.9G    174.9M      1.7G   9% /
tmpfs                     4.0M         0      4.0M   0% /mnt
/dev/sda1               126.0M      5.1M    118.4M   4% /boot
/dev/sda1               126.0M      5.1M    118.4M   4% /boot
tmpfs                   512.0K         0    512.0K   0% /dev
cgroup                    1.9G         0      1.9G   0% /sys/fs/cgroup
tmpfs                   512.0K         0    512.0K   0% /sys/module/md_mod/parameters/new_array
/dev/sda3                 1.9G    174.9M      1.7G   9% /overlay/upper/opt/docker
overlay                   1.9G    174.9M      1.7G   9% /overlay/upper/opt/docker/overlay2/a003e0405b6bb4ffca45eb370e95794830667c6d13421464e77fd11fbc9cfb39/merged
root@iStoreOS:~#
------------------------------------------------------------------------

599662746, 2023-04-01:
# 我虚拟的iStore系统扩容到了10G，然后，我分了6G给根目录使用。如下：

## 修改分区后，不可在线升级： 

```bash

root@iStoreOS:~# lsblk
NAME   MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda      8:0    0   10G  0 disk
├─sda1   8:1    0  128M  0 part /boot
├─sda2   8:2    0  256M  0 part /rom
├─sda3   8:3    0    2G  0 part /overlay
└─sda4   8:4    0    6G  0 part /
sr0     11:0    1 1024M  0 rom
root@iStoreOS:~#
root@iStoreOS:~#
root@iStoreOS:~#
root@iStoreOS:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               131.0M    131.0M         0 100% /rom
tmpfs                     1.9G      3.7M      1.9G   0% /tmp
/dev/sda3                 1.9G    760.0M      1.1G  39% /overlay
/dev/sda4                 5.8G      1.4G      4.2G  25% /
tmpfs                     4.0M         0      4.0M   0% /mnt
/dev/sda1               126.0M      5.1M    118.4M   4% /boot
/dev/sda1               126.0M      5.1M    118.4M   4% /boot
tmpfs                   512.0K         0    512.0K   0% /dev
cgroup                    1.9G         0      1.9G   0% /sys/fs/cgroup
tmpfs                   512.0K         0    512.0K   0% /sys/module/md_mod/parameters/new_array
/dev/sda3                 1.9G    760.0M      1.1G  39% /overlay/upper/opt/docker
overlay                   1.9G    760.0M      1.1G  39% /overlay/upper/opt/docker/overlay2/66dc842520b003292fc36ee2380784369c5aded6bea674572220eff7980b0459/merged
overlay                   1.9G    760.0M      1.1G  39% /overlay/upper/opt/docker/overlay2/4f7863188849d1d990b82e85657ed4d3ae0e78f8f33c8b5751f5ceabf3f720f1/merged
root@iStoreOS:~#

```

## 默认分区，可在线升级：

```bash

root@iStoreOS:~# lsblk
NAME   MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda      8:0    0  2.4G  0 disk
├─sda1   8:1    0  128M  0 part /boot
├─sda2   8:2    0  256M  0 part /rom
└─sda3   8:3    0    2G  0 part /overlay
root@iStoreOS:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               131.0M    131.0M         0 100% /rom
tmpfs                     1.9G      4.5M      1.9G   0% /tmp
/dev/sda3                 1.9G    174.9M      1.7G   9% /overlay
overlayfs:/overlay        1.9G    174.9M      1.7G   9% /
tmpfs                     4.0M         0      4.0M   0% /mnt
/dev/sda1               126.0M      5.1M    118.4M   4% /boot
/dev/sda1               126.0M      5.1M    118.4M   4% /boot
tmpfs                   512.0K         0    512.0K   0% /dev
cgroup                    1.9G         0      1.9G   0% /sys/fs/cgroup
tmpfs                   512.0K         0    512.0K   0% /sys/module/md_mod/parameters/new_array
/dev/sda3                 1.9G    174.9M      1.7G   9% /overlay/upper/opt/docker
overlay                   1.9G    174.9M      1.7G   9% /overlay/upper/opt/docker/overlay2/a003e0405b6bb4ffca45eb370e95794830667c6d13421464e77fd11fbc9cfb39/merged
root@iStoreOS:~#

```

jjm2473, 2023-04-11:
你都把根目录做成其他盘了，怎么自动升级啊。升级程序只会写原始的squashfs分区。而且你改了根目录以后，升级脚本也不一定能找到启动盘，可能会没法升级或者写错地方

closed 2023-04-18 by jjm2473