istoreos/istoreos #688: 畅网N6005物理安装iStore只有单核运行的问题

21527271, 2023-04-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用物理机安装iStore的时候只有一个核心被调用，其余核心显示闲置状态，使用coremark -h 跑分时，CPU使用率仅有25%

### (2) 路由器型号和固件版本

A：路由器型号是 畅网微控N6005 v4，路由器固件版本：

主机名 | iStoreOS
-- | --
型号 | Default string Default string
架构 | Intel(R) Pentium(R) Silver N6005 @ 2.00GHz
目标平台 | x86/64
固件版本 | iStoreOS 21.02.3 2023033111 / LuCI istoreos-21.02 branch git-23.088.34636-f8cc111

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2023-04-06:
coremask是默认单线程的吗？

jjm2473, 2023-04-07:
那不是你执行命令的参数有问题吗？跟系统有啥关系？openssl的测试命令也可以单线程或者多线程，需要自己指定参数啊。真是莫名其妙

closed 2023-04-07 by jjm2473