istoreos/istoreos #690: r2s 跑不满千兆

HalcyonDaysssss, 2023-04-03:
### (1) 具体问题

A：r2s跑不满千兆

### (2) 路由器型号和固件版本

A：软路由r2s，21.02.3 2023033111

### (3) 详细日志和/或截图
[image]

A：在百度上搜索过，恩山论坛说将wan口跟landui对调，依旧不行

HalcyonDaysssss, 2023-04-03:
好像说要开启usb3.0，但是不知道如何开启

jjm2473, 2023-04-06:
本来就有usb3啊
[image]

试试 20230310 固件性能怎么样吧，反正r2s的USB本身也不稳定

HalcyonDaysssss, 2023-04-06:
> 本来就有usb3啊 [image]
> 
> 试试 20230310 固件性能怎么样吧，反正r2s的USB本身也不稳定

但是我用QiuSimons/YAO这位大佬的固件，基本稳定跑千兆。。。。。不知道为啥

jjm2473, 2023-04-06:
哦，可能 2月24号 升级了 r8152 驱动导致的

jjm2473, 2023-04-06:
试试2月24号之前的固件

jjm2473, 2023-04-07:
@HalcyonDaysssss 老固件恢复出厂设置再试试

jjm2473, 2023-04-15:
已发布

closed 2023-04-15 by jjm2473

HoGarfield, 2024-02-15:
iStoreOS 22.03.6 2024011910
千兆宽带，旁路由模式。内网测速能跑千兆，但是电信测速就只能跑600M左右，假如不走软路由，电信测速就能跑满千兆。