istoreos/istoreos #691: F3322 DDNS无法更新

tommytianhm, 2023-04-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：f3322 DDNS 永不更新。

### (2) 路由器型号和固件版本

A：X86硬路由（非虚拟机），固件版本，iStoreOS 21.02.3 2023033111

### (3) 详细日志和/或截图

A： 064203       : force interval: 4320 seconds
 064203       : retry interval: 60 seconds
 064203       : retry counter : 0 times
 064203       : 'SIGTERM' was send to old process
 064203  note : PID '18023' terminated by 'SIGTERM' at 2023-04-04 06:42
 064203       : last update: never
 064203       : Detect registered/public IP
 064203       : #> /usr/bin/host -t A tommyxxx.f3322.net  >/var/run/ddns/myddns_ipv4.dat 2>/var/run/ddns/myddns_ipv4.err
 064203       : Registered IP '113.116.73.23' detected
 064203  info : Starting main loop at 2023-04-04 06:42
 064203       : Detect local IP on 'network'
 064203       : Local IP '113.92.95.134' detected on network 'wan'
 064203       : Update needed - L: '113.92.95.134' <> R: '113.116.73.23'
 064203       : Force communication via IP '113.92.95.134'
 064203       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err --bind-address=113.92.95.134 --no-proxy 'http://tommyxxx:***PW***@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134'
 064203 ERROR : GNU Wget Error: '8'
 064203       : http://tommyxxx:*password*@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134:
2023-04-04 06:42:03 ERROR 404: Not Found.
 064203  WARN : Transfer failed - retry 1/0 in 60 seconds
 064303       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err --bind-address=113.92.95.134 --no-proxy 'http://tommyxxx:***PW***@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134'
 064303 ERROR : GNU Wget Error: '8'
 064303       : http://tommyxxx:*password*@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134:
2023-04-04 06:43:03 ERROR 404: Not Found.
 064303  WARN : Transfer failed - retry 2/0 in 60 seconds
 064403       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err --bind-address=113.92.95.134 --no-proxy 'http://tommyxxx:***PW***@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134'
 064403 ERROR : GNU Wget Error: '8'
 064403       : http://tommyxxx:*password*@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134:
2023-04-04 06:44:03 ERROR 404: Not Found.
 064403  WARN : Transfer failed - retry 3/0 in 60 seconds
 064503       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err --bind-address=113.92.95.134 --no-proxy 'http://tommyxxx:***PW***@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134'
 064503 ERROR : GNU Wget Error: '8'
 064503       : http://tommyxxx:*password*@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134:
2023-04-04 06:45:03 ERROR 404: Not Found.
 064503  WARN : Transfer failed - retry 4/0 in 60 seconds
 064603       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err --bind-address=113.92.95.134 --no-proxy 'http://tommyxxx:***PW***@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134'
 064603 ERROR : GNU Wget Error: '8'
 064603       : http://tommyxxx:*password*@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134:
2023-04-04 06:46:03 ERROR 404: Not Found.
 064603  WARN : Transfer failed - retry 5/0 in 60 seconds
 064703       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err --bind-address=113.92.95.134 --no-proxy 'http://tommyxxx:***PW***@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134'
 064703 ERROR : GNU Wget Error: '8'
 064703       : http://tommyxxx:*password*@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134:
2023-04-04 06:47:03 ERROR 404: Not Found.
 064703  WARN : Transfer failed - retry 6/0 in 60 seconds
 064803       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err --bind-address=113.92.95.134 --no-proxy 'http://tommyxxx:***PW***@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134'
 064803 ERROR : GNU Wget Error: '8'
 064803       : http://tommyxxx:*password*@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134:
2023-04-04 06:48:03 ERROR 404: Not Found.
 064803  WARN : Transfer failed - retry 7/0 in 60 seconds
[... 176 lines cut ...]
 071630       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err --bind-address=113.92.95.134 --no-proxy 'http://tommyxxx:***PW***@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134'
 071630 ERROR : GNU Wget Error: '8'
 071630       : http://tommyxxx:*password*@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134:
2023-04-04 07:16:30 ERROR 404: Not Found.
 071630  WARN : Transfer failed - retry 6/0 in 60 seconds
 071730       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err --bind-address=113.92.95.134 --no-proxy 'http://tommyxxx:***PW***@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134'
 071730 ERROR : GNU Wget Error: '8'
 071730       : http://tommyxxx:*password*@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134:
2023-04-04 07:17:30 ERROR 404: Not Found.
 071730  WARN : Transfer failed - retry 7/0 in 60 seconds
 071830       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err --bind-address=113.92.95.134 --no-proxy 'http://tommyxxx:***PW***@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134'
 071830 ERROR : GNU Wget Error: '8'
 071830       : http://tommyxxx:*password*@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134:
2023-04-04 07:18:30 ERROR 404: Not Found.
 071830  WARN : Transfer failed - retry 8/0 in 60 seconds
 071930       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err --bind-address=113.92.95.134 --no-proxy 'http://tommyxxx:***PW***@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134'
 071930 ERROR : GNU Wget Error: '8'
 071930       : http://tommyxxx:*password*@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134:
2023-04-04 07:19:30 ERROR 404: Not Found.
 071930  WARN : Transfer failed - retry 9/0 in 60 seconds
 072030       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err --bind-address=113.92.95.134 --no-proxy 'http://tommyxxx:***PW***@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134'
 072030 ERROR : GNU Wget Error: '8'
 072030       : http://tommyxxx:*password*@members.3322.org/dyndns/update?system=dyndns&hostname=tommyxxx.f3322.net&myip=113.92.95.134:
2023-04-04 07:20:30 ERROR 404: Not Found.
 072030  WARN : Transfer failed - retry 10/0 in 60 seconds

jjm2473, 2023-04-06:
没看到404吗？

tommytianhm, 2023-04-06:
> 没看到404吗？

自己摸索出来解决方案了。
固件默认配置的f3322的ddns服务器地址为f3322.org,实际连不上，所以404.
把配置文件里的f3322.org改为f3322.net就可以了。

jjm2473, 2023-04-07:
可能根本就不是同一家

closed 2023-04-07 by jjm2473