istoreos/istoreos #696: R2S的lan口经常掉线

jx1949, 2023-04-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
已经发生过两次了，都是无缘武功lan口掉线，灯灭，重启又好了
### (1) 具体问题

A：

### (2) 路由器型号和固件版本
FriendlyElec NanoPi R2S---------iStoreOS 21.02.3 2023031713 / LuCI istoreos-21.02 branch git-23.037.57600-91128de
A：

### (3) 详细日志和/或截图

A：

jjm2473, 2023-04-06:
升级到最新固件看看有没有问题。
如果有问题，那再试试 20230310 固件有没有问题。
如果还有问题，那就没法修复了，r2s的USB就是不稳定。

Ardin963, 2023-04-06:
跟你一样的问题，我的也是不知道从哪个固件开始，就Lan口每天必掉一次线

jx1949, 2023-04-06:
> 升级到最新固件看看有没有问题。 如果有问题，那再试试 20230310 固件有没有问题。 如果还有问题，那就没法修复了，r2s的USB就是不稳定。
之前用minimal的固件没有出现这个情况，会不会是哪里不兼容，因为我发现istoreOS对SD卡也有要求，之前那张卡无法重启，后来换了张卡就可以重启了，那这个lan口掉线会不会是驱动哪里的问题。。

jx1949, 2023-04-06:
> 跟你一样的问题，我的也是不知道从哪个固件开始，就Lan口每天必掉一次线

我是差不多两天，之前的固件没有这种情况

jjm2473, 2023-04-07:
可能是升级r8152驱动导致的，使用 20230224 之前的固件应该就没问题

colindog, 2023-04-08:
R2S如果没有特别的需要 建议用 221216或者230203

jjm2473, 2023-04-12:
修复了r2s的多个问题，例如部分机器识别不了lan口，lan口掉线，速度慢，而且改了多核配置，实测稳定千兆下载，这是测试固件  https://we.tl/t-4z3Nhk9aMM

jjm2473, 2023-04-15:
已发布

closed 2023-04-15 by jjm2473

mycastlex, 2023-04-18:
> R2S如果没有特别的需要 建议用 221216或者230203

旁路由玩法，主路由开DHCP，指定旁路由网关和DNS，也碰到lan口经常断掉的问题，本来没发现软路由对游戏的延迟会有影响，最近pad连wifi玩原神时蓝牙手柄在2.4g下输入延迟过高，体验非常不好，于是切到5g频段下后手柄输入延迟基本消失，然后我就发现直连主路由和通过软路由时的原神游戏延迟（非输入延迟）会有不同，而且延迟经常从两位数跳到三位数，于是各种给路由器调wifi最佳信道，刷wifi固件，都没能解决。昨天把R2S固件刷到230414后虽然lan口没有掉线但是原神依旧不稳定，而且是在直连路由的时候不稳定，推测是软路由导致的问题 （不严谨），于是尝试着刷到221216后瞬间延迟稳如老狗，连软路由时的延迟反而更低了，（这点我先保留，等用几天一直这么稳定的话再下结论，假如换软路由固件后原神依旧不稳那么我再试试直接拔掉软路由的效果。）

感谢提醒换221216！ @colindog

jjm2473, 2023-04-25:
@mycastlex 230421还有这个问题？

jx1949, 2023-04-25:
> @mycastlex 230421还有这个问题？

还没有测试0421。
之前用了那个测试版是有问题的，也会掉线，后来在线升级了0417，用了几天好像没问题，但不敢保证，因为我设置了每周一自动重启，所以用的天数也不多，等这个星期过完，就知道0417有无问题。

jx1949, 2023-04-25:
> > @mycastlex 230421还有这个问题？
> 
> 还没有测试0421。
> 之前用了那个测试版是有问题的，也会掉线，后来在线升级了0417，用了几天好像没问题，但不敢保证，因为我设置了每周一自动重启，所以用的天数也不多，等这个星期过完，就知道0417有无问题。

说错了是041417

mycastlex, 2023-04-25:
> > > @mycastlex 230421还有这个问题？
> > 
> > 
> > 还没有测试0421。
> > 之前用了那个测试版是有问题的，也会掉线，后来在线升级了0417，用了几天好像没问题，但不敢保证，因为我设置了每周一自动重启，所以用的天数也不多，等这个星期过完，就知道0417有无问题。
> 
> 说错了是041417

对，测试版是会掉lan口的

mycastlex, 2023-04-25:
> @mycastlex 230421还有这个问题？
230421，看样子这回是正常了，原神在wifi直连主路由下ping值比连旁路由低，5G-1 延迟比5G-2低（游戏中同一条传送点路线来回测试了好几遍）。用221216虽然不断lan口，但是原神测试时连旁路由延迟更低，5G-1延迟比5G-2低，应该是不正常的。

接下来就看稳定性如何了，用一个星期再来更新。

主路由：华硕RT-AC5300, 大佬的固件386.7_2
旁路由：R2S, 固件230421

@jjm2473 @jx1949

mycastlex, 2023-04-25:
> @mycastlex 230421还有这个问题？

230421，看样子这回是正常了，原神在wifi直连主路由下ping值比连旁路由低，5G-1 延迟比5G-2低（游戏中同一条传送点路线来回测试了好几遍）。用221216虽然不断lan口，但是原神测试时连旁路由延迟更低，5G-1延迟比5G-2高，应该是不正常的。

接下来就看稳定性如何了，用一个星期再来更新。

主路由：华硕RT-AC5300, 大佬的固件386.7_2
旁路由：R2S, 固件230421

@jjm2473 @jx1949

jx1949, 2023-04-28:
> > @mycastlex 230421还有这个问题？
> 
> 230421，看样子这回是正常了，原神在wifi直连主路由下ping值比连旁路由低，5G-1 延迟比5G-2低（游戏中同一条传送点路线来回测试了好几遍）。用221216虽然不断lan口，但是原神测试时连旁路由延迟更低，5G-1延迟比5G-2高，应该是不正常的。
> 
> 接下来就看稳定性如何了，用一个星期再来更新。
> 
> 主路由：华硕RT-AC5300, 大佬的固件386.7_2 旁路由：R2S, 固件230421
@jjm2473 @mycastlex 
刚才lan口掉了，固件是在线升级到的041417，大概用了三四天，现在准备升级最新版试试

jx1949, 2023-05-01:
> > @mycastlex 230421还有这个问题？
> 
> 230421，看样子这回是正常了，原神在wifi直连主路由下ping值比连旁路由低，5G-1 延迟比5G-2低（游戏中同一条传送点路线来回测试了好几遍）。用221216虽然不断lan口，但是原神测试时连旁路由延迟更低，5G-1延迟比5G-2高，应该是不正常的。
> 
> 接下来就看稳定性如何了，用一个星期再来更新。
> 
> 主路由：华硕RT-AC5300, 大佬的固件386.7_2 旁路由：R2S, 固件230421
> 
> @jjm2473 @jx1949

R2s问题太多了...
搜多一个回答，但不知怎么弄
“ 看一下 `dmesg` 里面有没有 `lan0` 的内核错误信息，我遇到过这种问题，需要禁用 `lan0` 的 tx offload ，否则会不定期 `lan0` 接口离线**”

jjm2473, 2023-05-04:
@jx1949 都说了最新固件已经修复了，现在是开了tx offload的，也不会出现内核错误。

jx1949, 2023-05-04:
> 日期： 2023年5月4日周四 11:10
> 主 题： Re: [istoreos/istoreos] R2S的lan口经常掉线 (Issue #696)

那不知道为啥了，我这还是掉线了，这几天换回了官方固件，如果同样出现，那就是这个机器有问题了

mycastlex, 2023-05-04:
我来分享一下最近一个多星期最新固件的使用体验啦：
lan口不再掉线，游戏ping值也稳定，R2S作为旁路由访问外网，主路由指定R2S为网关并开启DHCP分配IP。
其实这样设置已经让R2S接管家里的所有设备了，没办法，家里平板手机电视电脑都需要访问外网，一台一台设置IP太麻烦。

主路由：华硕RT-AC5300, 大佬的固件386.7_2 旁路由：R2S, 固件230421
@jjm2473 @jx1949

mycastlex, 2023-06-14:
@jx1949 @jjm2473 我好像发现你掉lan口的原因可能是出在ssrplus+或者passwall上了，更新4211用are-u-ok的离线包安装用shadowsocks是不会掉线的，但是手动升级ssrplus后会掉，内核原因有些依赖也装不上（有可能装上了也会掉），用passwall或者ssrplus里面的xray+ss协议也会掉（我没试过别的vless,vmess协议等，大家可以试试）。还有r2s做旁路由时重启主路由偶尔也会导致r2s掉lan口。

mycastlex, 2023-06-14:
> > @jx1949 @jjm2473 我好像发现你掉lan口的原因可能是出在ssrplus+或者passwall上了，更新4211用are-u-ok的离线包安装用shadowsocks是不会掉线的，但是手动升级ssrplus后会掉，内核原因有些依赖也装不上（有可能装上了也会掉），用passwall或者ssrplus里面的xray+ss协议也会掉（我没试过别的vless,vmess协议等，大家可以试试）。还有r2s做旁路由时重启主路由偶尔也会导致r2s掉lan口。
> 
> **你好，我最近很久没折腾软路由了，该弄的都已经学会了。关于R2s, 目前发现：在不合适的系统、内存卡搭配下，存在重启失效、重启丢失配置、Lan口经常掉线三个严重的问题. 需要说明的是我一直用的是openclash。IStoreOS 界面很漂亮，但是我一直存在lan掉线问题，试过多个版本，一直无法解决。所以一直在寻找一个稳定好用的系统，直到上次将系统换成了友善官方系统，安装了openclash, 一切正常，体验效果和第三方没有任何差异，已经用了差不多一个多月了吧，都没出现任何问题，也不会再去折腾了** 下面是友善系统截图，自建VPS，外挂了硬盘，路由器是华为Q6, R2s设定是每周一凌晨自动重启。运行稳定，现在觉得R2s还挺完美的！ [image]

看起来是ss软件和istoreOS的兼容性问题比较可能

jjm2473, 2023-06-14:
不是，之前USB的驱动有问题，在部分R2S上甚至会识别不了lan口（我手上这台就是），后来更新USB驱动以后就没问题了。

mycastlex, 2023-06-15:
> 不是，之前USB的驱动有问题，在部分R2S上甚至会识别不了lan口（我手上这台就是），后来更新USB驱动以后就没问题了。

按照我上面的描述，4211更新USB驱动后的版本除了用脚本安装的ssrplus 用new ss version是稳定的，其他情况下用xray或者v2ray下的ss协议都会掉lan（大概一天左右就掉），甚至我主路由重启时偶尔都会掉，我试着换过4211后的所有固件，都有这个问题，因此我才又回复了一下上次那位朋友出现的问题。那位 @jx1949 用的openclash，我这边由于服务器装的只有ss，也不能测试别的协议，所以我也只能推测是ss的兼容性问题了。

jjm2473, 2023-06-15:
一般应用层的错误不会导致内核的问题，除非内存泄漏导致整个系统的内存被占用光，有些核心的进程也被杀了。

mycastlex, 2023-06-15:
> 一般应用层的错误不会导致内核的问题，除非内存泄漏导致整个系统的内存被占用光，有些核心的进程也被杀了。

但是确实我这就只有脚本装的ssrplus里用new ss version时才不会掉，刷机刷了好多遍了0421后的版本也都用过了。
反正新版本的ssrplus编译后直接后台上传更新脚本装的也会出问题。
我再强调一下哦，出问题的时候基本上是在用了比如ssrplus或者passwall里面的v2ray or xray但是是用的ss的协议，非vmess vless or trojan等等。

mycastlex, 2023-06-22:
@jx1949 謝謝你的回復，我也試了一下友善的固件，只用ss的話確實很穩定，passwall和ssrplus最新固件都可以裝上。iStoreOS刷機比較方便一些，功能也多，以後有大更新的時候再試試。
請教一下你的R2S定時重啟用的是哪個插件？

jx1949, 2023-06-22:
> @jx1949 謝謝你的回復，我也試了一下友善的固件，只用ss的話確實很穩定，passwall和ssrplus最新固件都可以裝上。iStoreOS刷機比較方便一些，功能也多，以後有大更新的時候再試試。 請教一下你的R2S定時重啟用的是哪個插件？

对吧，官方固件超级稳定，好久都没动过了，我看了下重启用的是luci-app-autoreboot

mycastlex, 2023-06-22:
> > @jx1949 謝謝你的回復，我也試了一下友善的固件，只用ss的話確實很穩定，passwall和ssrplus最新固件都可以裝上。iStoreOS刷機比較方便一些，功能也多，以後有大更新的時候再試試。 請教一下你的R2S定時重啟用的是哪個插件？
> 
> 对吧，官方固件超级稳定，好久都没动过了，我看了下重启用的是luci-app-autoreboot

試了一下無法重啟，查了一下說是sd卡的問題，手上兩張都不行，三星Endurance U1和SandDisk A1，兩張應該都是U1，算了下次買卡的時候再說，反正不死機也沒必要一直重啟。你的SD卡是什麼類型的？

jx1949, 2023-06-22:
> > > @jx1949 謝謝你的回復，我也試了一下友善的固件，只用ss的話確實很穩定，passwall和ssrplus最新固件都可以裝上。iStoreOS刷機比較方便一些，功能也多，以後有大更新的時候再試試。 請教一下你的R2S定時重啟用的是哪個插件？
> > 
> > 
> > 对吧，官方固件超级稳定，好久都没动过了，我看了下重启用的是luci-app-autoreboot
> 
> 試了一下無法重啟，查了一下說是sd卡的問題，手上兩張都不行，三星Endurance U1和SandDisk A1，兩張應該都是U1，算了下次買卡的時候再說，反正不死機也沒必要一直重啟。你的SD卡是什麼類型的？

嗯，不能重启很正常，系统和卡的问题都有可能，我用的是官方推荐的  川宇 8G手机内存卡 TF 8G 卡存储卡 C10 高速 Class10 microSD卡

mycastlex, 2023-06-22:
> 嗯，不能重启很正常，系统和卡的问题都有可能，我用的是官方推荐的 川宇 8G手机内存卡 TF 8G 卡存储卡 C10 高速 Class10 microSD卡

ok，謝謝

simplysexyboy, 2023-08-15:
我也晕倒这个问题 谁知道什么问题呢？

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.176 (build@buildserver) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 2023070614) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Thu Aug 3 02:21:10 2023
[    0.000000] Machine model: FriendlyElec NanoPi R2S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] On node 0 totalpages: 261632
[    0.000000]   DMA zone: 4088 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 261632 pages, LIFO batch:63
[    0.000000] cma: Reserved 8 MiB at 0x000000003f400000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 21 pages/cpu s47064 r8192 d30760 u86016
[    0.000000] pcpu-alloc: s47064 r8192 d30760 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1001708K/1046528K available (8638K kernel code, 2404K rwdata, 2424K rodata, 1728K init, 734K bss, 36628K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000009] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001403] Console: colour dummy device 80x25
[    0.001876] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.002824] pid_max: default: 32768 minimum: 301
[    0.003485] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004167] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.007398] rcu: Hierarchical SRCU implementation.
[    0.008192] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.009916] smp: Bringing up secondary CPUs ...
[    0.011100] Detected VIPT I-cache on CPU1
[    0.011204] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.012039] Detected VIPT I-cache on CPU2
[    0.012114] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.012954] Detected VIPT I-cache on CPU3
[    0.013027] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.013164] smp: Brought up 1 node, 4 CPUs
[    0.016391] SMP: Total of 4 processors activated.
[    0.016823] CPU features: detected: 32-bit EL0 Support
[    0.017296] CPU features: detected: CRC32 instructions
[    0.018113] CPU: All CPU(s) started at EL2
[    0.018598] alternatives: patching kernel code
[    0.030321] KASLR disabled due to lack of seed
[... 444 lines cut ...]
[   23.125057] br-lan: port 1(eth1) entered forwarding state
[   23.204507] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:00] driver [RTL8211E Gigabit Ethernet] (irq=POLL)
[   23.205460] rk_gmac-dwmac ff540000.ethernet: Set MAC tx_delay: 0x24 rx_delay: 0x18 for PHY 0x001cc915
[   23.213531] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[   23.214252] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[   23.214868] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[   24.171340] br-lan: port 1(eth1) entered disabled state
[   26.334200] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   26.335060] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   26.832995] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[   26.834107] r8152 4-1:1.0 eth1: carrier on
[   26.835272] br-lan: port 1(eth1) entered blocking state
[... cut ...]
[  103.531864] overlayfs: upperdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  103.533182] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  103.534359] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  103.535522] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.

jx1949, 2023-08-15:
> 我也晕倒这个问题 谁知道什么问题呢？
> 
> [ 0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034] [ 0.000000] Linux version 5.10.176 (build@buildserver) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 2023070614) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Thu Aug 3 02:21:10 2023 [ 0.000000] Machine model: FriendlyElec NanoPi R2S [ 0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '') [ 0.000000] printk: bootconsole [uart8250] enabled [ 0.000000] Zone ranges: [ 0.000000] DMA [mem 0x0000000000200000-0x000000003fffffff] [ 0.000000] DMA32 empty [ 0.000000] Normal empty [ 0.000000] Movable zone start for each node [ 0.000000] Early memory node ranges [ 0.000000] node 0: [mem 0x0000000000200000-0x000000003fffffff] [ 0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000003fffffff] [ 0.000000] On node 0 totalpages: 261632 [ 0.000000] DMA zone: 4088 pages used for memmap [ 0.000000] DMA zone: 0 pages reserved [ 0.000000] DMA zone: 261632 pages, LIFO batch:63 [ 0.000000] cma: Reserved 8 MiB at 0x000000003f400000 [ 0.000000] psci: probing for conduit method from DT. [ 0.000000] psci: PSCIv1.1 detected in firmware. [ 0.000000] psci: Using standard PSCI v0.2 function IDs [ 0.000000] psci: MIGRATE_INFO_TYPE not supported. [ 0.000000] psci: SMC Calling Convention v1.2 [ 0.000000] percpu: Embedded 21 pages/cpu s47064 r8192 d30760 u86016 [ 0.000000] pcpu-alloc: s47064 r8192 d30760 u86016 alloc=21_4096 [ 0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [ 0.000000] Detected VIPT I-cache on CPU0 [ 0.000000] CPU features: detected: ARM erratum 845719 [ 0.000000] Built 1 zonelists, mobility grouping on. Total pages: 257544 [ 0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait [ 0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear) [ 0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear) [ 0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off [ 0.000000] Memory: 1001708K/1046528K available (8638K kernel code, 2404K rwdata, 2424K rodata, 1728K init, 734K bss, 36628K reserved, 8192K cma-reserved) [ 0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1 [ 0.000000] rcu: Preemptible hierarchical RCU implementation. [ 0.000000] rcu: RCU event tracing is enabled. [ 0.000000] rcu: RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4. [ 0.000000] Trampoline variant of Tasks RCU enabled. [ 0.000000] Tracing variant of Tasks RCU enabled. [ 0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies. [ 0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4 [ 0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0 [ 0.000000] GIC: Using split EOI/Deactivate mode [ 0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys). [ 0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns [ 0.000009] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns [ 0.001403] Console: colour dummy device 80x25 [ 0.001876] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000) [ 0.002824] pid_max: default: 32768 minimum: 301 [ 0.003485] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear) [ 0.004167] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear) [ 0.007398] rcu: Hierarchical SRCU implementation. [ 0.008192] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build [ 0.009916] smp: Bringing up secondary CPUs ... [ 0.011100] Detected VIPT I-cache on CPU1 [ 0.011204] CPU1: Booted secondary processor 0x0000000001 [0x410fd034] [ 0.012039] Detected VIPT I-cache on CPU2 [ 0.012114] CPU2: Booted secondary processor 0x0000000002 [0x410fd034] [ 0.012954] Detected VIPT I-cache on CPU3 [ 0.013027] CPU3: Booted secondary processor 0x0000000003 [0x410fd034] [ 0.013164] smp: Brought up 1 node, 4 CPUs [ 0.016391] SMP: Total of 4 processors activated. [ 0.016823] CPU features: detected: 32-bit EL0 Support [ 0.017296] CPU features: detected: CRC32 instructions [ 0.018113] CPU: All CPU(s) started at EL2 [ 0.018598] alternatives: patching kernel code [ 0.030321] KASLR disabled due to lack of seed [ 0.030967] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns [ 0.031875] futex hash table entries: 1024 (order: 4, 65536 bytes, linear) [ 0.033159] pinctrl core: initialized pinctrl subsystem [ 0.035560] NET: Registered protocol family 16 [ 0.038838] DMA: preallocated 128 KiB GFP_KERNEL pool for atomic allocations [ 0.039689] DMA: preallocated 128 KiB GFP_KERNEL|GFP_DMA pool for atomic allocations [ 0.040599] DMA: preallocated 128 KiB GFP_KERNEL|GFP_DMA32 pool for atomic allocations [ 0.042461] thermal_sys: Registered thermal governor 'step_wise' [ 0.042473] thermal_sys: Registered thermal governor 'power_allocator' [ 0.043477] cpuidle: using governor menu [ 0.044566] ASID allocator initialised with 65536 entries [ 0.045220] Serial: AMBA PL011 UART driver [ 0.077392] HugeTLB registered 1.00 GiB page size, pre-allocated 0 pages [ 0.078018] HugeTLB registered 32.0 MiB page size, pre-allocated 0 pages [ 0.078625] HugeTLB registered 2.00 MiB page size, pre-allocated 0 pages [ 0.079239] HugeTLB registered 64.0 KiB page size, pre-allocated 0 pages [ 0.082060] cryptd: max_cpu_qlen set to 1000 [ 0.084877] reg-fixed-voltage sdmm [... cut ...]  in undefined behavior. [ 103.533182] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior. [ 103.534359] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior. [ 103.535522] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.

请看之前回复，我这很久没管了，已经稳定运行好几个月了

simplysexyboy, 2023-08-15:
我看了 你换了sd卡好了么？

我买了sandisk C10 高速 Class10 32g 
看看会怎么样。。

jjm2473, 2023-08-15:
应该跟sd卡无关，sd卡最多就是导致重启设备失败，应该也修复了

simplysexyboy, 2023-08-15:
那你怎么解决的：D

jjm2473, 2023-08-15:
换网线试试

wuming5622, 2024-11-05:
友善官方系统确实能够解决掉线的问题，我已经稳定运行了好几个月，有问题直接刷成官方系统即可

yalunkong, 2025-01-12:
iStoreOS 22.03.7 2024120615
lan口掉线，拔电源才行，重启lan口都无效