istoreos/istoreos #703: R68S在使用tproxy模式时出错，提示 RTNETLINK answers: Address family not supported by protocol。

popalanluck, 2023-04-08:
### (1) 具体问题

A：r68s刷任意版本的istoreos，在shellclash使用tproxy模式时，都会出错。
提示为：RTNETLINK answers: Address family not supported by protocol
               RTNETLINK answers: File exists

执行命令  ip -4 rule，同样出错
RTNETLINK answers: Address family not supported by protocol
Dump terminated

执行命令 ip -6 rule则正常

   **据说是固件编译的问题，需要先配置
    IP_MULTIPLE_TABLES=y**

r68s刷同样为4.19内核的lede固件，完全正常。希望istoreos能尽快解决这个问题。

### (2) 路由器型号和固件版本

A： 路由器型号为 r68s   
       固件版本  iStoreOS 21.02.3 2023040712

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2023-04-11:
试试这个固件有没有问题 https://we.tl/t-F6F84Jyz6O

popalanluck, 2023-04-11:
@jjm2473 **非常感谢您。R68S刷入新固件后，ip  route和tproxy模式都能正常使用。此问题已经解决。**

[image]

jjm2473, 2023-04-12:
好，感谢反馈

jjm2473, 2023-04-15:
发布了

closed 2023-04-15 by jjm2473