istoreos/istoreos #716: 22.03.4 固件测试反馈

jjm2473, 2023-04-15:
目前 x86 有 22.03.4 的测试固件，从这里 https://fw.koolcenter.com/iStoreOS/alpha/ 下载，文件名是 `istoreos-test-22.03.4-*-x86-64-generic-squashfs-combined*.img.gz` 。
测试固件支持了 N5105 的硬解，从iStore安装jellyfin插件，并且部署以后，在jellyfin的“控制台”-“播放”选择 QuickSync 转码器，剩下的硬解选项除了 AV1 以外都可以勾选上。

注意事项：
1. 测试固件尚未支持 `FullconeNAT` ，不要勾选。
2. 不要使用在线升级，不然会降级到线上正式版本。
3. 如果之前使用其他版本固件，刷机前备份，尽量不保留配置升级。
4. 发现有其他问题可以在这个issue跟帖，不要单独开issue

Eircodong, 2023-04-17:
测试固件 网易UU加速器无法使用

jjm2473, 2023-04-18:
> 测试固件 网易UU加速器无法使用

好像uu确实不支持nftables，也就是只能使用iptables。听说uu还不支持openwrt 22官方版本，在iStoreOS的测试固件应该也一样。

honestpony, 2023-04-19:
只支持5105嘛 同代的N6000可否开启硬解

jjm2473, 2023-04-20:
> 只支持5105嘛 同代的N6000可否开启硬解

需要调试，但我没设备。之后找用户远程调试看看

mengxunlingxue, 2023-04-20:
> > 只支持5105嘛 同代的N6000可否开启硬解
> 
> 需要调试，但我没设备。之后找用户远程调试看看

我可以测试下，我有n6005设备

jjm2473, 2023-04-20:
> > > 只支持5105嘛 同代的N6000可否开启硬解
> > 
> > 
> > 需要调试，但我没设备。之后找用户远程调试看看
> 
> 我可以测试下，我有n6005设备

那你私信我 jjm2473@gmail.com ，之后微信联系

Eircodong, 2023-04-20:
> > 测试固件 网易UU加速器无法使用
> 
> 好像uu确实不支持nftables，也就是只能使用iptables。听说uu还不支持openwrt 22官方版本，在iStoreOS的测试固件应该也一样。

是的 我用22.3.4 换用iptables就可以支持

jjm2473, 2023-04-20:
@Eircodong 试试 2023042012，nftables 加上了 tproxy 支持，dnsmasq也加上了nftset支持，至少passwall可以正常使用了，uu可能也行。

jjm2473, 2023-04-21:
> 只支持5105嘛 同代的N6000可否开启硬解

支持了

ly173d, 2023-04-22:
用万兆光sriov方式，提示需要开启pci_mmconfig，有解决方案吗？

jjm2473, 2023-04-22:
> 用万兆光sriov方式，提示需要开启pci_mmconfig，有解决方案吗？

不知道这是啥，虚拟机吗？

54fellow, 2023-04-22:
n4505支持吗？

Eircodong, 2023-04-22:
> @Eircodong 试试 2023042012，nftables 加上了 tproxy 支持，dnsmasq也加上了nftset支持，至少passwall可以正常使用了，uu可能也行。
好的，马上测试一下

Eircodong, 2023-04-22:
> 

已测试，还是无法加速，应该是跟nftables有关系 uu可能不兼容

jjm2473, 2023-04-22:
@Eircodong  应该可以同时加上iptables，nftables和iptables可以共存

jjm2473, 2023-04-22:
> n4505支持吗？

不知道，没机器测试。你刷机以后如果不支持，就把内核日志发出来看看。

Eircodong, 2023-04-22:
> @Eircodong 应该可以同时加上iptables，nftables和iptables可以共存

共存啊？ 我是用22.3.4+上iptables 就是不用这个nftables就没有这些兼容问题。如果可以也建议istore先考虑这种过渡。等插件全部兼容nftables再更新。

a447948814, 2023-04-23:
istoreos-21.02.3-2023042111-x86-64-squashfs-combined-efi.img.gz
这个uu也有问题。手机app检测不到主机，必须重启op才能识别，过一会又不识别了。。又要重启。。

Eircodong, 2023-04-23:
> istoreos-21.02.3-2023042111-x86-64-squashfs-combined-efi.img.gz 这个uu也有问题。手机app检测不到主机，必须重启op才能识别，过一会又不识别了。。又要重启。。

21.2的uu是正常的奥，你是做主路由还是旁路由？

a447948814, 2023-04-23:
我是旁路由。不管是官方的安装方法（直接在终端安装）还是istore商店安装  我都试了，uu的app都会时不时不显示主机设备，如果开了常用设备自动加速，有时候连上网的瞬间显示一下，马上就消失了。实际上主机的ip发现已经自动加速成功。后来发现只要重启op 打开app就能稳定显示主机设备。不清楚是uu的问题还是istore的问题。找到一个帖子您看看：https://www.right.com.cn/FORUM/forum.php?mod=viewthread&amp;tid=6429299
 

郑好
***@***.***

jjm2473, 2023-04-23:
@a447948814 看帖子uu官方早就修复了啊

a447948814, 2023-04-23:
故障相似  可能问题不同吧  不清楚是哪边的问题

a447948814, 2023-04-23:
我是旁路由，刚才晚上回家又试了下，发现，正常加速情况下，断网后uu主机设备消失，再次将主机联网，uu就不显示设备了，此时将op重启，即可在uu重新显示主机设备。

94486, 2023-04-24:
N6000核显转码似乎正常，但转码过程出现死机重启还没有排除是过热还是驱动问题。J4105核显转码似乎反而有点问题，跑不了。

a447948814, 2023-04-24:
哥  我说的是uu……我不用转码

jjm2473, 2023-04-24:
@a447948814 不是回复你，这是测试反馈

iso-lib, 2023-04-25:
FullconeNAT 开启不能联网，关闭则使用正常
主机硬件：J3455
主机系统：PVE7.4-3 内核5.15
虚拟环境：LXC
路由系统： IStoreOS 22.03.4_2023042012_x86/64

第一次启动后，设置PPPOE拨号，能获取运营商IP,包括IPV6也获取到了，但局域网电脑不能连接外部网络
怀疑没有DNS服务，直接PING 223.5.5.5也不通
IStoreOS内部，网络诊断PING 域名和IP都不通
感觉跟防火墙有关，进防火墙，看到FullconeNAT打勾，取消勾选，应用，一切正常。

不知道是不是因为我用LXC有关，没有测试虚拟机安装

ycy119744220, 2023-04-25:
我用21.03跑在LXC上面也是一样的，PPPoE能拨号，iStoreOS本身可以访问互联网，但是局域网内的所有电脑都不能正常访问，ping 223.5.5.5也不通，不知道什么原因。

ycy119744220, 2023-04-26:
> FullconeNAT 开启不能联网，关闭则使用正常 主机硬件：J3455 主机系统：PVE7.4-3 内核5.15 虚拟环境：LXC 路由系统： IStoreOS 22.03.4_2023042012_x86/64
> 
> 第一次启动后，设置PPPOE拨号，能获取运营商IP,包括IPV6也获取到了，但局域网电脑不能连接外部网络 怀疑没有DNS服务，直接PING 223.5.5.5也不通 IStoreOS内部，网络诊断PING 域名和IP都不通 感觉跟防火墙有关，进防火墙，看到FullconeNAT打勾，取消勾选，应用，一切正常。
> 
> ## 不知道是不是因为我用LXC有关，没有测试虚拟机安装
> 后续： 上网正常后，再次主动勾选FullconeNAT,应用后，网络也还是正常的 不过貌似没什么作用，NAT类型测试工具测试还是NAT3

老哥，你LXC有没有直通网口?能不能把配置发出来给我看看。

LopoaySyen, 2023-04-29:
N6000 安装这个固件后 qbit aria transmission 都无法运行

ycy119744220, 2023-04-30:
LXC下，FullconeNAT开启不能联网，关闭再开机则正常使用，重启之后仍需，关闭再开启操作。
LXC下，UPNP无法运行，直接报错。
`Sun Apr 30 19:12:07 2023 daemon.err miniupnpd[2992]: try_sendto(sock=6, len=469, dest=239.255.255.250:1900): sendto: No such device`

magitwit, 2023-04-30:
主机硬件：R86s-G3
主机系统：IStoreOS 22.03.4_2023042012_x86/64
直接物理安装再emmc上，遇到如下几个问题“
1.全新写盘后，FullconeNAT开启不能联网，关闭开关才能正常使用。
2.upnp服务会自动关闭，无法生效
3.MallonX驱动，sfp+接口，如果重启过交换机之后，不会重新握手，需要在交换机正确启动后，重启一次r86s才能重新握手。比较影响使用
  （这个问题可能是内置驱动的问题，提个小建议，iKuai 5.10最新固件，内置的同型号万兆网卡驱动并没有这个问题，可以考虑提取，并更新到那边固件内置的网卡驱动)

以下是日志：
问题2：
Sun Apr 30 16:11:39 2023 daemon.notice miniupnpd[13940]: shutting down MiniUPnPd
Sun Apr 30 16:11:48 2023 daemon.notice miniupnpd[20932]: perform_stun: 1 response out of 4 received
Sun Apr 30 16:11:48 2023 daemon.warn miniupnpd[20932]: STUN: ext interface pppoe-wan has now public IP address IP地址 but firewall filters incoming connections set by miniunnpd
Sun Apr 30 16:11:48 2023 daemon.warn miniupnpd[20932]: Check configuration of firewall on local machine and also on upstream router
Sun Apr 30 16:11:48 2023 daemon.warn miniupnpd[20932]: Port forwarding is now disabled

问题3：
iKuai 5.10内核固件里面是显示这个名字：Mellanox Technologies MT27500 Family [ConnectX-3]

希望能略尽绵力，510快点转正 希望小宝团队的iStoreOS越做越好

magitwit, 2023-05-02:
接上补充问题4

jjm2473, 2023-05-04:
@magitwit 

1. “FullconeNAT开启不能联网，关闭开关才能正常使用。”
  你说的是内网设备不能上网吗，还是软路由自己不能上网，我测试过FullconeNAT没问题的。
2. upnp的问题，试试在upnp的高级设置里，关掉stun，同时打开强制转发。
3. 驱动的话，我看看是不是正确
4. 应用过滤可能是因为没适配nftables，之后我再测试看看

magitwit, 2023-05-04:
@jjm2473 
回复再反馈,谢谢回复
1.就是全新裸跑LEDe，写盘刷机之后，表现跟楼下LXC同学反映的情况一样。默认开启防火墙FullconeNAT开关的话。
 a.软路由本身上网一切正常
 b.内网设备只能访问少部分优先v6的网址,163/qq。其他大部分网站都是。设备nslookup 能找到地址 不能上网
c.反馈下我的网络情况：广东电信，FTTH，桥接，双栈公网ipv4/v6

2.关掉stun，软件就能正常添加upnp内容了。强制转发是upnp的默认配置

3.再新反馈一个今早发现的bug，日志出现满屏如下错误，每5分钟一次 一直在刷屏（现在还在刷)
Thu May  4 11:32:42 2023 daemon.err nmbd[1937]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.19.1 for name WORKGROUP<1d>.
Thu May  4 11:32:42 2023 daemon.err nmbd[1937]:   This response was from IP 192.168.19.11, reporting an IP address of 192.168.19.11.
Thu May  4 11:37:33 2023 daemon.err nmbd[1937]: [2023/05/04 11:37:33.665848,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)

YiGeRen-yl, 2023-05-09:
旁路由下 网易UU不识别设备 怎么解决

jjkle, 2023-05-12:
n6000从正式版2023042111升级到测试版2023051216，/dev/dri下有驱动文件，但安装jellyfin后还是无法硬解

jjm2473, 2023-05-13:
@jjkle 终端里面执行下 `lspci -tv` ，把结果帖出来。还有执行 `dmesg | grep i915`，也把结果帖出来。

jjkle, 2023-05-13:
> @jjkle 终端里面执行下 `lspci -tv` ，把结果帖出来。还有执行 `dmesg | grep i915`，也把结果帖出来。

root@iStoreOS:~# lspci -tv
-[0000:00]-+-00.0  Intel Corporation Device 4e26
           +-02.0  Intel Corporation JasperLake [UHD Graphics]
           +-04.0  Intel Corporation Dynamic Tuning service
           +-14.0  Intel Corporation Device 4ded
           +-14.2  Intel Corporation Device 4def
           +-16.0  Intel Corporation Management Engine Interface
           +-17.0  Intel Corporation Device 4dd3
           +-1c.0-[01]----00.0  MAXIO Technology (Hangzhou) Ltd. Device 1602
           +-1c.3-[02]----00.0  Sandisk Corp WD Blue SN550 NVMe SSD
           +-1c.4-[03]----00.0  Intel Corporation Device 125c
           +-1c.5-[04]----00.0  Intel Corporation Device 125c
           +-1c.6-[05]----00.0  Intel Corporation Device 125c
           +-1c.7-[06]----00.0  Intel Corporation Device 125c
           +-1f.0  Intel Corporation Device 4d87
           +-1f.3  Intel Corporation Jasper Lake HD Audio
           +-1f.4  Intel Corporation Jasper Lake SMBus
           \-1f.5  Intel Corporation Jasper Lake SPI Controller
root@iStoreOS:~# dmesg | grep i915
[    7.332883] i915 0000:00:02.0: [drm] Incompatible option enable_guc=3 - GuC submission is N/A
[    7.345303] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/icl_dmc_ver1_09.bin (v1.9)
[    8.004080] i915 0000:00:02.0: [drm] failed to retrieve link info, disabling eDP
[    8.042073] i915 0000:00:02.0: [drm] GuC firmware i915/ehl_guc_33.0.4.bin version 33.0 submission:disabled
[    8.042079] i915 0000:00:02.0: [drm] HuC firmware i915/ehl_huc_9.0.0.bin version 9.0 authenticated:yes
[    8.043520] [drm] Initialized i915 1.6.0 20200917 for 0000:00:02.0 on minor 0
[    8.045584] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    8.045682] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes

jjm2473, 2023-05-13:
@jjkle 驱动应该没问题了，你的jellyfin是从istore里面安装的吗？

jjkle, 2023-05-13:
> @jjkle 驱动应该没问题了，你的jellyfin是从istore里面安装的吗？

是的，需要重新安装istoreos么

jjm2473, 2023-05-13:
@jjkle 那你安装完配置了jellyfin的硬解吗

jjkle, 2023-05-13:
> @jjkle 那你安装完配置了jellyfin的硬解吗
[image]
这样对么？
[image]

m0eak, 2023-05-13:
能增加fullconenat高性能模式吗

jjm2473, 2023-05-13:
@jjkle 看看jellyfin的版本号

jjm2473, 2023-05-13:
> 能增加fullconenat高性能模式吗

谁tm又造了个词，truboacc？flow-offload？

jjkle, 2023-05-13:
> @jjkle 看看jellyfin的版本号

服务器：iStoreOS

版本：10.7.7

操作系统：Linux

架构：X64

m0eak, 2023-05-13:
> > 能增加fullconenat高性能模式吗
> 
> 谁tm又造了个词，truboacc？flow-offload？

lede新编译的防火墙fullcone里的

jjm2473, 2023-05-13:
@jjkle 你这个jellyfin肯定是旧版本的。把docker配置里面的镜像仓库删掉，然后重新部署一遍，jellyfin起码得10.8

jjm2473, 2023-05-13:
@jjkle 10.7.7已经是两年前的版本了（ https://github.com/jellyfin/jellyfin/releases/tag/v10.7.7 ），谁知道是不是支持这个驱动

jjm2473, 2023-05-13:
@m0eak 那你测试了吗，有什么提升吗？

jjkle, 2023-05-13:
> @jjkle 你这个jellyfin肯定是旧版本的。把docker配置里面的镜像仓库删掉，然后重新部署一遍，jellyfin起码得10.8

容器镜像都删除后，自动拉取的latest，但版本是10.7.7

jjm2473, 2023-05-13:
@jjkle 都说了让你删掉“镜像仓库”配置，也就是mirror，因为有些mirror不会更新latest

jjm2473, 2023-05-13:
@jjkle  被翻译成了注册表镜像 
[image]

jjkle, 2023-05-13:
> @jjkle 被翻译成了注册表镜像 [image]

搞定了，删掉后拉取就是10.8.10了

weijie66, 2023-05-16:
此固件是否支持N5095硬解 谢谢

jjm2473, 2023-05-16:
@weijie66 你在终端执行 `lspci -tvnn` ，把结果帖出来看看

jjm2473, 2023-05-16:
@magitwit  试试这里的 https://fw.koolcenter.com/iStoreOS/alpha/x86/  2023051618 版本，还有没有问题。（应用过滤删掉了，因为官方master版本也无效，可能之后换成iptables才能用）

weijie66, 2023-05-16:
> 

[image]

jjm2473, 2023-05-17:
@weijie66 驱动应该支持的，可能没加载firmware，你再执行下`dmesg | grep i915`看看

[102 more replies]

closed 2023-07-07 by jjm2473