istoreos/istoreos #720: passwall  SSrPLUS  clash安装失败

liliang233, 2023-04-18:
安装passwall clash之类的软件出现这个报错，以前安装没有，但是ota之后就有问题了

主要目前很多机场都是只有ss的，但是明显ss无法正常安装

是不是新版本禁用了这些功能

安装PassWall_a53.run，报错
Error relocating /usr/bin/ss-local: mbedtls_cipher_setkey: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_finish: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_hmac: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_hmac_starts: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_reset: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_hmac_update: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md5_ret: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_starts: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_update: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_set_iv: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_get_size: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_init: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_setup: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_setup: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_free: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_info_from_string: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_hmac_finish: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_cipher_auth_encrypt: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_info_from_string: symbol not found
Error relocating /usr/bin/ss-local: mbedtls_md_init: symbol not found

系统版本
[image]

liliang233, 2023-04-18:
用了很多方法  把这个目录打包替换新装的系统里面都不行

liliang233, 2023-04-19:
d

> 谁还用ss啊，xray不支持这些协议吗，ss-redir呢？

虽然我自建的机场还是用vmess 但是奶昔的机场全是ss呀 😭

jjm2473, 2023-04-21:
我是说，用xray也可以支持ss的协议啊，你选择xray不就行了

closed 2023-05-18 by jjm2473