istoreos/istoreos #726: istoreos开启防火墙后设备无法访问外网ipv6

stjiji, 2023-04-20:
N4100软路由安装istoreos拨号上网能正常获取公网ipv6地址。
默认开启ipv6防火墙后，内网设备(手机，PC等)无法通过外网的ipv6测试。

请问这是正常的吗，需要在通信规则里自行配置从lan到wan的ipv6流量吗
[image]

[image]
[image]
[image]

jjm2473, 2023-04-21:
你的内网设备有获取到ipv6地址吗

stjiji, 2023-04-21:
> 你的内网设备有获取到IPv6地址吗

有的，内网设备都能获取到ipv6地址。

jjm2473, 2023-04-21:
那你在内网设备执行 `ping6 240c::6666` ，还有 `nslookup taobao.com 192.168.100.1` (192.168.100.1改成软路由地址)

jjm2473, 2023-04-21:
还有 `traceroute6 240c::6666`

stjiji, 2023-04-21:
> 那你在内网设备操作`ping6 240c::6666`，还有`nslookup taobao.com 192.168.100.1`(192.168.100.1改成软件地址由地址)

[image]
这是用内网群晖Ping后的结果，不知道能不能看出大概是哪里的问题。非常感谢您的回复。感激。

closed 2023-04-21 by stjiji

jjm2473, 2023-04-22:
看起来没问题啊，dns正常，路由也正常。除非你的内网设备的dns不是192.168.3.1？

stjiji, 2023-04-25:
> 看起没有问题啊，dns正常，路由也正常。除了你的内网设备的dns不是192.168.3.1？

非常感谢您的解答