istoreos/istoreos #729: 怎么使用终端配置端口转发呢

WuMing-dp, 2023-04-21:
已经配置好了ddns，之前在设置里配置好了一下基础的端口转发规则，现在可以使用域名进入终端，如图，**已经在/etc/config/firewall添加了新的端口转发规则**，但是不知道怎么让它生效，常用的Linux firewall命令无效，请问op的防火墙restart命令是什么，或者怎么使用终端配置端口转发。
[image]

然后我还修改了/etc/config/uhttpd 这个文件，想将op的web页面用8080端口打开，然后可以通过设置端口转发设置外网访问op的web页，但是修改重启后貌似还是80端口

[image]

jjm2473, 2023-04-21:
防火墙修改以后，执行 `/etc/init.d/firewall reload` 重新加载，其他插件也都类似。配置文件和脚本的对应关系一般是名称对应，如果不清楚，可以看看 `/etc/config/ucitrack` 里面的记录

jjm2473, 2023-04-21:
防火墙还支持加载外部脚本，可以自己写iptables命令，可能更方便你维护

closed 2023-04-21 by jjm2473