istoreos/istoreos #731: AMD5900hx物理机安装温度监控不显示

4538640, 2023-04-21:
反馈bug/问题模板，提建议请删除

1.用物理机5900hx安装istoreOS，温度监控不能正常显示

2. 详细叙述

(1) 具体问题

A：温度监控不显示

 (2) 路由器型号和固件版本

A：AMD5900hx 固件版本
iStoreOS 21.02.3 2023042111

(3) 详细日志和/或截图

[image]

4538640, 2023-04-21:
[image]
sensors可以返回温度

jjm2473, 2023-05-12:
你这个温度不是scsi的吗，也就是硬盘的吧？

4538640, 2023-05-12:
您发给我的信件已经收到。--------------------------------------------中文邮箱第一品牌——QQ免费邮 mail.qq.comThis is an automatic reply, confirming that your e-mail was received.Thank you

4538640, 2023-05-12:
这个是什么温度我也不太清楚，就是首页的温度监控无法显示cpu核心温度

terryzung, 2023-05-22:
AMD都不能显示温度，我也是这样

jjm2473, 2023-05-26:
你们执行这行命令看看，把结果复制出来 `(cd /sys/class/thermal ; ls | grep thermal_zone | xargs -rn1 sh -c 'echo -n "$0 "; cat $0/type | tr -d "\\n"; echo -n " " ; cat $0/temp')`

terryzung, 2023-05-26:
> 你们在终或ssh执行这行命令看看，把结果复制出来 `(cd /sys/class/thermal ; ls | grep thermal_zone | xargs -rn1 sh -c 'echo -n "$0 "; cat $0/type | tr -d "\\n"; echo -n " " ; cat $0/temp')`

执行了没有什么结果的

4538640, 2023-06-14:
> 

我也执行了没反应

jjm2473, 2023-06-15:
那就是amd机型都没在内核注册温度传感器

terryzung, 2023-06-15:
> 那就是amd机型都没在内核注册温度传感器

曾经有个一个版本显示过，很早前第二个版本。

4538640, 2023-06-15:
> 那就是amd机型都没在内核注册温度传感器

大神能在固件中添加amd处理器的温度显示插件吗

jjm2473, 2023-07-17:
没法支持，内核都没注册传感器给用户端。

closed 2023-07-17 by jjm2473