istoreos/istoreos #733: docker ps 命令执行报错

zhao624, 2023-04-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用 docker ps 查看容器状态和在luci上操作，均无法成功

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志和/或截图

A：
[image]
```
Sat Apr 22 21:25:00 2023 daemon.err dockerd[9252]: time="2023-04-22T13:25:00.294057474Z" level=error msg="Handler for GET /v1.41/containers/sonarqube/top returned error: ps: ps: unrecognized option: e"
Sat Apr 22 21:26:41 2023 daemon.err dockerd[9252]: time="2023-04-22T13:26:41.584958776Z" level=error msg="Handler for GET /containers/8dca6dfcfbe6c969dac27bb59e1d070aec1e034363b5fc30e30c972be63aabc9/top returned error: ps: ps: unrecognized option: a"
Sat Apr 22 21:26:41 2023 daemon.err dockerd[9252]: time="2023-04-22T13:26:41.593630661Z" level=error msg="Handler for GET /containers/8dca6dfcfbe6c969dac27bb59e1d070aec1e034363b5fc30e30c972be63aabc9/top returned error: ps: ps: unrecognized option: e"
```

jjm2473, 2023-05-12:
这是你的容器有问题，可能容器里面的ps不支持e参数，跟docker自身或者固件都无关。

closed 2023-05-12 by jjm2473

weiyideai520, 2024-12-09:
> 这是你的容器有问题，可能容器里面的ps不支持e参数，跟docker自身或者固件都无关。

我认为这不是容器的问题，因为相同的容器，在群晖NAS上执行时正常的。在iStoreOS上就报错了

jjm2473, 2024-12-17:
docker ps输出什么