istoreos/istoreos #736: iStoreos网页端的易有云文件管理器新建文件时间是错误的

zzj6116, 2023-04-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue
X
## 2. 详细叙述

### (1) 具体问题

A：iStoreos网页端的易有云文件管理器新建文件时间是错误的，已查，软路由系统时间是正确的，在iStoreos网页端首页自带的文件管理器，列表模式，新建文件时间是错误的，慢了大约8小时。易有云APP上显示的文件时间也是错误的，但是Windows端通过SMB挂载的istoreos的硬盘，同一个文件，时间又是正确的
### (2) 路由器型号和固件版本

A：X86，j4125, 固件版本iStoreOS 21.02.3 2023042111

### (3) 详细日志和/或截图

A：
[image]

[image]
[image]

Bugaoxingxx, 2023-05-05:
8个小时，要么检查下OS内的时区是不是UTC啥的？

jjm2473, 2023-05-12:
那应该是界面上显示的时区有问题

N1k1Zhang, 2023-12-24:
2023年12月24日，我也有这个同样的问题。

closed 2024-04-15 by jjm2473