istoreos/istoreos #737: 首页系统时间比本地时间快8小时

wcbspp, 2023-04-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：首页系统时间比本地时间快8小时

### (2) 路由器型号和固件版本

A：R4SE -21.02.3 2023042111

### (3) 详细日志和/或截图

A：
[image]

rootphantomer, 2023-04-24:
这种一般系统时区里没选择utc+8北京时区，显示的是utc时区

wcbspp, 2023-04-24:
> 这种一般系统时区里没选择utc+8北京时区，显示的是utc时区

好想没有utc➕8北京时区这个选项呢 现在选的是ASia/Shanghai

jjm2473, 2023-04-24:
那就是你的电脑时间不对啊，可能电脑设置成了UTC时区，然后却按北京时间设置，然后还同步到 R4SE 了，那不就错误了吗

wcbspp, 2023-04-24:
> 那就是你的电脑时间不对啊，可能电脑设置成了UTC时间区，然后却按北京时间设置，然后又同步到R4SE了，那不就错了吗

应该不是 电脑是UTC+8  istore os 在设置里是这样
**
[image]
**

jjm2473, 2023-04-24:
软路由重启以后还有这个问题吗？

wcbspp, 2023-04-24:
> 软路由重启以后还有这个问题吗？

是的 很早之前就有这个问题了 一些插件的日志时间都相差了八小时  opkg install zoneinfo-asia 之后其他插件的时间对了
后台看是这样
[image]

jjm2473, 2023-04-24:
跟date -R无关，首页实际是读取 `ubus call system info` 里面的时间轴，然后转换成字符串，转换过程在设备上，不在浏览器。

jjm2473, 2023-04-25:
不知道怎么解决，我也是用北京时间，没遇到过这个问题

wcbspp, 2023-04-27:
> 不知道怎么解决，我也是用北京时间，没遇到过这个问题
获取的时间戳确实比本地时间快了8小时 头痛
[image]
[image]

jjm2473, 2023-05-04:
我看了下自己的设备，localtime确实不是UTC时间轴，而是本地的时间轴，我看看后端那边能不能指定时区。

jjm2473, 2023-05-05:
修复了。

这个问题的复现方式就是 `opkg install zoneinfo-asia` ，然后重启。

后端改成UTC就不受影响了。

wcbspp, 2023-05-05:
> 修复了。
> 
> 这个问题的复现方式就是 `opkg install zoneinfo-asia` ，然后重启。
> 
> 后端改成UTC就不受影响了。

牛~ 需要等待新版本更新不

jjm2473, 2023-05-05:
如果你用了沙箱，我可以给你个ipk安装，以后升级固件退出沙箱就行。如果没用沙箱，那还是等固件更新吧，以免麻烦。

这是已经编译的quickstart，如果你有会用也可以自己尝试，但是如果你替换了系统里面的quickstart，之后固件更新就不会同步升级了（除非用沙箱）。  https://github.com/linkease/istore-packages/releases/download/prebuilt/quickstart-binary-0.7.6.tar.gz

wcbspp, 2023-05-05:
> 如果你用了沙箱，我可以给你个ipk安装，以后升级固件退出沙箱就行。如果没用沙箱，那还是等固件更新吧，以免麻烦。
> 
> 这是已经编译的quickstart，如果你有会用也可以自己尝试，但是如果你替换了系统里面的quickstart，之后固件更新就不会同步升级了（除非用沙箱）。 https://github.com/linkease/istore-packages/releases/download/prebuilt/quickstart-binary-0.7.6.tar.gz

非常感谢~！

closed 2023-05-12 by jjm2473

yuankan1986, 2023-11-17:
> 修复了。
> 
> 这个问题的复现方式就是 `opkg install zoneinfo-asia` ，然后重启。
> 
> 后端改成UTC就不受影响了。

怎么解决的，我也遇到了同样的问题。感谢。

jjm2473, 2023-11-19:
升级最新固件