istoreos/istoreos #744: 关闭ipv6操作后无法恢复

supertrf, 2023-04-27:
使用系统-->系统便利工具-->关闭ipv6操作后，想再恢复ipv6，重新启用了wan6接口，依然无法获取ipv6地址，请求帮助。谢谢！
另外有个建议是，是否可以考虑加一个一键恢复ipv6的便利工具？

如本小白对ipv6的工作方式理解有误，敬请谅解

thomas-affie, 2023-06-24:
与楼主同样问题，敢问楼主是否已经解决咧

supertrf, 2023-07-30:
> 

还没有解决。。。等待开发者注意到我们

jinxiao, 2023-08-04:
通过手动修改接口里面的配置，可以重新启动v6，就是有点麻烦

wolvth, 2023-08-06:
是的，我手痒关掉了，现在恢复不了，手动设置没启作用，关闭后要改动的项好像有点多，不想折腾了，怕越搞问题越多。
希望有个恢复的设置。

kid-oh, 2023-08-11:
wan口打开自动获取ipv6地址
lan口dhcp服务器里将RA 服务和DHCPv6 服务改为服务器模式
dns高级设置里把忽略ipv6解析的勾取消
最后把设备的网重新连接一下应该就恢复了

supertrf, 2023-09-07:
> wan口打开自动获取ipv6地址 lan口dhcp服务器里将RA 服务和DHCPv6 服务改为服务器模式 dns高级设置里把忽略ipv6解析的勾取消 最后把设备的网重新连接一下应该就恢复了

全部操作都试了一下，还是没恢复，wan口那边也没有看到自动获取ipv6的选项
[image]
[image]

Mr-XiaoYi, 2023-10-19:
[OpenWRT系统获取到Ipv6地址但不下发Ipv6地址解决过程](https://www.scczz.com/1056.html)
https://www.scczz.com/1056.html
关于ipv6的开关真的是有点多了，折腾起来可难受了
折腾一天找到的完整解决方案，希望帮到各位，有心的朋友点点赞赏😘

zwtocn, 2023-12-01:
我是能获取到ipv6地址，然后外网能够ping通这个ipv6地址，但是无法ping通外网地址。已经快一年了，找不到原因，各种方法用尽

zizhaolaisuimuqu, 2023-12-23:
同求