istoreos/istoreos #745: 求宝哥团队添加n7505核显驱动！！！谢谢

sosisxjf, 2023-04-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

sosisxjf, 2023-05-20:
> 开机以后执行 lspci -tvnn ，结果发出来看看

宝哥为啥我的固件不能安装关机插件提示Collected errors:
 * check_data_file_clashes: Package luci-i18n-poweroff-zh-cn wants to
install file /usr/lib/lua/luci/i18n/poweroff.zh-cn.lmo
        But that file is already provided by package  * luci-app-poweroff
 * opkg_install_cmd: Cannot install package app-meta-poweroff.

sosisxjf, 2023-05-21:
***@***.***:~# lspci -tvnn
-+-[0000:02]-+-00.0  VMware PVSCSI SCSI Controller [15ad:07c0]
 |           +-01.0  VMware USB1.1 UHCI Controller [15ad:0774]
 |           +-02.0  VMware VMXNET3 Ethernet Controller [15ad:07b0]
 |           +-03.0  VMware USB2 EHCI Controller [15ad:0770]
 |           +-04.0  VMware SATA AHCI controller [15ad:07e0]
 |           \-05.0  Intel Corporation Device [8086:9a78]
 \-[0000:00]-+-00.0  Intel Corporation 440BX/ZX/DX - 82443BX/ZX/DX Host
bridge [8086:7190]
             +-01.0-[01]--
             +-07.0  Intel Corporation 82371AB/EB/MB PIIX4 ISA [8086:7110]
             +-07.1  Intel Corporation 82371AB/EB/MB PIIX4 IDE [8086:7111]
             +-07.3  Intel Corporation 82371AB/EB/MB PIIX4 ACPI [8086:7113]
             \-07.7  VMware Virtual Machine Communication Interface
[15ad:0740]

jjm2473, 2023-07-12:
你这不是虚拟机吗，而且都没看到虚拟机里面有显卡

closed 2023-07-12 by jjm2473