istoreos/istoreos #747: R68s 平均负载

binfoo, 2023-04-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

binfoo, 2023-04-28:
2. 详细叙述
(1) 具体问题
A：R68S的状态，CPU平均负载达到10以上

(2) 路由器型号和固件版本
A：FastRhino R68s 型号；固件版本：iStoreOS 21.02.3 2023042111 / LuCI openwrt-21.02 branch git-23.093.57360-e98243e

(3) 详细日志和/或截图
A：截图如下
[image]
[image]

jjm2473, 2023-05-12:
看看是不是哪个进程死循环了。恢复出厂设置如果有问题再来反馈

closed 2023-05-12 by jjm2473