istoreos/istoreos #748: R68S 升级到iStoreOS 21.02.3 2023042111，恢复出厂设置后，网口显示仍然错误。

popalanluck, 2023-04-28:
## 1.关于你要提交的问题
Q：是否用关键词搜索了issue? (使用 "x" 选择)
 没有类似的issue

2. 详细叙述

 (1) 具体问题

A：
     R68S原物理网口顺序为：1023， 外网线插在eth1物理口。升级到iStoreOS 21.02.3 2023042111并恢复出厂设置后，外网口显示仍然是eth0，并不是实际插线的eth1。而上一版istoreos-21.02.3-2023041417是正确显示网口顺序的。

既然为了方便维护，物理 WAN 口统一映射到 eth0，能不能将r68s的物理网口顺序定义为： 0123，让逻辑WAN口与物理wan口保持一致？ 

### (2) 路由器型号和固件版本

A：            FastRhino R68s
固件版本   iStoreOS 21.02.3 2023042111 / LuCI istoreos-21.02 branch git-23.088.34636-f8cc111

### (3) 详细日志和/或截图
A：
[image]

jjm2473, 2023-05-12:
跟这个界面显示的无关，eth0是不是第一个网口，如果是就对了。

jjm2473, 2023-05-12:
这个界面的显示以后可能会按接口名称排序。不过跟我说的wan口统一eth0没有任何关系

jjm2473, 2023-05-16:
之后的固件首页会按eth0开始的顺序显示