istoreos/istoreos #749: 或许这是一个小bug，路由重启，断电自动开机就拨不上号

guugg, 2023-04-28:
* 类似的issue
#672   这是我上个月提的问题

## 2. 详细叙述

  上个月使用的N3160主板+R8125BG (4+2.5G)网卡 ，物理机安装`iStoreOS `。使用过程中出现断网拨不上号的问题，中间还找过电信刷新宽带几次，问运维是不是我重拨太多限制拨号了。后面尝试重启我的猫棒`MA5671A` 。 发现可以拨号了，就早早关闭了问题。  中间因为没有重启过机子一直稳定使用。

  今天把旧电脑`h61+i53570`做软路由。esxi6.7 主板网口做管理 + 四网口直通`iStoreOS` 。  当我设置好IPV6重启动就出现了。

```
Fri Apr 28 23:26:38 2023 daemon.notice netifd: Interface 'wan' is now down
Fri Apr 28 23:26:38 2023 daemon.notice netifd: Interface 'wan' is setting up now
Fri Apr 28 23:26:38 2023 
daemon.err insmod: module is already loaded - slhc
Fri Apr 28 23:26:38 2023 daemon.err insmod: module is already loaded - ppp_generic
Fri Apr 28 23:26:38 2023 daemon.err insmod: module is already loaded - pppox
Fri Apr 28 23:26:38 2023 daemon.err insmod: module is already loaded - pppoe
Fri Apr 28 23:26:38 2023 daemon.notice netifd: wan (10419): uci: Entry not found
Fri Apr 28 23:26:38 2023 daemon.info pppd[10436]: Plugin pppoe.so loaded.
Fri Apr 28 23:26:38 2023 daemon.info pppd[10436]: PPPoE plugin from pppd 2.4.9
Fri Apr 28 23:26:38 2023 daemon.notice pppd[10436]: pppd 2.4.9 started by root, uid 0
Fri Apr 28 23:26:53 2023 daemon.warn pppd[10436]: Timeout waiting for PADO packets
Fri Apr 28 23:26:53 2023 daemon.err pppd[10436]: Unable to complete PPPoE Discovery
```

### (1) 具体问题
>测试不同设置的重启。全是拨号测试 。  间隔很久不会频繁拨号
1.  新安装 -  重启偶尔会出现拨不上，重启猫棒就好
2. 新安装 +设置 ipv6 -   直接重启/3次就有1次拨不了

### (2) 路由器型号和固件版本

- X86 主机
- istoreos-21.02.3-2023040712-x86-64-squashfs-combined.img

### (3) 详细日志和/或截图

- 我就不截这么多了。都是重复的。  
- 下面的代码是我在拔插猫棒的过程。
```
Fri Apr 28 23:31:37 2023 daemon.info pppd[16031]: Exit.
Fri Apr 28 23:31:37 2023 daemon.notice netifd: Interface 'wan' is now down
Fri Apr 28 23:31:37 2023 daemon.notice netifd: Interface 'wan' is setting up now
Fri Apr 28 23:31:37 2023 daemon.err insmod: module is already loaded - slhc
Fri Apr 28 23:31:37 2023 daemon.err insmod: module is already loaded - ppp_generic
Fri Apr 28 23:31:37 2023 daemon.err insmod: module is already loaded - pppox
Fri Apr 28 23:31:37 2023 daemon.err insmod: module is already loaded - pppoe
Fri Apr 28 23:31:37 2023 daemon.notice netifd: wan (16365): uci: Entry not found
Fri Apr 28 23:31:37 2023 daemon.info pppd[16382]: Plugin pppoe.so loaded.
Fri Apr 28 23:31:37 2023 daemon.info pppd[16382]: PPPoE plugin from pppd 2.4.9
Fri Apr 28 23:31:37 2023 daemon.notice pppd[16382]: pppd 2.4.9 started by root, uid 0
Fri Apr 28 23:31:48 2023 kern.info kernel: [  411.531224] r8125: eth0: link down
Fri Apr 28 23:31:48 2023 daemon.notice netifd: Network device 'eth0' link is down
Fri Apr 28 23:31:48 2023 daemon.notice netifd: Interface 'wan' has link connectivity loss
Fri Apr 28 23:31:48 2023 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Fri Apr 28 23:31:48 2023 daemon.err pppd[16382]: select (waitForPADO): Interrupted system call
Fri Apr 28 23:31:48 2023 daemon.warn pppd[16382]: Timeout waiting for PADO packets
Fri Apr 28 23:31:48 2023 daemon.err pppd[16382]: Unable to complete PPPoE Discovery
Fri Apr 28 23:31:48 2023 daemon.info pppd[16382]: Exit.
Fri Apr 28 23:31:48 2023 daemon.notice netifd: Interface 'wan' is now down
Fri Apr 28 23:31:48 2023 daemon.notice netifd: Interface 'wan6' is now down
Fri Apr 28 23:31:51 2023 daemon.notice netifd: Network device 'eth0' link is up
Fri Apr 28 23:31:51 2023 daemon.notice netifd: Interface 'wan' has link connectivity
Fri Apr 28 23:31:51 2023 daemon.notice netifd: Interface 'wan' is setting up now
Fri Apr 28 23:31:51 2023 daemon.notice netifd: Interface 'wan6' has link connectivity
Fri Apr 28 23:31:51 2023 daemon.notice netifd: Interface 'wan6' is setting up now
Fri Apr 28 23:31:51 2023 kern.info kernel: [  414.594363] r8125: eth0: link up
Fri Apr 28 23:31:51 2023 daemon.err insmod: module is already loaded - slhc
Fri Apr 28 23:31:51 2023 daemon.err insmod: module is already loaded - ppp_generic
Fri Apr 28 23:31:51 2023 daemon.err insmod: module is already loaded - pppox
[... 55 lines cut ...]
Fri Apr 28 23:32:37 2023 daemon.info pppd[17953]: Using interface pppoe-wan
Fri Apr 28 23:32:37 2023 daemon.notice pppd[17953]: Connect: pppoe-wan <--> eth0
Fri Apr 28 23:32:37 2023 daemon.info pppd[17953]: syncppp not active
Fri Apr 28 23:32:37 2023 daemon.info pppd[17953]: Remote message: Authentication Successful,Welcome!
Fri Apr 28 23:32:37 2023 daemon.notice pppd[17953]: PAP authentication succeeded
Fri Apr 28 23:32:37 2023 daemon.notice pppd[17953]: peer from calling number CC:1A:FA:EB:B7:E0 authorized
Fri Apr 28 23:32:37 2023 daemon.notice pppd[17953]: local  IP address 100.113.39.95
Fri Apr 28 23:32:37 2023 daemon.notice pppd[17953]: remote IP address 100.113.0.1
Fri Apr 28 23:32:37 2023 daemon.notice pppd[17953]: primary   DNS address 202.103.224.68
Fri Apr 28 23:32:37 2023 daemon.notice pppd[17953]: secondary DNS address 202.103.225.68
Fri Apr 28 23:32:37 2023 daemon.info avahi-daemon[5769]: Joining mDNS multicast group on interface pppoe-wan.IPv4 with address 100.113.39.95.
Fri Apr 28 23:32:37 2023 daemon.info avahi-daemon[5769]: New relevant interface pppoe-wan.IPv4 for mDNS.
Fri Apr 28 23:32:37 2023 daemon.info avahi-daemon[5769]: Registering new address record for 100.113.39.95 on pppoe-wan.IPv4.
Fri Apr 28 23:32:37 2023 daemon.notice pppd[17953]: local  LL address fe80::a236:9f56:5350:c307
Fri Apr 28 23:32:37 2023 daemon.notice pppd[17953]: remote LL address fe80::ce1a:faff:feeb:b7e0
Fri Apr 28 23:32:37 2023 daemon.info avahi-daemon[5769]: Joining mDNS multicast group on interface pppoe-wan.IPv6 with address fe80::a236:9f56:5350:c307.
Fri Apr 28 23:32:37 2023 daemon.info avahi-daemon[5769]: New relevant interface pppoe-wan.IPv6 for mDNS.
Fri Apr 28 23:32:37 2023 daemon.info avahi-daemon[5769]: Registering new address record for fe80::a236:9f56:5350:c307 on pppoe-wan.*.
Fri Apr 28 23:32:37 2023 daemon.warn dnsmasq[4538]: failed to create listening socket for fe80::ce1a:faff:feeb:b7e0%pppoe-wan: Address not available
Fri Apr 28 23:32:37 2023 daemon.warn dnsmasq[4538]: failed to create listening socket for fe80::ce1a:faff:feeb:b7e0%pppoe-wan: Address not available
Fri Apr 28 23:32:37 2023 daemon.notice netifd: Network device 'pppoe-wan' link is up
Fri Apr 28 23:32:37 2023 daemon.notice netifd: Interface 'wan' is now up
```

##  我使用其他的固件没有这个问题出现

guugg, 2023-04-28:
是不是istoreos固件无法获取猫棒数据？ 每次istoreos重启需要猫棒下发数据？

guugg, 2023-04-28:
刚刚又重启一次。
```
Sat Apr 29 00:28:24 2023 daemon.err odhcp6c[8555]: Failed to send RS (Address not available)
Sat Apr 29 00:28:24 2023 daemon.err insmod: module is already loaded - ppp_generic
Sat Apr 29 00:28:24 2023 daemon.err insmod: module is already loaded - pppox
Sat Apr 29 00:28:24 2023 daemon.err insmod: module is already loaded - pppoe
Sat Apr 29 00:28:24 2023 daemon.notice netifd: wan (8540): uci: Entry not found
Sat Apr 29 00:28:24 2023 daemon.info pppd[8579]: Plugin pppoe.so loaded.
Sat Apr 29 00:28:24 2023 daemon.info pppd[8579]: PPPoE plugin from pppd 2.4.9
Sat Apr 29 00:28:25 2023 daemon.notice pppd[8579]: pppd 2.4.9 started by root, uid 0
Sat Apr 29 00:28:25 2023 daemon.err odhcp6c[8555]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Sat Apr 29 00:28:25 2023 daemon.err odhcpd[5120]: Failed to send to fe80::b96f:436:976:95af%lan@br-lan (Address not available)
Sat Apr 29 00:28:25 2023 daemon.err odhcpd[5120]: Failed to send to ff02::1%lan@br-lan (Address not available)
Sat Apr 29 00:28:26 2023 daemon.err modprobe: failed to find a module named aufs
Sat Apr 29 00:28:26 2023 daemon.err dockerd[7608]: time="2023-04-28T16:28:26.138847101Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Sat Apr 29 00:28:26 2023 daemon.err dockerd[7608]: time="2023-04-28T16:28:26.139599674Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Sat Apr 29 00:28:26 2023 kern.warn kernel: [   27.241776] overlayfs: upperdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Sat Apr 29 00:28:26 2023 kern.warn kernel: [   27.241878] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Sat Apr 29 00:28:26 2023 kern.warn kernel: [   27.241969] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Sat Apr 29 00:28:26 2023 kern.warn kernel: [   27.242060] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Sat Apr 29 00:28:26 2023 daemon.info dnsmasq[7868]: read /etc/hosts - 4 addresses
Sat Apr 29 00:28:26 2023 daemon.info dnsmasq[7868]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sat Apr 29 00:28:26 2023 daemon.info dnsmasq-dhcp[7868]: read /etc/ethers - 0 addresses
Sat Apr 29 00:28:26 2023 daemon.err dockerd[7608]: time="2023-04-28T16:28:26.200744216Z" level=warning msg="Your kernel does not support cgroup blkio weight"
Sat Apr 29 00:28:26 2023 daemon.err dockerd[7608]: time="2023-04-28T16:28:26.200769678Z" level=warning msg="Your kernel does not support cgroup blkio weight_device"
Sat Apr 29 00:28:26 2023 daemon.err odhcp6c[8555]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Sat Apr 29 00:28:26 2023 daemon.info modprobe: nf_conntrack is already loaded
Sat Apr 29 00:28:26 2023 daemon.info modprobe: nf_conntrack_netlink is already loaded
Sat Apr 29 00:28:26 2023 daemon.info avahi-daemon[5803]: Joining mDNS multicast group on interface eth0.IPv6 with address fe80::a236:9fff:fe50:c307.
Sat Apr 29 00:28:26 2023 daemon.info avahi-daemon[5803]: New relevant interface eth0.IPv6 for mDNS.
Sat Apr 29 00:28:26 2023 daemon.info avahi-daemon[5803]: Registering new address record for fe80::a236:9fff:fe50:c307 on eth0.*.
Sat Apr 29 00:28:26 2023 daemon.info avahi-daemon[5803]: Joining mDNS multicast group on interface docker0.IPv4 with address 172.17.0.1.
Sat Apr 29 00:28:26 2023 daemon.info avahi-daemon[5803]: New relevant interface docker0.IPv4 for mDNS.
Sat Apr 29 00:28:26 2023 daemon.info avahi-daemon[5803]: Registering new address record for 172.17.0.1 on docker0.IPv4.
Sat Apr 29 00:28:27 2023 kern.info kernel: [   28.349672] r8125: eth1: link up
Sat Apr 29 00:28:27 2023 kern.info kernel: [   28.349686] br-lan: port 1(eth1) entered blocking state
Sat Apr 29 00:28:27 2023 kern.info kernel: [   28.349687] br-lan: port 1(eth1) entered forwarding state
Sat Apr 29 00:28:27 2023 daemon.notice netifd: Network device 'eth1' link is up
Sat Apr 29 00:28:27 2023 kern.warn kernel: [   28.601865] overlayfs: upperdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Sat Apr 29 00:28:27 2023 kern.warn kernel: [   28.601867] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Sat Apr 29 00:28:27 2023 kern.warn kernel: [   28.601868] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Sat Apr 29 00:28:27 2023 kern.warn kernel: [   28.601869] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Sat Apr 29 00:28:27 2023 daemon.info procd: - init complete -
Sat Apr 29 00:28:27 2023 daemon.info dnsmasq[7868]: read /etc/hosts - 4 addresses
Sat Apr 29 00:28:27 2023 daemon.info dnsmasq[7868]: read /tmp/hosts/odhcpd - 0 addresses
Sat Apr 29 00:28:27 2023 daemon.info dnsmasq[7868]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sat Apr 29 00:28:27 2023 daemon.info dnsmasq-dhcp[7868]: read /etc/ethers - 0 addresses
[... cut ...]
Sat Apr 29 00:31:26 2023 daemon.err insmod: module is already loaded - ppp_generic
Sat Apr 29 00:31:26 2023 daemon.err insmod: module is already loaded - pppox
Sat Apr 29 00:31:26 2023 daemon.err insmod: module is already loaded - pppoe
Sat Apr 29 00:31:26 2023 daemon.notice netifd: wan (14048): uci: Entry not found
Sat Apr 29 00:31:26 2023 daemon.info pppd[14065]: Plugin pppoe.so loaded.
Sat Apr 29 00:31:26 2023 daemon.info pppd[14065]: PPPoE plugin from pppd 2.4.9
Sat Apr 29 00:31:26 2023 daemon.notice pppd[14065]: pppd 2.4.9 started by root, uid 0
```

hy817120, 2023-05-01:
我也有这种问题,只要开启了ipv6,隔几分钟或者十几二十分钟断一次网,或者说网口重启, 我把ipv6关了,一直稳定.
头都整大了,现在都是把ipv6关了. 我自己做过测试, 把ipv6关了, 一直稳定,没出现过断网重拨.
但一开启问题就多了.

jjm2473, 2023-05-15:
你这个猫棒是插光口的吗？还是用网线连接的？