istoreos/istoreos #752: 关于出现xt_fullconenat：warning：   failed to register a conntrack notifier.disable active gc formatpings

lretro-ok, 2023-05-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
正常运行出现xt_fullconenat：warning：   failed to register a conntrack notifier.disable active gc formatpings应该如何解决 此问题导致nat类型不稳定及OS上面的docker容器的时延高（以前就是5ms这样）

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
如上 因为我已经重新刷过三次系统了，只有这一次刷入最新的出现这个奇怪问题，我并没有什么设置，无法详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：N5105 X86

[istoreos-21.02.3-2023042111-x86-64-squashfs-combined-efi.img.gz](https://fw0.koolcenter.com/iStoreOS/x86_64_efi/istoreos-21.02.3-2023042111-x86-64-squashfs-combined-efi.img.gz)
[image]

### (3) 详细日志和/或截图

A：

lretro-ok, 2023-05-01:
[image]

lretro-ok, 2023-05-01:
[image]

lretro-ok, 2023-05-02:
又运行大概4w秒左右再次出现问题 这次是外接硬盘盒直接掉盘

lretro-ok, 2023-05-02:
补充一下 当出现问题时候 只能进入到路由器后台登录界面 输入密码也无法进入里面 似乎无响应一样

lretro-ok, 2023-05-04:
[image]
当运行到5w小时左右就会出现问题….

jjm2473, 2023-05-12:
只要打开fullcone-nat都会有这个警告，不管哪个版本都一样。

硬盘盒的问题就复杂了，自己研究。

closed 2023-05-12 by jjm2473

antsman9, 2023-06-02:
昨天开始我也是出现同样的问题，N5105，用ventoy安装。公网 pppoe拨号，联通。
晚上9点左右，突然断网，查看后台，无法从192.168.100.1 进入。
查看到重复的语句：类似上面的xt_fullconenat：warning：   failed to register a conntrack notifier.disable active gc for matpings
重置、 重新下载新的固件，均无法进入后台，查看连接情况，无法获取到ip。
重新下载安装，quickstart，还是同样的问题，大概出现的时间是从14秒开始。
显示器连接软路由，可以输入命令。
一个SSD，一个HDD，HDD卸掉，依旧不管用。
SSD为光威256G。
麻烦可以指导一下，如何解决吗？

jjm2473, 2023-06-02:
都说了这只是个fullconenat的警告，不管在哪个机器上都有的。