istoreos/istoreos #759: 希望加入nat高性能模式

18312447181, 2023-05-05:
能否加入nat高性能模式，要不然跑网心云网络很不稳定。最新高性能Fullcone 新版PCDN专用优化固件发布，支持30w+并发NAT高能连接~~防火墙中手动设为高性能模式即可

guozenghuang, 2023-05-06:
关注一下。ikuai同样设备网心收益会比istoreos高，没特殊设置，两者都是开启upnp，而ikuai有nat1设置项，而istoreos只有上面提到的防火墙FullconeNAT，无其他设置选项。

jjm2473, 2023-05-12:
跟upnp完全没关系好吧，upnp是给内网设备用的

GreenTeodoro839, 2023-05-24:
那东西是lean搞的，官方openwrt源码没有

jjm2473, 2023-05-27:
@guozenghuang  fullcone-nat就是最高级别的NAT了，也就是NAT1。你的网心云是运行在内网设备还是网关设备上？如果是网关设备，那默认已经开了50000以上端口，只要你有公网IP，那就已经开放出去了，不需要看UPNP和NAT（不要看网心云的傻逼提示）。如果是在内网设备，那网关需要提供NAT或者UPNP，UPNP的高级设置关掉STUN，同时打开强制转发。

GreenTeodoro839, 2023-05-27:
> @guozenghuang fullcone-nat就是最高级别的NAT了，也就是NAT1。你的网心云是运行在内网设备还是网关设备上？如果是网关设备，那默认已经开了50000以上端口，只要你有公网IP，那就已经开放出去了，不需要看UPNP和NAT（不要看网心云的傻逼提示）。如果是在内网设备，那网关需要提供NAT或者UPNP，UPNP的高级设置关掉STUN，同时打开强制转发。

他说的是lean的fullcone高性能模式，相同配置可以承受更高的连接数

DolphinAC, 2023-09-07:
> @guozenghuang fullcone-nat就是最高级别的NAT了，也就是NAT1。你的网心云是运行在内网设备还是网关设备上？如果是网关设备，那默认已经开了50000以上端口，只要你有公网IP，那就已经开放出去了，不需要看UPNP和NAT（不要看网心云的傻逼提示）。如果是在内网设备，那网关需要提供NAT或者UPNP，UPNP的高级设置关掉STUN，同时打开强制转发。

> @guozenghuang fullcone-nat就是最高级别的NAT了，也就是NAT1。你的网心云是运行在内网设备还是网关设备上？如果是网关设备，那默认已经开了50000以上端口，只要你有公网IP，那就已经开放出去了，不需要看UPNP和NAT（不要看网心云的傻逼提示）。如果是在内网设备，那网关需要提供NAT或者UPNP，UPNP的高级设置关掉STUN，同时打开强制转发。

> @guozenghuang fullcone-nat就是最高级别的NAT了，也就是NAT1。你的网心云是运行在内网设备还是网关设备上？如果是网关设备，那默认已经开了50000以上端口，只要你有公网IP，那就已经开放出去了，不需要看UPNP和NAT（不要看网心云的傻逼提示）。如果是在内网设备，那网关需要提供NAT或者UPNP，UPNP的高级设置关掉STUN，同时打开强制转发。

50000以上端口默认打开是哪条规则，在哪里看。我找了一下好像没看到哇

DolphinAC, 2023-09-07:
> @guozenghuang fullcone-nat就是最高级别的NAT了，也就是NAT1。你的网心云是运行在内网设备还是网关设备上？如果是网关设备，那默认已经开了50000以上端口，只要你有公网IP，那就已经开放出去了，不需要看UPNP和NAT（不要看网心云的傻逼提示）。如果是在内网设备，那网关需要提供NAT或者UPNP，UPNP的高级设置关掉STUN，同时打开强制转发。

50000以上端口默认打开是哪条规则，在哪里看。我找了一下好像没看到哇

GreenTeodoro839, 2023-09-07:
> > @guozenghuang fullcone-nat就是最高级别的NAT了，也就是NAT1。你的网心云是运行在内网设备还是网关设备上？如果是网关设备，那默认已经开了50000以上端口，只要你有公网IP，那就已经开放出去了，不需要看UPNP和NAT（不要看网心云的傻逼提示）。如果是在内网设备，那网关需要提供NAT或者UPNP，UPNP的高级设置关掉STUN，同时打开强制转发。
> 
> 50000以上端口默认打开是哪条规则，在哪里看。我找了一下好像没看到哇

不是规则，lean改了很多地方，没必要移植。版本跨度太大，移植很费事，而且不一定能在最新openwrt正常工作

jjm2473, 2023-09-08:
@DolphinAC 从istore安装的网心云会自动打开50000以上端口，如果是自己手动用docker命令安装的话，那就自己在“防火墙-通信规则”那边添加个允许入站的规则，可以自己设置端口范围，建议只开放10000以上的端口

jjm2473, 2023-09-08:
@DolphinAC 如果不是软路由上运行网心云，那么就是在上级路由开端口转发

GreenTeodoro839, 2023-09-08:
> @DolphinAC 从istore安装的网心云会自动打开50000以上端口，如果是自己手动用docker命令安装的话，那就自己在“防火墙-通信规则”那边添加个允许入站的规则，可以自己设置端口范围，建议只开放10000以上的端口

那个不是开放端口，lean的fullcone-nat可以跑到30万连接数，普通的linux最多是5万多

jjm2473, 2023-09-10:
@GreenTeodoro839 我的意思是软路由安装网心云的话，跟NAT没有任何关系。有谁不知道NAT跟开放端口不一样？

GreenTeodoro839, 2023-09-10:
> @GreenTeodoro839 我的意思是软路由安装网心云的话，跟NAT没有任何关系。有谁不知道NAT跟开放端口不一样？

不是nat，那个高性能模式是突破linux默认的连接数限制，提高pcdn性能

closed 2024-01-12 by jjm2473