istoreos/istoreos #764: webdav服务 客户端无法重名移动文件

GitHubmnxz, 2023-05-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：webdav服务 客户端无法重名移动文件  读写都是正常的
      Samba服务一切都正常
各类客户端都报错403
[image]

[image]

### (2) 路由器型号和固件版本

A：固件版本
iStoreOS 21.02.3 2023042111
内核版本
5.4.188

### (3) 详细日志和/或截图

A：
[image]

[image]

GitHubmnxz, 2023-05-09:
Gowebdav服务也无法启动

jjm2473, 2023-05-12:
是不是跨根目录移动了