istoreos/istoreos #767: 光猫路由模式，istoreos后台已经有公网IPV6地址，但是下属设备无法获取IPV6

928882650, 2023-05-10:
反馈bug/问题模板，提建议请删除

### (1) 具体问题

A：光猫路由模式，istoreos后台已经有公网IPV6地址，但是下属设备无法获取IPV6

### (2) 路由器型号和固件版本

A：R4S/ iStoreOS 21.02.3 2023042111

### (3) 详细日志和/或截图

A：
[image]

有会弄的可以付费928882650@qq.com

magitwit, 2023-05-13:
不用付费 官方有教程“iStoreOS中Ipv6设置方法” ，就一个开关的问题
https://www.bilibili.com/video/BV1g84y1r7nY

928882650, 2023-05-14:
> 不用付费 官方有教程“iStoreOS中Ipv6设置方法” ，就一个开关的问题 https://www.bilibili.com/video/BV1g84y1r7nY

你发的这个视频好像是光猫桥接，软路由拨号的，我的是光猫拨号的，试了很多次了，软路由后面的设备都没有办法通过IPV6测试，如果你会弄，可以付费帮我设置下

alexmadgame, 2023-05-15:
个人猜测，你这个v6地址已经是/128了，/128不能再往下分配v6地址，建议换成路由拨号（虽然我这也有设备是光猫/64下级路由也是/64，比较奇怪）

My-By, 2023-05-25:
得先确定你的光猫有开启ipv6，试着直接接网线到电脑，查看适配器选项里有没有获取到ipv6地址，如果有，就是设置问题，如果没有就是光猫没开ipv6。

928882650, 2023-05-25:
有的

 

AiGeek
***@***.***

 

 
&mdash;

My-By, 2023-05-25:
我晚点加你微信帮你看看，现在出门，

928882650, 2023-05-25:
好的 不急！
 

AiGeek
***@***.***

aaa3313, 2023-06-11:
请问问题解决了吗，我和你一样的情况，求指导

aaa3313, 2023-06-11:
> 
>  
>  
>   
>   我晚点加你微信帮你看看，现在出门，

大佬，顺便指导一下我，问题和他一样

jjm2473, 2023-07-12:
看起来没分配到ipv6-pd，应该没法再往下分了

shaojs321, 2023-07-12:
我也是光猫拨号，软路由当二级路由，这是我换别的版本在系统->高级设置中网络和DHCP的配置
网络：
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdc4:ec98:e4ab::/48'
	option packet_steering '1'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth1'

config device
	option name 'eth1'
	option macaddr 'XX:XX:XX:XX:XX:XX'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.3.1'
	option netmask '255.255.255.0'
	option ip6assign '60'
	list ip6class 'wan6'

config interface 'wan'
	option device 'eth0'
	option proto 'dhcp'

config interface 'wan6'
	option device 'eth0'
	option proto 'dhcpv6'
	option reqaddress 'try'
	option reqprefix 'auto'
	list ip6class 'wan6'

DHCP配置：
config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '0'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option localservice '0'
	option ednspacket_max '1232'
	option port '53'
	option dns_redirect '1'
	option allservers '1'
	option min_ttl '3600'
	option dnsforwardmax '8192'
	option localuse '1'
	option noresolv '1'
	option cachesize '0'
	option filter_aaaa '0'
	list server '127.0.0.1#7874'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '60'
	option dhcpv4 'server'
	option dhcpv6 'server'
	option ra 'server'
	option force '1'
	list ra_flags 'other-config'
	option ra_maxinterval '10'
	option ra_mininterval '1'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

config host
	option name 'debian12'
	option dns '1'
	option mac 'XX:XX:XX:XX:XX:XX'
	option ip '192.168.3.170'
	option leasetime 'infinite'

你可以对照着来

shaojs321, 2023-07-12:
> 我也是光猫拨号，软路由当二级路由，这是我换别的版本在系统->高级设置中网络和DHCP的配置 网络： config interface 'loopback' option device 'lo' option proto 'static' option ipaddr '127.0.0.1' option netmask '255.0.0.0'
>
> config globals 'globals' [...]

[image]
lan口得这样才可以

GitHubScorpiocat, 2023-08-08:
建议开发者在系统工具中添加 一键开启IPV6的 功能~

Satone7, 2023-09-24:
> [image] lan口得这样才可以

您好，请教一下是怎么配置让lan口拿到ipv6的

My-By, 2023-09-24:
> > > 我也是光猫拨号，软路由当路由路由，这是我换其他版本在系统->高级设置中网络和DHCP的配置网络：config interface 'loopback' option device 'lo' option proto 'static' option ipaddr '127.0.0.1' 选项网络掩码 '255.0.0.0'
> > > 配置全局变量 'globals' 选项 ula_prefix 'fdc4:ec98:e4ab::/48' 选项 packet_steering '1' 配置
> > > 设备选项名称 'br-lan' 选项类型 'bridge'列出端口 'eth1'
> > > 配置设备选项名称 'eth1' 选项 macaddr 'XX:XX:XX:XX:XX:XX'
> > > 配置接口 'lan' 选项设备 'br-lan' 选项 proto 'static' 选项 ipaddr '192.168.3.1 ' 选项网络掩码 '255.255.255。0' 选项 ip6assign '60' 列表 ip6class 'wan6'
> > > 配置接口 'wan' 选项设备 'eth0' 选项原型 'dhcp'
> > > config interface 'wan6' option device 'eth0' option proto 'dhcpv6' option reqaddress 'try' option reqprefix 'auto' list ip6class 'wan6' DHCP配置： config dnsmasq option domainneeded '1' 
> > > option localise_queries '1' option rebind_protection '0'选项本地'/lan/' 选项域'lan' 选项expandhosts '1' 选项权威'1' 选项readethers '1' 选项leasefile '/tmp/dhcp.租约'选项localservice'0'选项ednspacket_max'1232'选项端口'53'选项dns_redirect'1'选项allservers'1'选项min_ttl'3600'选项dnsforwardmax'8192'选项localuse'1'选项noresolv'1'选项cachesize' 0'选项filter_aaaa'0'列表服务器'127.0.0.1#7874'
> > > 配置dhcp'lan'选项接口'lan'选项启动'100'选项限制'150'选项leasetime'60'选项dhcpv4'服务器'选项dhcpv6'服务器'选项ra'服务器'选项强制'1'列表ra_flags'其他-配置'选项ra_maxinterval'10'选项ra_mininterval'1'
> > > 配置dhcp'wan'选项接口'wan'选项忽略'1'
> > > 配置odhcpd'odhcpd'选项maindhcp'0'选项leasefile'/tmp/hosts/odhcpd'选项leasetrigger' /usr/sbin/odhcpd-update' 选项 loglevel '4'
> > > 配置主机选项名称 'debian12' 选项 dns '1' 选项 mac 'XX:XX:XX:XX:XX:XX' 选项 ip '192.168.3。170' 选项租约时间'无限'
> > > 你可以对照着来
> > 
> > 
> > [image]lan口得这样才可以
> 
> 您好，请教一下怎么配置让lan口获得ipv6的

你是要给下级设备分配ipv6还是要给软路由获取ipv6。

jjm2473, 2023-09-24:
> 建议开发者在系统工具中添加 一键开启IPV6的 功能~

默认就是开的，只是DNS不解析，可以在DHCP/DNS的高级设置关掉“禁用AAAA解析”就行。
但是现在的问题是地址都没分配。

rumianoesa, 2023-10-12:
光猫拨号，openwrt做二级路由的话ipv6可以用中继模式
wan6的dhcp服务器ipv6设置勾上指定主接口，下面全部选择中继模式
lan口的dhcp服务器也全都设置成中继模式

Mr-XiaoYi, 2023-10-19:
[OpenWRT系统获取到Ipv6地址但不下发Ipv6地址解决过程](https://www.scczz.com/1056.html) 
https://www.scczz.com/1056.html
折腾一天找到的解决方案，希望帮到各位，有心的朋友点点赞赏😘

jjm2473, 2023-10-19:
如果是软路由拨号，试试将软路由lan口的ipv6服务全部禁用。
如果是光猫拨号，软路由作为二级路由，参考 @rumianoesa  提供的方法。

jjm2473, 2023-12-15:
在配置好联网以后，istore里面安装系统便利工具（systools），然后执行“开启IPv6”试试，会自动根据当前联网方式调整ipv6设置，可能可以解决问题

luckymrwang, 2023-12-15:
> 在配置好联网以后，istore里面安装系统便利工具（systools），然后执行“开启IPv6”试试，会自动根据当前联网方式调整ipv6设置，可能可以解决问题

感谢提供解决思路，我试试

138161, 2024-04-05:
第一步，先搞定wan6口
1、网络--接口--wan6--编辑
1.1、常规设置
     协议：DHCPv6客户端
         设备：eth0
         开机自动运行：选择
         请求IPv6地址：try
         请求指定长度的IPv6前缀：自动
       
1.2、高级设置
         强制链路：不选
         请求DHCP时发送的客户端ID：空
         使用默认网关：选择
         自动获取DNS服务器：选择
         DNS 权重：空
     使用网关跃点：空
     覆盖 IPv4 路由表：未指定
     覆盖 IPv6 路由表：未指定
     IPv6 源路由：选择
     委托 IPv6 前缀：选择
     IPv6 分配长度：已禁用
     IPv6 前缀过滤器：-- 请选择 --
     IPv6 后缀：::1
     IPv6 优先级：空
     
                          
1.3、防火墙设置
     创建和分配防火墙区域：wan和wan6
     
1.4、DHCP服务器--常规设置
     忽略此接口：不选
     
1.5、DHCP服务器--IPv6设置
     指定的主接口：选择
     RA 服务：中继模式
     DHCPv6 服务：中继模式
     NDP 代理：中继模式
     学习路由：选择
     

第二步，搞定lan口
2、网络--接口--lan--编辑
2.1、常规设置
     协议：静态地址
     设备：br-lan
     开机自动运行：选择
     IPv4 地址：192.168.100.1
     IPv4 子网掩码：255.255.255.0
     IPv4 网关：灰色无法更改
     IPv4 广播地址：灰色无法更改
     IPv6 地址：空
     IPv6 网关：空
     IPv6 路由前缀：空
     
2.2、高级设置
     强制链路：选择
     使用默认网关：选择
     使用自定义的 DNS 服务器：空
     DNS 搜索域名：空
     DNS 权重：空
     使用网关跃点：空
     覆盖 IPv4 路由表：未指定
     覆盖 IPv6 路由表：未指定
     委托 IPv6 前缀：选择
     IPv6 分配长度：已禁用
     IPv6 前缀过滤器：-- 请选择 --
     IPv6 后缀：::1
     IPv6 优先级：空
     
                       
[... 167 lines cut ...]
tcpdump  -nn -i eth0 icmp6 -v
tcpdump  -nn -i eth0 icmp6 -vv

查看邻居mac地址
ip -6 neighbor show 

检查哪些端口和应用在运行，如过滤19422相关内容
netstat -anp -p udp
netstat -anp -p udp | grep 19422

ipv6 DNS服务器
2001:4860:4860::8888
2001:4860:4860::8844
240e:4c:4008::1
240e:4c:4808::1
2001:dc7:1000::1
240C::6666
240C::6644
2400:3200::1
2402:4e00::
2400:da00::6666

重启网络服务
/etc/init.d/firewall restart
/etc/init.d/network restart

138161, 2024-04-05:
上面的都瞎扯，slaac方式下，icmpv6消息被防火墙挡住了，不改防火墙配置，怎么连？

xiazunyang, 2024-09-25:
> [OpenWRT系统获取到Ipv6地址但不下发Ipv6地址解决过程](https://www.scczz.com/1056.html) https://www.scczz.com/1056.html 折腾一天找到的解决方案，希望帮到各位，有心的朋友点点赞赏😘

6的，按你的方法成功让局域网的设备获取到了IPV6