istoreos/istoreos #770: 这次是迅游加速器无法正常使用

cero-ss, 2023-05-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
使用迅游加速主机无效但是加速pc却能正常使用
### (1) 具体问题

A：之前我用的uu加速器，以前uu也用不了，后来修复了可以正常使用，最近会员过期了我就想试试迅游加速器，这样也能多一个选择，我在使用时发现了这个问题，当插件识别到pc时不管是steam社区还是英雄联盟台服都能正常加速，加速后按照提示安装证书就能正常使用了，而我用插件加速ps5时却一直是无效的，界面提示加速成功，但无论是加速日服还是美服，进游戏里比如apex英雄都是默认的港服。

既然pc能够加速成功，为什么主机却不行，pc能用说明插件有在正常工作，我把问题反馈给官方客服，人家说可能是openwrt系统和不同路由器型号兼容性问题。神奇的是迅游的插件是可以手动更改识别的设备的，我把默认识别的ps5改成了pc，然后选择加速apex-steam选择日服，进入ps5居然加速成功了能够正常联机，我大为震撼！
我还测试了我的xbox和ps5是一样的问题，改成pc后也能正常加速联机，虽然我还有ns但我估计也是一样的。

### (2) 路由器型号和固件版本

A：FriendlyElec NanoPi R5S

### (3) 详细日志和/或截图

A：

[image]
[image]

jjm2473, 2023-05-16:
固件版本

closed 2023-05-16 by jjm2473

jjm2473, 2023-05-18:
根据你描述的，应该是迅游自己的策略有问题

cero-ss, 2023-05-18:
> 根据你描述的，应该是迅游自己的策略有问题

我也是这么和客服说的，但是人家说什么openwrt系统和不同路由器型号兼容性问题，也说会把我的问题记录一下，他们自己也和硬路由厂商合作，难道只有软路由插件会有这样的问题吗，我也没有其它设备可以测试。还有其它可能影响加速的插件我都关闭了，这些我在之前已经测试过了

jjm2473, 2023-05-27:
你这个固件什么版本，issue模板里面不是让提供固件版本吗，复制一下版本号这么难？

cero-ss, 2023-05-27:
> 你这个固件什么版本，issue模板里面不是让提供固件版本吗，复制一下版本号这么难？

我记得我前面有写是最新版本吧
[image]