istoreos/istoreos #771: 关于端口转发，不转发来源IP的逻辑问题

yh317, 2023-05-13:
关于端口转发的逻辑问题
（未找到类似issue）

### (1) 具体问题

端口转发，转发的数据包中不含数据包的来原公网IP，来源为路由器的IP。
这使得某些防火墙策略（如群晖通过来源IP屏蔽某些恶意攻击的IP地址）无效。
应该增加一个选项，转发数据包时是否保留来源IP地址。

### (2) 路由器型号和固件版本

E3845工控机
iStoreOS 21.02.3 2023042111 / LuCI istoreos-21.02 branch git-23.088.34636-f8cc111

### (3) 详细日志和/或截图

A：群晖后台登陆截图：
[image]

jjm2473, 2023-05-15:
自己看看防火墙里面有没有这样的设置，如果没有，那就没有，我们不会去实现的。防火墙是openwrt的基础功能，如果你觉得不够用，应该去openwrt的项目提建议。

jjm2473, 2023-05-15:
如果按你说的，那路由器自己也没办法访问群晖啊。

jjm2473, 2023-05-15:
路由器转发以后，来源ip是不是变成路由器的内网IP了？如果这个路由器的内网IP都访问不了群晖，那不就代表路由器自己是访问不了群晖的吗？

yh317, 2023-05-15:
是路由器转发外网数据包，但是来源IP并不是非得是路由器的。正常的情况应该能显示来源的公网IP地址。
我也找到解决方案了。在网络-防火墙里面配置一下nat规则就可以了。
在NAT规则里配置下“防止重写源”就可以了。
[image]
配置好后，群晖的显示应该是这样的：
[image]
虽然也是由路由器转发，但是可以识别到外网IP地址了。这样对某些IP的屏蔽规则就生效了。
总之，还是感谢大佬的帮助。