istoreos/istoreos #781: R4S最新istoreos固件USB-AC55网卡没有支持

tinobara, 2023-05-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：R4S 使用新版固件,升级后,无线网卡不能工作
移去后新增加提示错误
TypeError
Cannot read properties of undefined (reading 'value')
[image]

重新卡刷后,没有了无线选项
[image]

### (2) 路由器型号和固件版本

路由器型号：R4S
固件版本：istoreos-21.02.3-2023051217-r4s-squashfs.img
USB网卡型号：USB-AC55双频Wireless-AC1300无线网卡
[image]

### (3) 详细日志和/或截图

A：
在固件istoreos-21.02.3-2023032012-r4s-squashfs.img运行 lsusb -tv 的结果
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003  
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    ID 1d6b:0002  
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003  
    |__ Port 1: Dev 2, If 0, Class=, Driver=mt76x2u, 5000M
        ID 0b05:17eb  
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    ID 1d6b:0002  
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
    ID 1d6b:0001  
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
    ID 1d6b:0001  
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    ID 1d6b:0002  
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    ID 1d6b:0002  

在固件istoreos-21.02.3-2023051217-r4s-squashfs.img运行 lsusb -tv 的结果
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003  
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    ID 1d6b:0002  
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003  
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    ID 1d6b:0002  
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
    ID 1d6b:0001  
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
    ID 1d6b:0001  
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    ID 1d6b:0002  
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    ID 1d6b:0002

jjm2473, 2023-05-18:
应该是之前删掉了很多内核模块导致的，你可以先用这个命令安装回去`opkg update; opkg install kmod-mt76x2u`

tinobara, 2023-05-18:
> 应该是之前删掉了很多内核模块导向的，你可以先用这个命令安装回去`opkg update; opkg install kmod-mt76x2u`

好的,最新固件通过命令安装后无线功能正常了

closed 2023-05-18 by tinobara