istoreos/istoreos #783: istoreos这么好用既然支持DDNSTO，为什么没有Tailscale呢？

ZTOA10, 2023-05-18:
如题：istoreos这么好用既然支持DDNSTO，为什么没有Tailscale呢？

jjm2473, 2023-05-18:
直接opkg就能安装啊，关键是要会配置

cero-ss, 2023-05-18:
我用的就是tailscale
手动下载安装tailscale和tailscaled
运行tailscale up --advertise-exit-node --accept-routes --advertise-routes=192.168.31.0/24   [这里填上软路由地址，如果是192.168.100.1就填192.168.100.0/24]，复制给的地址到浏览器上
然后在接口里新建一个选择tailscale设备，填上给的虚拟ip地址就行了
[image]
[image]

qzydustin, 2023-06-02:
OpenWRT的教程可以在iStoreOS上正常使用，包括exit node和subnet router
https://openwrt.org/docs/guide-user/services/vpn/tailscale/start

jjm2473, 2023-06-08:
tailscale还有个docker版，好像界面更好用吧

closed 2023-06-08 by jjm2473

h6fYe34KFSiH, 2023-09-26:
手动安装成功了，分享下。下载对应的ipk，然后安装。

wget https://downloads.openwrt.org/snapshots/packages/aarch64_generic/packages/tailscale_1.48.2-1_aarch64_generic.ipk
wget https://downloads.openwrt.org/snapshots/packages/aarch64_generic/packages/tailscaled_1.48.2-1_aarch64_generic.ipk

opkg install tailscale_1.48.2-1_aarch64_generic.ipk tailscaled_1.48.2-1_aarch64_generic.ipk

zhouxiaoxing91, 2024-01-09:
https://www.zyh8.com/archives/135/openwrt%E5%AE%89%E8%A3%85%E9%85%8D%E7%BD%AEtailscale/
OpenWrt安装配置Tailscale

这个教程说的很好

jjm2473, 2024-01-10:
istore早已经上架了tailscale